luxonis/depthai-core #324: Dependencies not installed alongside depthai

NatalieWolfe, 2022-01-12:
Built and installed following instructions from [the readme](https://github.com/luxonis/depthai-core#building):
```sh
$ sudo apt install libusb-1.0-0-dev libopencv-dev
$ git clone git@github.com:luxonis/depthai-core
$ cd depthai-core
$ git submodule update --init --recursive
$ cmake -S. -Bbuild -D'CMAKE_INSTALL_PREFIX=/usr/local'
$ cmake --build build --parallel 30
$ sudo cmake --build build --target install
```

Attempt to compile anything which includes `depthai/depthai.hpp` such as:
```cpp
// test.cpp
#include <depthai/depthai.hpp>

int main() {
  return 0;
}
```

And the result is missing headers for `nop`.
```sh
$ g++ test.cpp 
In file included from /usr/local/include/depthai-shared/common/Point3f.hpp:7,
                 from /usr/local/include/depthai-shared/common/Extrinsics.hpp:6,
                 from /usr/local/include/depthai-shared/common/CameraInfo.hpp:4,
                 from /usr/local/include/depthai-shared/common/EepromData.hpp:7,
                 from /usr/local/include/depthai/device/CalibrationHandler.hpp:6,
                 from /usr/local/include/depthai/depthai.hpp:9,
                 from test.cpp:2:
/usr/local/include/depthai-shared/utility/Serialization.hpp:9:10: fatal error: nop/serializer.h: No such file or directory
    9 | #include <nop/serializer.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

These deps are clearly cached somewhere by cmake as the library built fine. Could they also be installed alongside depthai?

Luxonis-Brandon, 2022-01-13:
Sorry about the delay.  Asking.

themarpe, 2022-01-13:
@NatalieWolfe do you consume the library with CMake afterwards or just point the include path to the installation directory (or leave it as is as that is the default)?
We don't install the rest of the dependencies to  the same installation directory, but to `[prefix]/lib/cmake/depthai/dependencies/...` (along some other paths) which you should point to manually if not using CMake based build system to consume depthai. Refer to: https://github.com/luxonis/depthai-core#non-cmake-integration-visual-studio-xcode-codeblocks- for more details

NatalieWolfe, 2022-01-14:
My project is using Bazel as its build system. Typically, CMake projects are trivial to integrate into a Bazel project as there is a build rule specifically for CMake. I have not encountered Hunter before, and it does not behave well with Bazel so I opted to try a system install of depthai instead.

After adding build rules for the various things under `/usr/local/lib/cmake/depthai/dependencies` and `/usr/local/include/depthai-shared/3rdparty` I have it building now. Apologies for missing that in the instructions and thank you for your help.

Now to figure out why my desktop wont detect the camera but my laptop does...

closed 2022-01-14 by NatalieWolfe

themarpe, 2022-01-14:
You may also consume DepthAI project after its installed (or only just build) if Bazel supports CMake. 
But not as adding it to the build system itself and building it, but just "find_package" consuming it (if Bazel supports that CMake part).

Glad to hear you got it working:)

(Note - check if udev rules are installed if perhaps your desktop is Linux and laptop Windows/macOS)

NatalieWolfe, 2022-01-14:
Bazel knows how to configure, build, and extract artifacts from projects with CMake, but it doesn't support CMake's packaging system. Bazel tries to treat it as a fancy Makefile.

The udev rules were indeed what I was missing. Both machines are linux, but I had played with the python demo on the laptop and that instructs you to add the rules if they're missing. Up and running now, thank you!

themarpe, 2022-08-09:
@cyberbeat 

`nop/serializer.h` is available under: `lib/cmake/depthai/dependencies/include/`. You must link to this target (with CMake, `lib/cmake/depthai/dependencies/lib/cmake/libnop/libnopConfig.cmake`) or if you are specifying include flags, add: `-I lib/cmake/depthai/dependencies/include` as well.

The 3rdparty folder, is a subset of shared external dependencies. Those are a bit misplaced and should preferably be in either private dependency or external.

`BUILD_SHARED_LIBS` is standard CMake variable: https://cmake.org/cmake/help/v3.0/variable/BUILD_SHARED_LIBS.html, which is why it isn't explicitly documented again.

cyberbeat, 2022-08-10:
@themarpe
this seems more like an ugly workaround than a solution? Or at least it should be documented? Please notice that the nop/serializer.h is included from the 
`#include <depthai/depthai.hpp>`
header. It should really "just work" if I have in CMakeLists.txt:

```
find_package(depthai CONFIG REQUIRED)
...
...
target_link_libraries(myproject PRIVATE ${DEPTHAI_SHARED_LIBS})
```
Perhaps you could set some include dir variable in your cmake files, which you could document, like

`INCLUDE_DIRECTORIES(${DEPTHAI_INCLUDE_DIR})`

like it is does in some other projects?

cyberbeat, 2022-08-10:
About the `BUILD_SHARED_LIBS`: the cmake documentation says, that you should explicit set this variable as an option:
https://cmake.org/cmake/help/latest/variable/BUILD_SHARED_LIBS.html
`
If present and true, this will cause all libraries to be built shared unless the library was explicitly added as a static library. This variable is often added to projects as an [option()](https://cmake.org/cmake/help/latest/command/option.html#command:option) so that each user of a project can decide if they want to build the project using shared or static libraries.
`

themarpe, 2022-08-10:
@cyberbeat 

libnop is included as part of `depthaiDependencies.cmake` as well as specified as PUBLIC linking.
This should indeed work out of the box. 

Do you mind posting an MRE where this happens? I haven't stumbled across this yet.
Which CMake version are you on?

> About the BUILD_SHARED_LIBS: the cmake documentation says, that you should explicit set this variable as an option

It doesn't say you `should` but you `may` - but I agree, its nice having that as option so it shows up in eg `ccmake` et al. 
Do you mind opening a PR against `develop` with this added?

cyberbeat, 2022-08-10:
Sorry, my cmake file was wrong, so first problem seems solved. The second one is a one liner, for example

`option(BUILD_SHARED_LIBS "Build shared libs" OFF)`
or
`option(BUILD_SHARED_LIBS "Build shared libs" ON)`