luxonis/depthai-core #327: v2.14.1 fails `tests/pipeline_test` with `mutex.cpp(64): mutex destroyed while busy`

diablodale, 2022-01-13:
Today I rebased on to v2.14.1 and now `tests/pipeline_test` does not pass. When it runs, I get reasonable output but the program hangs before exiting.

The same code rebased on v2.13.3 (my setup yesterday) does not have this problem.

### Setup

* Microsoft Windows [Version 10.0.19044.1466]
* VS 16.11.9
* depthai-core v2.14.1 + xlink/move fixes

### Repro

1. config x64, debug, shared libs, all tests, all examples
2. build
3. run `build/tests/pipeline_test.exe`

### Result

```
===============================================================================
All tests passed (11 assertions in 4 test cases)

d:\a01\_work\12\s\src\vctools\crt\github\stl\src\mutex.cpp(64): mutex destroyed while busy
```

...and the program never exits. It is hung.

When debugging...
1. I get the two exceptions on lines 47 and 59 as expected, and continue after the 2nd exception
2. the program is hung or looping somewhere, so I pause the app to look at the threads

There is only one thread. Here is call stack
```
ntdll.dll!00007ff88a170764() (Unknown Source:0)
ntdll.dll!00007ff88a134021() (Unknown Source:0)
KernelBase.dll!00007ff887b7ce89() (Unknown Source:0)
msvcp140d.dll!00007ff8605327d7() (Unknown Source:0)
msvcp140d.dll!00007ff860532723() (Unknown Source:0)
msvcp140d.dll!00007ff860532b65() (Unknown Source:0)
depthai-cored.dll!std::condition_variable::wait(std::unique_lock<std::mutex> & _Lck) Line 609 (c:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\mutex:609)
depthai-cored.dll!std::condition_variable::wait<bool <lambda>(void)>(std::unique_lock<std::mutex> & _Lck, backward::SignalHandling::crash_handler::__l9::bool <lambda>(void) _Pred) Line 615 (c:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\mutex:615)
depthai-cored.dll!backward::SignalHandling::crash_handler(int skip, _CONTEXT * ct) Line 4426 (c:\.hunter\_Base\9b4e732\07a9f17\7dfc749\Install\include\backward.hpp:4426)
depthai-cored.dll!backward::SignalHandling::signal_handler(int __formal) Line 4385 (c:\.hunter\_Base\9b4e732\07a9f17\7dfc749\Install\include\backward.hpp:4385)
ucrtbased.dll!00007ff860398e05() (Unknown Source:0)
ucrtbased.dll!00007ff86039ab29() (Unknown Source:0)
msvcp140d.dll!00007ff8605339e1() (Unknown Source:0)
msvcp140d.dll!00007ff8605336b2() (Unknown Source:0)
depthai-cored.dll!std::_Mutex_base::~_Mutex_base() Line 45 (c:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\mutex:45)
depthai-cored.dll!std::mutex::~mutex() (Unknown Source:0)
depthai-cored.dll!dai::Resources::~Resources() Line 373 (c:\njs\depthai-core\src\utility\Resources.cpp:373)
depthai-cored.dll!`dai::Resources::getInstance'::`2'::`dynamic atexit destructor for 'instance''() (Unknown Source:0)
ucrtbased.dll!00007ff8603a2931() (Unknown Source:0)
ucrtbased.dll!00007ff8603a2105() (Unknown Source:0)
```

Destructor `~Resources()` is running. It is either exiting the destructor or at the last line `if(lazyThreadBootloader.joinable()) lazyThreadBootloader.join();`

Then the thread bounces in/out of the OS to reach `backward::SignalHandling::signal_handler(int)`
and then goes "no place good" trying to use conditional variables and/or mutex during program shutdown...which it could be that some of those are being destroyed by the runtime while catch is trying to use them. 🤷‍♀️

Rebuilding without backwards...
The program makes the same console output above and then a standard C runtime dialog box appears
[image]

If I run in the debugger, I get slightly different callstack
```
ntdll.dll!00007ff88a1e2174() (Unknown Source:0)
ntdll.dll!00007ff88a17ae33() (Unknown Source:0)
KernelBase.dll!00007ff887b7f9b9() (Unknown Source:0)
combase.dll!00007ff889b06c87() (Unknown Source:0)
combase.dll!00007ff889b069ae() (Unknown Source:0)
combase.dll!00007ff889b37ac0() (Unknown Source:0)
combase.dll!00007ff889b32aff() (Unknown Source:0)
ntdll.dll!00007ff88a0e9a1d() (Unknown Source:0)
ntdll.dll!00007ff88a13c1e7() (Unknown Source:0)
ntdll.dll!00007ff88a13bf7a() (Unknown Source:0)
ntdll.dll!00007ff88a13c000() (Unknown Source:0)
ntdll.dll!00007ff88a10d937() (Unknown Source:0)
ntdll.dll!00007ff88a0efbae() (Unknown Source:0)
ntdll.dll!00007ff88a0e73e4() (Unknown Source:0)
ntdll.dll!00007ff88a0e6af4() (Unknown Source:0)
KernelBase.dll!00007ff887b3ade2() (Unknown Source:0)
user32.dll!00007ff8892f2e57() (Unknown Source:0)
ntdll.dll!00007ff88a170ba4() (Unknown Source:0)
win32u.dll!00007ff887f81ec4() (Unknown Source:0)
user32.dll!00007ff8892d7d1b() (Unknown Source:0)
user32.dll!00007ff8892d66c1() (Unknown Source:0)
user32.dll!00007ff8892fd957() (Unknown Source:0)
user32.dll!00007ff88934cbdb() (Unknown Source:0)
user32.dll!00007ff88934b4f5() (Unknown Source:0)
user32.dll!00007ff88934c2e7() (Unknown Source:0)
user32.dll!00007ff88934c36e() (Unknown Source:0)
ucrtbased.dll!00007ff8432a5367() (Unknown Source:0)
ucrtbased.dll!00007ff8432b346d() (Unknown Source:0)
ucrtbased.dll!00007ff8432b3333() (Unknown Source:0)
ucrtbased.dll!00007ff8432b3547() (Unknown Source:0)
ucrtbased.dll!00007ff8432b44a7() (Unknown Source:0)
ucrtbased.dll!00007ff8432b5443() (Unknown Source:0)
ucrtbased.dll!00007ff8432b6edd() (Unknown Source:0)
ucrtbased.dll!00007ff8432b5330() (Unknown Source:0)
ucrtbased.dll!00007ff8432a2e51() (Unknown Source:0)
ucrtbased.dll!00007ff8432a3003() (Unknown Source:0)
ucrtbased.dll!00007ff8432bab0d() (Unknown Source:0)
msvcp140d.dll!00007ff85c3b39e1() (Unknown Source:0)
msvcp140d.dll!00007ff85c3b36b2() (Unknown Source:0)
depthai-cored.dll!std::_Mutex_base::~_Mutex_base() Line 45 (c:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\mutex:45)
depthai-cored.dll!std::mutex::~mutex() (Unknown Source:0)
depthai-cored.dll!dai::Resources::~Resources() Line 373 (c:\njs\depthai-core\src\utility\Resources.cpp:373)
depthai-cored.dll!`dai::Resources::getInstance'::`2'::`dynamic atexit destructor for 'instance''() (Unknown Source:0)
ucrtbased.dll!00007ff8432c2931() (Unknown Source:0)
ucrtbased.dll!00007ff8432c2105() (Unknown Source:0)
ucrtbased.dll!00007ff8432c2257() (Unknown Source:0)
ucrtbased.dll!00007ff8432c2b44() (Unknown Source:0)
depthai-cored.dll!__scrt_dllmain_uninitialize_c() Line 399 (d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\utility\utility.cpp:399)
depthai-cored.dll!dllmain_crt_process_detach(const bool is_terminating) Line 182 (d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\dll_dllmain.cpp:182)
depthai-cored.dll!dllmain_crt_dispatch(HINSTANCE__ * const instance, const unsigned long reason, void * const reserved) Line 220 (d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\dll_dllmain.cpp:220)
depthai-cored.dll!dllmain_dispatch(HINSTANCE__ * const instance, const unsigned long reason, void * const reserved) Line 293 (d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\dll_dllmain.cpp:293)
depthai-cored.dll!_DllMainCRTStartup(HINSTANCE__ * const instance, const unsigned long reason, void * const reserved) Line 335 (d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\dll_dllmain.cpp:335)
ntdll.dll!00007ff88a0e9a1d() (Unknown Source:0)
ntdll.dll!00007ff88a12db91() (Unknown Source:0)
ntdll.dll!00007ff88a12da2d() (Unknown Source:0)
kernel32.dll!00007ff88861e0ab() (Unknown Source:0)
ucrtbased.dll!00007ff8432c1bba() (Unknown Source:0)
ucrtbased.dll!00007ff8432c1b65() (Unknown Source:0)
ucrtbased.dll!00007ff8432c1f06() (Unknown Source:0)
pipeline_test.exe!__scrt_common_main_seh() Line 297 (d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:297)
pipeline_test.exe!__scrt_common_main() Line 331 (d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:331)
pipeline_test.exe!mainCRTStartup(void * __formal) Line 17 (d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_main.cpp:17)
kernel32.dll!00007ff888617034() (Unknown Source:0)
ntdll.dll!00007ff88a122651() (Unknown Source:0)
```

Notice the `atexit()` handler is running. So this is program shutdown. The handler is destructing a `Resources` class.
https://github.com/luxonis/depthai-core/blob/0eaab54a7e3415ec135d315dbd33d931475c0097/src/utility/Resources.cpp#L366-L370

The first line has no issues. The second line has no issues. Is is something after that which has the mutex issues and hanging.
The `Resource` class does have two mutex..
I suspect this is a race condition. This test runs very fast. It never opens a connection to a device.
But...is still loads Resources. Perhaps the `Resource` is being created, is spins up two threads to read the firmware/resources. Maybe those threads are still working or have mutex locked in some manner that conflicts with a rapid program shutdown.

### Expected

All tests and examples to pass.

diablodale, 2022-01-13:
test 9 `serialization_test` fails with same mutex error. It is also a very fast test that never connects to a device.
All 62 other tests passed.

themarpe, 2022-01-13:
Hi @diablodale 
Can you test against this changes: `resources_lazy_load_tsan`
Was just resolving some issues yesterday and I think this way should be better.

diablodale, 2022-01-13:
Agreed, unlocking before the `notify_all()` is preferred: your closing the scope -or- calling `.unlock()`.
Your write two? Is the second in `Device::startPipelineImpl()`?

I tried your branch (with and without your above edit) and both tests now pass.
I'll start a full test pass with your scope edit above.

diablodale, 2022-01-13:
All 64 tests passed with your branch + EDIT above.
I also ran hundreds of the `serialization` and `pipeline` tests in a looped BAT file. All passed with no errant behavior.

themarpe, 2022-01-13:
> Your write two? Is the second in Device::startPipelineImpl()?

Sorry, had waiting part in mind, where there are two (`mtxDevice`, `mtxBootloader`).

Great, I'll fix and merge

diablodale, 2022-01-13:
It might be appropriate also in the that Device function. Worth trying since it is a callback that I think is repeatedly used.

diablodale, 2022-01-13:
I made similar unlocking change in `Device`. All tests passed, and fast loop BAT file also passed.
While there wasn't a known issue, this is likely a very minor performance improvement.

themarpe, 2022-01-14:
Addressed both in the PR - thanks for reporting:)

(not sure why the closing words aren't working..., I guess I have to merge via GH UI...)

closed 2022-01-14 by themarpe

diablodale, 2022-01-14:
Your code worked well, thanks! :-)

PR #328 (2022-01-13): Resources lazy load fixes
Close: #327 

Moves lazy loading finished notification to condition_variable+bool instead of just holding a mutex. Also improves some existing usage.