luxonis/depthai-core #330: feature request: add `const` to `dai::Buffer::getData()`

diablodale, 2022-01-14:
Request api adjustment to enable more code uses. Change `dai::Buffer::getData()` -> `dai::Buffer::getData() const`
Made in a separate PR (trivial I can make) or as part of the zerocopy buffer effort.

Code change needed is add the 6 chars to the signature declaration in Buffer.hpp and its signature in the implementation.
The change enables code like...

```c++
void myfunc(const dai::ImgFrame& frame) {
    const auto size = frame.getData().size();
    const auto ptr = frame.getData().data();
    ...
}
```

themarpe, 2022-01-14:
@diablodale 
I'd suggest opening a PR, separate, for this, to merge faster, without zerocopy - I'm not sure how soon I'll be able to get back to zerocopy and Messages refactor (probably next week, more likely week after that...)

diablodale, 2022-01-15:
forthcoming PR has above change and two additional changes

* add same const `ADatatype::getRaw() const` 
* add/clarify copy and move overloads for `Buffer::setData()`

diablodale, 2022-02-03:
verified the fix was merged into `develop` and all tests are passing using 04d5149e7cfddc4dc871d55d1a3d9ee270b76835