luxonis/depthai-core #334: incompatible BUILD_SHARED_LIBS + WINDOWS_EXPORT_ALL_SYMBOLS + INTERPROCEDURAL_OPTIMIZATION

diablodale, 2022-01-16:
On Windows, all Release builds fail when `BUILD_SHARED_LIBS=ON` and `CMAKE_INTERPROCEDURAL_OPTIMIZATION=ON`.

I think because CMakeLists.txt line 15 sets `CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS=ON`.
And that `...EXPORT_ALL_SYMBOLS=ON` feature of cmake is incompatible with `CMAKE_INTERPROCEDURAL_OPTIMIZATION=ON`.

When both are `ON` and `BUILD_SHARED_LIBS=ON` then the `exports.def file` created by the `CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS` feature is empty. With no def file, nothing is exported from the DLL and the linker fails.

Because depthai-core uses `CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS` instead of exporting via Windows standard approachs (e.g __declspec), it appears (from what I know so far) that the `INTERPROCEDURAL_OPTIMIZATION=ON` setting can not be used on Windows with the MSVC toolchain. FYI, `INTERPROCEDURAL_OPTIMIZATION=ON` cmake setting enables link-time code generation on MSVC with `/LTCG`.

### Setup

* Microsoft Windows [Version 10.0.19044.1466]
* VS2019 v16.11.9
* depthai-core `develop` 3d5dc2ee77b78ace7c20c49ec1ad2b13545aa4e6

### Repro

1. set `BUILD_SHARED_LIBS=ON`, `CMAKE_INTERPROCEDURAL_OPTIMIZATION=ON`, and `DEPTHAI_ENABLE_BACKWARD=ON`
2. cmake config for Release, x64, shared lib
3. cmake build

### Result

```
[build] [87/88  98% :: 26.555] C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\cl.exe   /TP -DDEPTHAI_BOOTLOADER_VERSION=\"0.0.15\" -DDEPTHAI_DEVICE_VERSION=\"2d72ac3bbbec5227f59873ea0b14b890a4fbf9bb\" -DDEPTHAI_PATCH_ONLY_MODE -DDEPTHAI_RESOURCE_COMPILED_BINARIES -DDEPTHAI_TARGET_CORE -DJSON_DIAGNOSTICS=0 -DJSON_USE_IMPLICIT_CONVERSIONS=1 -DLIBARCHIVE_STATIC -DLZMA_API_STATIC -DSPDLOG_COMPILED_LIB -DXLINK_USE_MX_ID_NAME=ON -D__PC__ -Ddepthai_core_EXPORTS -I..\include -IC:\njs\depthai-shared\include -I..\shared\depthai-bootloader-shared\include -I..\include\depthai -I..\src -IC:\njs\depthai-shared\src -I..\shared\depthai-bootloader-shared\src -IC:\njs\XLink\cmake\..\include -IC:\njs\vcpkg\installed\x64-windows-static-md-v142-sdk10\include -IC:\njs\depthai-shared\3rdparty -I_cmrc\include -IC:\.hunter\_Base\9b4e732\07a9f17\0dd2504\Install\include -DWINVER=0x0A00 -D_WIN32_WINNT=0x0A00 -DNTDDI_VERSION=0x0A000005 /arch:AVX /Gw /QIntel-jcc-erratum /guard:cf /GS /w15240 -D_CRT_SECURE_NO_WARNINGS  /wd5030 /wd4267 /wd5030 /wd4267   /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG /GL /Zc:preprocessor -std:c++14 /showIncludes /FoCMakeFiles\depthai-core.dir\src\pipeline\node\NeuralNetwork.cpp.obj /FdCMakeFiles\depthai-core.dir\ /FS -c ..\src\pipeline\node\NeuralNetwork.cpp
[build] Microsoft (R) C/C++ Optimizing Compiler Version 19.29.30139 for x64
[build] Copyright (C) Microsoft Corporation.  All rights reserved.
[build] 
[build] [88/88 100% :: 26.604] cmd.exe /C "cmd.exe /C ""C:\Program Files\CMake\bin\cmake.exe" -E __create_def C:\njs\depthai-core\build\CMakeFiles\depthai-core.dir\.\exports.def C:\njs\depthai-core\build\CMakeFiles\depthai-core.dir\.\exports.def.objs && cd C:\njs\depthai-core\build" && "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_dll --intdir=CMakeFiles\depthai-core.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100203~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100203~1.0\x64\mt.exe --manifests  -- C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\link.exe  CMakeFiles\depthai-core.dir\C_\njs\depthai-shared\src\datatype\DatatypeEnum.cpp.obj CMakeFiles\depthai-core.dir\C_\njs\depthai-shared\src\utility\Checksum.cpp.obj CMakeFiles\depthai-core.dir\shared\depthai-bootloader-shared\src\SBR.c.obj CMakeFiles\depthai-core.dir\shared\depthai-bootloader-shared\src\Bootloader.cpp.obj CMakeFiles\depthai-core.dir\src\device\Device.cpp.obj CMakeFiles\depthai-core.dir\src\device\DeviceBase.cpp.obj CMakeFiles\depthai-core.dir\src\device\DeviceBootloader.cpp.obj CMakeFiles\depthai-core.dir\src\device\DataQueue.cpp.obj CMakeFiles\depthai-core.dir\src\device\CallbackHandler.cpp.obj CMakeFiles\depthai-core.dir\src\device\CalibrationHandler.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\Pipeline.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\AssetManager.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\Node.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\XLinkIn.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\XLinkOut.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\ColorCamera.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\MonoCamera.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\StereoDepth.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\NeuralNetwork.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\ImageManip.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\VideoEncoder.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\DetectionNetwork.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\Script.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\SpatialDetectionNetwork.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\SystemLogger.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\SpatialLocationCalculator.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\ObjectTracker.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\IMU.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\EdgeDetector.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\SPIIn.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\node\FeatureTracker.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\Buffer.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\ImgFrame.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\ImageManipConfig.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\CameraControl.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\NNData.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\ImgDetections.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\SpatialImgDetections.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\SystemInformation.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\StreamMessageParser.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\SpatialLocationCalculatorData.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\SpatialLocationCalculatorConfig.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\Tracklets.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\IMUData.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\StereoDepthConfig.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\EdgeDetectorConfig.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\TrackedFeatures.cpp.obj CMakeFiles\depthai-core.dir\src\pipeline\datatype\FeatureTrackerConfig.cpp.obj CMakeFiles\depthai-core.dir\src\utility\Initialization.cpp.obj CMakeFiles\depthai-core.dir\src\utility\Resources.cpp.obj CMakeFiles\depthai-core.dir\src\utility\Platform.cpp.obj CMakeFiles\depthai-core.dir\src\xlink\XLinkConnection.cpp.obj CMakeFiles\depthai-core.dir\src\xlink\XLinkStream.cpp.obj CMakeFiles\depthai-core.dir\src\openvino\OpenVINO.cpp.obj CMakeFiles\depthai-core.dir\src\openvino\BlobReader.cpp.obj CMakeFiles\depthai-core.dir\src\bspatch\bspatch.c.obj  /out:depthai-core.dll /implib:depthai-core.lib /pdb:depthai-core.pdb /dll /version:0.0 /machine:x64 /ignore:4099 /INCREMENTAL:NO  /INCREMENTAL:NO /LTCG /DEF:CMakeFiles\depthai-core.dir\.\exports.def -LIBPATH:C:\PROGRA~1\NVIDIA~2\CUDA\v10.1\lib\x64 depthai-resources.lib  XLink\XLink.lib  C:\.hunter\_Base\9b4e732\07a9f17\0dd2504\Install\lib\bz2.lib  C:\.hunter\_Base\9b4e732\07a9f17\0dd2504\Install\lib\archive_static.lib  C:\.hunter\_Base\9b4e732\07a9f17\0dd2504\Install\lib\spdlog.lib  C:\.hunter\_Base\9b4e732\07a9f17\0dd2504\Install\lib\zlib.lib  C:\.hunter\_Base\9b4e732\07a9f17\0dd2504\Install\lib\lzma.lib  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib  && cd ."
[build] FAILED: depthai-core.dll depthai-core.lib 
[build] cmd.exe /C "cmd.exe /C ""C:\Program Files\CMake\bin\cmake.exe" -E __create_def C:\njs\depthai-core\build\CMakeFiles\depthai-core.dir\.\exports.def C:\njs\depthai-core\build\CMakeFiles\depthai-core.dir\.\exports.def.objs && cd C:\njs\depthai-core\build" && "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_dll --intdir=CMakeFiles\depthai-core.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100203~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100203~1.0\x64\mt.exe --manifests  -- C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\link.exe  CMakeFiles\depthai-core.dir\C_\njs\depthai-s [... cut ...]
1. Shared builds (dll) set `CMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF` or `INTERPROCEDURAL_OPTIMIZATION=OFF`. This is the default.
2. Static builds do not have this problem. You can set the optimization ON or OFF.

### Related

https://vcpkg.readthedocs.io/en/latest/maintainers/maintainer-guide/#do-not-add-cmake_windows_export_all_symbols
https://gitlab.kitware.com/cmake/cmake/-/issues/23035
https://manpages.debian.org/bullseye-backports/cmake-data/cmake-properties.7.en.html#WINDOWS_EXPORT_ALL_SYMBOLS

diablodale, 2022-01-16:
I have a workaround. For any DLL that will export functions/classes, the `/GL` compile flag has to be disabled so that cmake's CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS feature can correctly scan the OBJ files. But the linker's `/LTCG` can still be enabled so that dependencies can be correctly linked into the DLL.

If LTCG isn't put on the linker flags, then when dependencies compiled GL+LTCG are linked without it, then the linker will run for seconds->minutes and then notice a dependency is GL+LTCG and discard the link in process and restart the link with LTCG. The linker will literally display that warning and the restart on console output.

The depthai-core project has three libraries which are affected: `depthai-core`, `depthai-opencv`, and `utility` from the examples subproject. Same cmake code can be used in all three locations by only changing the target's name. For example...

```cmake
# disable /GL and enable /LTCG for bug https://github.com/luxonis/depthai-core/issues/334
if(WIN32 AND MSVC AND CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS)
    get_target_property(_INTER_OPT ${TARGET_CORE_NAME} INTERPROCEDURAL_OPTIMIZATION)
    if(_INTER_OPT)
        message(STATUS "Workaround MSVC dll exports with INTERPROCEDURAL_OPTIMIZATION")
        set_target_properties(${TARGET_CORE_NAME} PROPERTIES INTERPROCEDURAL_OPTIMIZATION OFF)
        target_link_options(${TARGET_CORE_NAME} PRIVATE /LTCG)
    endif()
    unset(_INTER_OPT)
endif()
```

Using that approach, I am able to enable INTERPROCEDURAL_OPTIMIZATION for Windows shared lib (DLL) builds.
All tests and examples passed.

Is this something the depthai-core project wants to include? If yes, I can make a PR.

themarpe, 2022-01-17:
Thanks for reporting and for the proposed workaround.

Unfortunate that this is required... 

I think it could fit - can you try performing this snippet by looping through `${targets_to_export}`, so it can be tucked away and only written once?

diablodale, 2022-01-17:
Sure. I will explore that looping, macro/func, and/or separate include file in `cmake/`.

The cmake code that does this is https://github.com/Kitware/CMake/blob/master/Source/bindexplib.cxx. After reviewing it and then reading https://docs.microsoft.com/en-us/cpp/build/reference/gl-whole-program-optimization?view=msvc-170 and https://docs.microsoft.com/en-us/cpp/build/reference/symbols?view=msvc-170

> .obj files produced with /GL aren't usable by linker utilities such as EDITBIN and DUMPBIN.
> Only the /HEADERS DUMPBIN option is available for use on files produced with the /GL compiler option.

This setting combination may be impossible to WINDOWS_EXPORT_ALL_SYMBOLS scan for symbols since even MSVC's own tool dumpbin can not do it.

themarpe, 2022-01-17:
> This setting combination may be impossible to WINDOWS_EXPORT_ALL_SYMBOLS scan for symbols since even MSVC's own tool dumpbin can not do it.

The interprocedular (`/LTCG`?) and the whole program optimization flags?

> Sure. I will explore that looping, macro/func, and/or separate include file in cmake/.

Thanks.

diablodale, 2022-01-17:
I think `/GL` and WINDOWS_EXPORT_ALL_SYMBOLS will be forever incompatible until Microsoft themselves exposes a method to scan for symbols. The compiler only knows about a single compilation unit. From what I learned, `/GL` precompiles headers and a subset of data into these global-opt `obj` files. The `/GL` option sets a internal mark meaning "I was compiled with /GL".

When the MSVC linker runs and links in obj files it looks for the `/GL` flag. If not there, it compiles like a standard toolset without global optimization. If it finds the internal "/GL" mark then the linker runs the compiler again and gives it global knowledge. New obj code is generated and the linker links that into an executable. 

I set the `/ltcg` flag...because when the linker eventually finds the "I was compiled with /GL" mark but it wasn't given the "/LTCG" link option, it discards the link in progress and restarts with the global approach -- very slow. The warning is `MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command line to improve linker performance`.

The subset of devs that encounter this are those with the combination: Windows, MSVC, global opt, cmake, building a DLL from code that needs WINDOWS_EXPORT_ALL_SYMBOLS. It is that last qualifier that depthai-core catches this limitation. Other projects use a macro and declspec to declare which APIs will be available (exported) outside the DLL. e.g.
https://github.com/opencv/opencv/blob/17234f82d025e3bbfbf611089637e5aa2038e7b8/modules/core/include/opencv2/core/cvdef.h#L406
https://github.com/madler/zlib/blob/cacf7f1d4e3d44d871b605da3b647f07d718623f/zconf.h#L335

Overall, this is a minor annoy and I don't see any urgent need for depthai-core to stop using `WINDOWS_EXPORT_ALL_SYMBOLS`. Only to have a workaround so that building a DLL will work. ;-)

closed 2022-02-23 by themarpe