luxonis/depthai-core #336: Feature Request: more information for tracked features

arthurBricq, 2022-01-17:
Dear Luxonis Team, 

It would be great to have additional information when receiving the tracked features from the feature tracker: At the moment, there is only the pixel position and the feature's ID, [according to the doc](https://docs.luxonis.com/projects/api/en/latest/references/cpp/#_CPPv4N3dai14TrackedFeatureE).

I am especially thinking of :
- The **Harris Score** of the feature. Even if I can control the Harris Threshold of the Feature Maintainer, it could be useful for run-time post-processing of the extracted features. 
- The feature's **timestamps** (*I know I can extract it with the `passthroughInputImage` as workaround, but it would still be better to be able to access the timestamp immediately*). 

Thanks in advance for the help !

themarpe, 2022-01-17:
Hi @arthurBricq 

I'll let @szabi-luxonis comment more on your first point.

Regarding second point - we'll address this with upcoming changes that will include timestamps and sequence/id numbers to all messages.

szabi-luxonis, 2022-01-17:
Hello @arthurBricq.
I reenabled the metadata of `FeatureTracker`. (Background on that: had to disable due to slow serializer, added too much CPU overhead, we changed the serializer recently, so it's possible to enable it again).

You can quickly install the library:
```
python3 -m pip install -U --prefer-binary --user --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local "depthai==2.14.1.0.dev+547e56d74c9a1a25a7eb074d8e157e8720ed5a28"
```
After building the wheels:
https://github.com/luxonis/depthai-python/actions/runs/1709583293

Available extra metadata: 
```
    /**
     *  Feature age in frames
     */
    uint32_t age;
    /**
     *  Feature harris score
     */
    float harrisScore;
    /**
     *  Feature tracking error
     */
    float trackingError;
```

arthurBricq, 2022-01-18:
Great ! Thanks for the updates, really appreciated :+1:

closed 2022-01-18 by arthurBricq