luxonis/depthai-core #338: libnop bug calling std::accumulate() cascades errors up to main apps (via headers)

diablodale, 2022-01-17:
libnop is cascading errors up to main apps because of the libnop headers, and then the main app builds break.
Often the errors are `'=': conversion from 'size_t' to '_Ty', possible loss of data`

Due to three places in libnop calling
`std::accumulate(value.cbegin(), value.cend(), 0U ...` 
All three need change to
`std::accumulate(value.cbegin(), value.cend(), static_cast<size_t>(0),`
so that the type of the accumulation var is the same (or larger) than the things being accumulated.

### Setup

* Microsoft Windows [Version 10.0.19044.1466]
* VS2019 v16.11.9
* depthai-core v2.14.1 + thread/move fixes

### Repro

1. cmake config for x64, debug, shared lib
2. cmake build

### Result

```
[build] Microsoft (R) C/C++ Optimizing Compiler Version 19.29.30139 for x64
[build] Copyright (C) Microsoft Corporation.  All rights reserved.
[build] 
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\numeric(35): error C2220: the following warning is treated as an error
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/map.h(55): note: see reference to function template instantiation '_Ty std::accumulate<std::_Tree_const_iterator<std::_Tree_val<std::_Tree_simple_types<std::pair<const std::string,std::vector<int,std::allocator<int>>>>>>,unsigned int,nop::Encoding<std::map<std::string,std::vector<int,std::allocator<int>>,std::less<std::string>,std::allocator<std::pair<const std::string,std::vector<int,std::allocator<int>>>>>,void>::Size::<lambda_bf97cd4b6d0a085dbaf5378705417e16>>(const _InIt,const _InIt,_Ty,_Fn)' being compiled
[build]         with
[build]         [
[build]             _Ty=unsigned int,
[build]             _InIt=std::_Tree_const_iterator<std::_Tree_val<std::_Tree_simple_types<std::pair<const std::string,std::vector<int,std::allocator<int>>>>>>,
[build]             _Fn=nop::Encoding<std::map<std::string,std::vector<int,std::allocator<int>>,std::less<std::string>,std::allocator<std::pair<const std::string,std::vector<int,std::allocator<int>>>>>,void>::Size::<lambda_bf97cd4b6d0a085dbaf5378705417e16>
[build]         ]
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/map.h(47): note: while compiling class template member function 'size_t nop::Encoding<std::map<std::string,std::vector<int,std::allocator<int>>,std::less<std::string>,std::allocator<std::pair<const std::string,std::vector<int,std::allocator<int>>>>>,void>::Size(const std::map<std::string,std::vector<int,std::allocator<int>>,std::less<std::string>,std::allocator<std::pair<const std::string,std::vector<int,std::allocator<int>>>>> &)'
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/types/detail/member_pointer.h(50): note: see reference to function template instantiation 'size_t nop::Encoding<std::map<std::string,std::vector<int,std::allocator<int>>,std::less<std::string>,std::allocator<std::pair<const std::string,std::vector<int,std::allocator<int>>>>>,void>::Size(const std::map<std::string,std::vector<int,std::allocator<int>>,std::less<std::string>,std::allocator<std::pair<const std::string,std::vector<int,std::allocator<int>>>>> &)' being compiled
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/types/detail/member_pointer.h(50): note: see reference to class template instantiation 'nop::Encoding<std::map<std::string,std::vector<int,std::allocator<int>>,std::less<std::string>,std::allocator<std::pair<const std::string,std::vector<int,std::allocator<int>>>>>,void>' being compiled
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/types/detail/member_pointer.h(49): note: while compiling class template member function 'size_t nop::MemberPointer<std::map<std::string,std::vector<int,std::allocator<int>>,std::less<std::string>,std::allocator<std::pair<const std::string,std::vector<int,std::allocator<int>>>>> dai::DetectionNetworkProperties::* ,pointer-to-member(0x80),void *,0x0,void>::Size(const Class &)'
[build]         with
[build]         [
[build]             Class=dai::DetectionNetworkProperties
[build]         ]
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/members.h(87): note: see reference to function template instantiation 'size_t nop::MemberPointer<std::map<std::string,std::vector<int,std::allocator<int>>,std::less<std::string>,std::allocator<std::pair<const std::string,std::vector<int,std::allocator<int>>>>> dai::DetectionNetworkProperties::* ,pointer-to-member(0x80),void *,0x0,void>::Size(const Class &)' being compiled
[build]         with
[build]         [
[build]             Class=dai::DetectionNetworkProperties
[build]         ]
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/members.h(87): note: see reference to class template instantiation 'nop::MemberPointer<std::map<std::string,std::vector<int,std::allocator<int>>,std::less<std::string>,std::allocator<std::pair<const std::string,std::vector<int,std::allocator<int>>>>> dai::DetectionNetworkProperties::* ,pointer-to-member(0x80),void *,0x0,void>' being compiled
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/members.h(87): note: see reference to function template instantiation 'size_t nop::Encoding<T,void>::Size<11>(const T &,nop::Index<11>)' being compiled
[build]         with
[build]         [
[build]             T=dai::DetectionNetworkProperties
[build]         ]
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/members.h(87): note: see reference to function template instantiation 'size_t nop::Encoding<T,void>::Size<11>(const T &,nop::Index<11>)' being compiled
[build]         with
[build]         [
[build]             T=dai::DetectionNetworkProperties
[build]         ]
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/members.h(45): note: see reference to function template instantiation 'size_t nop::Encoding<T,void>::Size<12>(const T &,nop::Index<12>)' being compiled
[build]         with
[build]         [
[build]             T=dai::DetectionNetworkProperties
[build]         ]
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/members.h(44): note: see reference to function template instantiation 'size_t nop::Encoding<T,void>::Size<12>(const T &,nop::Index<12>)' being compiled
[build]         with
[build]         [
[build]             T=dai::DetectionNetworkProperties
[build]         ]
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/members.h(43): note: while compiling class template member function 'size_t nop::Encoding<T,void>::Size(const T &)'
[build]         with
[... 22 lines cut ...]
[build]             T=dai::DetectionNetworkProperties
[build]         ]
[build] C:\njs\depthai-core\.install\Debug\include\depthai-shared/utility/Serialization.hpp(105): note: see reference to function template instantiation 'nop::Status<void> nop::Serializer<dai::utility::VectorWriter>::Write<T>(const T &)' being compiled
[build]         with
[build]         [
[build]             T=dai::DetectionNetworkProperties
[build]         ]
[build] C:\njs\depthai-core\.install\Debug\include\depthai-shared/properties/Properties.hpp(18): note: see reference to function template instantiation 'bool dai::utility::serialize<Derived>(const T &,std::vector<uchar,std::allocator<uchar>> &)' being compiled
[build]         with
[build]         [
[build]             Derived=dai::DetectionNetworkProperties,
[build]             T=dai::DetectionNetworkProperties
[build]         ]
[build] C:\njs\depthai-core\.install\Debug\include\depthai-shared/properties/Properties.hpp(17): note: while compiling class template member function 'void dai::PropertiesSerializable<dai::NeuralNetworkProperties,dai::DetectionNetworkProperties>::serialize(std::vector<uchar,std::allocator<uchar>> &) const'
[build] C:\njs\depthai-core\.install\Debug\include\depthai-shared/properties/DetectionNetworkProperties.hpp(16): note: see reference to class template instantiation 'dai::PropertiesSerializable<dai::NeuralNetworkProperties,dai::DetectionNetworkProperties>' being compiled
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\numeric(35): warning C4267: '=': conversion from 'size_t' to '_Ty', possible loss of data
[build]         with
[build]         [
[build]             _Ty=unsigned int
[build]         ]
```

### Expected

No potential loss of data errors -> clean app build.

themarpe, 2022-01-17:
Thanks, will update the libnop library

diablodale, 2022-01-18:
Hi. For the future, your libnop fork doesn't have issues or a `develop` branch.
Should we post issues here like I did?
And what branch should we target for PRs?

themarpe, 2022-01-18:
Added issues and `develop` branch.
Also opened https://github.com/luxonis/libnop/pull/1 which addresses libnop related issues from `core`

You may post issues and PRs there now (develop) :)

diablodale, 2022-02-03:
verified libnop fix is being included in depthai-core `develop` 04d5149e7cfddc4dc871d55d1a3d9ee270b76835
I consider this issue resolved.

PR #344 (2022-01-20): Updated libnop library
Fixes: #338 #339
closed 2022-02-03 by diablodale