luxonis/depthai-core #339: minor libnop tweaks for compile-time eval and move exceptions

diablodale, 2022-01-17:
During compile with MSVC code analyzer, some libnop code changes are suggested

### Setup
* Microsoft Windows [Version 10.0.19044.1466]
* VS2019 v16.11.9
* depthai-core v2.14.1 + thread/move fixes

### Repro
1. Setup msvc code analyzer with `/analyze:plugin EspXEngine.dll` and env vars `esp.extensions=CppCoreCheck.dll` `esp.annotationbuildlevel=ignore`
2. cmake config for x64, debug, shared lib
3. cmake build

### Result

A move constructor can not throw, it should be marked as such.
```c++
//  nop\types\result.h line 202
constexpr Result(Result&& other) : error_{other.error_} { other.clear(); }
// change to
constexpr Result(Result&& other) noexcept : error_{other.error_} { other.clear(); }

// same result.h line 212
constexpr Result& operator=(Result&& other) {
// change to
constexpr Result& operator=(Result&& other) noexcept {
```

The const variable 'element_size' can be computed at compile-time. Use constexpr
```c++
// nop\utility\buffer_reader.h line 60
const std::size_t element_size = sizeof(T);
// change to
constexpr std::size_t element_size = sizeof(T);
```

themarpe, 2022-01-17:
Thanks - will update libnop with proposed changes.

Also @diablodale, if you happen to know anyone who'd be willing to take a look at the libnop's macros and make an old MSVC preprocessor alternatives for it, I'd be happy to have a chat

diablodale, 2022-01-17:
I don't know anyone that might be passionate about that  ;-)

My approach on libnop was to change my projects and dependencies to be new preprocessor compliant. I knew them so well -- I could make the compliance changes rapidly.

themarpe, 2022-01-18:
> My approach on libnop was to change my projects and dependencies to be new preprocessor compliant. I knew them so well -- I could make the compliance changes rapidly.

Thats nice - luckily we have a large user base using Python bindings especially on Windows. This preprocessor change didn't cause any issues there as well.

Thanks anyways, will look around a bit more:)

diablodale, 2022-02-03:
verified suggested libnop changes were merged into libnop
verified depthai-code `develop` at 04d5149e7cfddc4dc871d55d1a3d9ee270b76835 is using those changed.
I consider this issue resolved.

closed 2022-02-03 by diablodale