luxonis/depthai-core #340: dai::PipelineImpl::create() uses raw new() -- not exception-safe

diablodale, 2022-01-17:
Found with MSVC code analyzer, `dai::PipelineImpl::create()` uses raw `new()` while creating a `shared_ptr`. In general raw `new()` should no longer be used. Also, using it during create of shared_ptr is not exception-safe.

Easy fix. Use `make_shared()`

### Setup

* Microsoft Windows [Version 10.0.19044.1466]
* VS2019 v16.11.9
* depthai-core v2.14.1 + thread/move fixes

### Repro

1. Setup msvc code analyzer with `/analyze:plugin EspXEngine.dll` and env vars `esp.extensions=CppCoreCheck.dll` `esp.annotationbuildlevel=ignore`
2. cmake config for x64, debug, shared lib
3. cmake build

### Result

```
[build] C:\njs\depthai-core\.install\Debug\include\depthai\pipeline\Pipeline.hpp(83) : warning C26409: Avoid calling new and delete explicitly, use std::make_unique<T> instead (r.11).
```

### Expected

Clean build

### Fix

https://github.com/luxonis/depthai-core/blob/988442242061362ee0e40851723e88ac45858da1/include/depthai/pipeline/Pipeline.hpp#L82-L83

change to 
```c++
auto node = std::make_shared<N>(itself, id);
```

themarpe, 2022-01-17:
AFAIK was modified as such as shared_ptr didn't have access to some parts.. will recheck and modify if possible or add the friend if clean

diablodale, 2022-01-17:
ok, though suspicious. 🧐

The `make_shared` change above works and passes tests+examples using MSVC. If you reproduce a problem, would be interesting to understand the cause as that would be some STL or compiler incompatibility.

diablodale, 2022-02-03:
verified fix appears in `develop` 04d5149e7cfddc4dc871d55d1a3d9ee270b76835 and it passes all tests.

closed 2022-02-03 by diablodale