luxonis/depthai-core #343: `StereoDepth::setDefaultProfilePreset()` probably uses wrong API for LR check

diablodale, 2022-01-18:
From what I can discern, `StereoDepth::setDefaultProfilePreset()` uses the wrong api.
As seen below it uses `setLeftRightCheck()`. But that method takes a boolean. Not an int.
Should it be instead `setLeftRightCheckThreshold()`

https://github.com/luxonis/depthai-core/blob/c49cd08cb1cad7d526ca2c2a2b561f0dcfa7020d/src/pipeline/node/StereoDepth.cpp#L177-L183

### Setup

* Everything
* depthai-core `main`

### Repro

Found during code review.

Luxonis-Brandon, 2022-01-18:
Thanks @diablodale !  Much appreciate it as usual.  CC: @szabi-luxonis

szabi-luxonis, 2022-01-18:
Thanks.
Fixed: https://github.com/luxonis/depthai-core/commit/5931f9600a7551ea796d51db7cd9caa183a5dfc3

closed 2022-01-18 by szabi-luxonis