luxonis/depthai-core #346: Lack file 【depthai-bootloader-shared...】 and 【depthai-shared】

berlin2049, 2022-01-22:
dowanload it and copy to  folder【depthai-core\shared】 can solve it.
【depthai-bootloader-shared...】：https://github.com/luxonis/depthai-bootloader-shared

szabi-luxonis, 2022-01-22:
> dowanload it and copy to folder【depthai-core\shared】 can solve it.
> 
> 【depthai-core\shared】：https://github.com/luxonis/depthai-shared 【depthai-bootloader-shared...】：https://github.com/luxonis/depthai-bootloader-shared

You need to run `git submodule update --init --recursive` after cloning or clone with `git clone git@github.com:luxonis/depthai-core.git --recursive`

berlin2049, 2022-01-23:
> You need to run `git submodule update --init --recursive` after cloning or clone with `git clone git@github.com:luxonis/depthai-core.git --recursive`

these two command take to much time in the region i'm at

closed 2022-01-23 by berlin2049

szabi-luxonis, 2022-01-23:
> 

You can use `git clone git@github.com:luxonis/depthai-core.git --recursive --depth 1`, it will download the latest.
Total downloaded size is less than 1MB.