luxonis/depthai-core #348: XLinkPlatformConnect - AddressSanitizer: heap-buffer-overflow (OakD POE)

jayv, 2022-01-23:
Just getting started with the OakD POE camera using the example from https://docs.luxonis.com/en/latest/pages/tutorials/getting-started-with-poe/#manually-specify-device-ip

However when I tried compiling/linking my project with `-g -fsanitize=undefined,address -fno-omit-frame-pointer` on gcc 9.3 (Ubuntu20.04) I'm getting this ERROR. 
Version v2.14.1 (same on older versions), and using pre-built binaries (`find_package(depthai CONFIG REQUIRED)`).

Did I miss anything?

```
=================================================================
==1284==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x602000012f3d at pc 0x7fbb4586fb35 bp 0x7ffcbb0fc1d0 sp 0x7ffcbb0fb978
READ of size 14 at 0x602000012f3d thread T0
    #0 0x7fbb4586fb34  (/lib/x86_64-linux-gnu/libasan.so.5+0xd1b34)
    #1 0x5604107de049 in XLinkPlatformConnect (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/modules/driver/tests/rr-sensors-oak-driver-test+0x82c049)
    #2 0x5604107d6cde in XLinkConnect (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/modules/driver/tests/rr-sensors-oak-driver-test+0x824cde)
    #3 0x5604107cb697 in dai::XLinkConnection::initDevice(dai::DeviceInfo const&, XLinkDeviceState_t) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/modules/driver/tests/rr-sensors-oak-driver-test+0x819697)
    #4 0x5604107cbeaf in dai::XLinkConnection::XLinkConnection(dai::DeviceInfo const&, XLinkDeviceState_t) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/modules/driver/tests/rr-sensors-oak-driver-test+0x819eaf)
    #5 0x56041074d044 in dai::DeviceBootloader::init(bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, tl::optional<dai::bootloader::Type>, bool) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/modules/driver/tests/rr-sensors-oak-driver-test+0x79b044)
    #6 0x56041074dad0 in dai::DeviceBootloader::DeviceBootloader(dai::DeviceInfo const&, bool) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/modules/driver/tests/rr-sensors-oak-driver-test+0x79bad0)
    #7 0x56041071b0c9 in dai::DeviceBase::init2(dai::DeviceBase::Config, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, tl::optional<dai::Pipeline const&>) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/modules/driver/tests/rr-sensors-oak-driver-test+0x7690c9)
    #8 0x56041071c848 in dai::DeviceBase::DeviceBase(dai::OpenVINO::Version, dai::DeviceInfo const&, bool) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/modules/driver/tests/rr-sensors-oak-driver-test+0x76a848)
    #9 0x56041070c9c2 in dai::Device::Device(dai::Pipeline const&, dai::DeviceInfo const&) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/modules/driver/tests/rr-sensors-oak-driver-test+0x75a9c2)
```

themarpe, 2022-01-23:
@jayv thanks for the report
We are currently fixing various issues of the underlying XLink library WRT multithreading & memory management...

Would you be able to provide a debug log with the issue when compiled in Debug mode? Also, try using latest `develop` branch as there were already some improvements to the XLink added there.

We'll circle back as we introduce improvements in this regard.

jayv, 2022-01-23:
@themarpe with `DEPTHAI_LEVEL=debug` this is what we get:
```
[2022-01-23 20:51:59.213] [debug] Device - OpenVINO version: 2021.4
[2022-01-23 20:51:59.324] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.15.tar.xz' open: 4ms, archive read: 108ms
[2022-01-23 20:51:59.729] [debug] Resources - Archive 'depthai-device-fwp-f503b7adfd44b696187b7825a12a134ded0ca3f3.tar.xz' open: 3ms, archive read: 513ms
=================================================================
==13381==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x602000012e9d at pc 0x7f900610eb35 bp 0x7ffec1ef6050 sp 0x7ffec1ef57f8
READ of size 14 at 0x602000012e9d thread T0
    #0 0x7f900610eb34  (/lib/x86_64-linux-gnu/libasan.so.5+0xd1b34)
    #1 0x564818407ce9 in XLinkPlatformConnect (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xcf9ce9)
    #2 0x56481840097e in XLinkConnect (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xcf297e)
    #3 0x5648183f5337 in dai::XLinkConnection::initDevice(dai::DeviceInfo const&, XLinkDeviceState_t) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xce7337)
    #4 0x5648183f5b4f in dai::XLinkConnection::XLinkConnection(dai::DeviceInfo const&, XLinkDeviceState_t) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xce7b4f)
    #5 0x5648183769c4 in dai::DeviceBootloader::init(bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, tl::optional<dai::bootloader::Type>, bool) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xc689c4)
    #6 0x564818377450 in dai::DeviceBootloader::DeviceBootloader(dai::DeviceInfo const&, bool) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xc69450)
    #7 0x564818344879 in dai::DeviceBase::init2(dai::DeviceBase::Config, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, tl::optional<dai::Pipeline const&>) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xc36879)
    #8 0x564818345ff8 in dai::DeviceBase::DeviceBase(dai::OpenVINO::Version, dai::DeviceInfo const&, bool) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xc37ff8)
    #9 0x564818336102 in dai::Device::Device(dai::Pipeline const&, dai::DeviceInfo const&) (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xc28102)
```

themarpe, 2022-01-23:
@jayv 
I meant the same sanitizer error log but of Debug compiled depthai library (so paths and locations of this issues is also printed along)

Great, let us know

jayv, 2022-01-24:
```
=================================================================
==1282==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x602000012e9d at pc 0x7ff43de4fb35 bp 0x7ffff8325d10 sp 0x7ffff83254b8
READ of size 14 at 0x602000012e9d thread T0
    #0 0x7ff43de4fb34  (/lib/x86_64-linux-gnu/libasan.so.5+0xd1b34)
    #1 0x55f3fe44f4d9 in XLinkPlatformConnect (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0x258f4d9)
    #2 0x55f3fe44816e in XLinkConnect (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0x258816e)
    #3 0x55f3fe3cbd85 in dai::XLinkConnection::initDevice(dai::DeviceInfo const&, XLinkDeviceState_t) /usr/src/depthai-core-v2.14.1/src/xlink/XLinkConnection.cpp:330
    #4 0x55f3fe3c80b0 in dai::XLinkConnection::XLinkConnection(dai::DeviceInfo const&, XLinkDeviceState_t) /usr/src/depthai-core-v2.14.1/src/xlink/XLinkConnection.cpp:167
    #5 0x55f3fdfe90b5 in void __gnu_cxx::new_allocator<dai::XLinkConnection>::construct<dai::XLinkConnection, dai::DeviceInfo&, XLinkDeviceState_t>(dai::XLinkConnection*, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/ext/new_allocator.h:147
    #6 0x55f3fdfe3a2c in void std::allocator_traits<std::allocator<dai::XLinkConnection> >::construct<dai::XLinkConnection, dai::DeviceInfo&, XLinkDeviceState_t>(std::allocator<dai::XLinkConnection>&, dai::XLinkConnection*, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/alloc_traits.h:484
    #7 0x55f3fdfda684 in std::_Sp_counted_ptr_inplace<dai::XLinkConnection, std::allocator<dai::XLinkConnection>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<dai::DeviceInfo&, XLinkDeviceState_t>(std::allocator<dai::XLinkConnection>, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr_base.h:548
    #8 0x55f3fdfc9e8f in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<dai::XLinkConnection, std::allocator<dai::XLinkConnection>, dai::DeviceInfo&, XLinkDeviceState_t>(dai::XLinkConnection*&, std::_Sp_alloc_shared_tag<std::allocator<dai::XLinkConnection> >, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr_base.h:679
    #9 0x55f3fdfad810 in std::__shared_ptr<dai::XLinkConnection, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<dai::XLinkConnection>, dai::DeviceInfo&, XLinkDeviceState_t>(std::_Sp_alloc_shared_tag<std::allocator<dai::XLinkConnection> >, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr_base.h:1344
    #10 0x55f3fdf8de96 in std::shared_ptr<dai::XLinkConnection>::shared_ptr<std::allocator<dai::XLinkConnection>, dai::DeviceInfo&, XLinkDeviceState_t>(std::_Sp_alloc_shared_tag<std::allocator<dai::XLinkConnection> >, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr.h:359
    #11 0x55f3fdf83e03 in std::shared_ptr<dai::XLinkConnection> std::allocate_shared<dai::XLinkConnection, std::allocator<dai::XLinkConnection>, dai::DeviceInfo&, XLinkDeviceState_t>(std::allocator<dai::XLinkConnection> const&, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr.h:702
    #12 0x55f3fdf73bdc in std::shared_ptr<dai::XLinkConnection> std::make_shared<dai::XLinkConnection, dai::DeviceInfo&, XLinkDeviceState_t>(dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr.h:718
    #13 0x55f3fdf4ef9d in dai::DeviceBootloader::init(bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, tl::optional<dai::bootloader::Type>, bool) /usr/src/depthai-core-v2.14.1/src/device/DeviceBootloader.cpp:258
    #14 0x55f3fdf4a4ef in dai::DeviceBootloader::DeviceBootloader(dai::DeviceInfo const&, bool) /usr/src/depthai-core-v2.14.1/src/device/DeviceBootloader.cpp:208
    #15 0x55f3fdd36518 in dai::DeviceBase::init2(dai::DeviceBase::Config, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, tl::optional<dai::Pipeline const&>) /usr/src/depthai-core-v2.14.1/src/device/DeviceBase.cpp:511
    #16 0x55f3fdd2f28b in dai::DeviceBase::init(dai::OpenVINO::Version, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/src/depthai-core-v2.14.1/src/device/DeviceBase.cpp:425
    #17 0x55f3fdd1c356 in dai::DeviceBase::DeviceBase(dai::OpenVINO::Version, dai::DeviceInfo const&, bool) /usr/src/depthai-core-v2.14.1/src/device/DeviceBase.cpp:233
    #18 0x55f3fdcd6eea in dai::Device::Device(dai::Pipeline const&, dai::DeviceInfo const&) /usr/src/depthai-core-v2.14.1/src/device/Device.cpp:46
```

jayv, 2022-01-24:
Same deal on build from `develop`:

```
=================================================================
==1245==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x602000012e9d at pc 0x7f646b537b35 bp 0x7ffd70eb7fd0 sp 0x7ffd70eb7778
READ of size 14 at 0x602000012e9d thread T0
    #0 0x7f646b537b34  (/lib/x86_64-linux-gnu/libasan.so.5+0xd1b34)
    #1 0x55ce62d68d09 in XLinkPlatformConnect (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xdecd09)
    #2 0x55ce62d6199e in XLinkConnect (/tmp/platform/sensors/oak/cmake-build-debug-dockerssh-oak-platform/application/rr-sensors-oak+0xde599e)
    #3 0x55ce62d43055 in dai::XLinkConnection::initDevice(dai::DeviceInfo const&, XLinkDeviceState_t) /usr/src/depthai-core/src/xlink/XLinkConnection.cpp:330
    #4 0x55ce62d421dd in dai::XLinkConnection::XLinkConnection(dai::DeviceInfo const&, XLinkDeviceState_t) /usr/src/depthai-core/src/xlink/XLinkConnection.cpp:167
    #5 0x55ce62c4f74e in void __gnu_cxx::new_allocator<dai::XLinkConnection>::construct<dai::XLinkConnection, dai::DeviceInfo&, XLinkDeviceState_t>(dai::XLinkConnection*, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/ext/new_allocator.h:147
    #6 0x55ce62c4e172 in void std::allocator_traits<std::allocator<dai::XLinkConnection> >::construct<dai::XLinkConnection, dai::DeviceInfo&, XLinkDeviceState_t>(std::allocator<dai::XLinkConnection>&, dai::XLinkConnection*, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/alloc_traits.h:484
    #7 0x55ce62c4bd66 in std::_Sp_counted_ptr_inplace<dai::XLinkConnection, std::allocator<dai::XLinkConnection>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<dai::DeviceInfo&, XLinkDeviceState_t>(std::allocator<dai::XLinkConnection>, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr_base.h:548
    #8 0x55ce62c484ac in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<dai::XLinkConnection, std::allocator<dai::XLinkConnection>, dai::DeviceInfo&, XLinkDeviceState_t>(dai::XLinkConnection*&, std::_Sp_alloc_shared_tag<std::allocator<dai::XLinkConnection> >, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr_base.h:679
    #9 0x55ce62c425df in std::__shared_ptr<dai::XLinkConnection, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<dai::XLinkConnection>, dai::DeviceInfo&, XLinkDeviceState_t>(std::_Sp_alloc_shared_tag<std::allocator<dai::XLinkConnection> >, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr_base.h:1344
    #10 0x55ce62c3b96e in std::shared_ptr<dai::XLinkConnection>::shared_ptr<std::allocator<dai::XLinkConnection>, dai::DeviceInfo&, XLinkDeviceState_t>(std::_Sp_alloc_shared_tag<std::allocator<dai::XLinkConnection> >, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr.h:359
    #11 0x55ce62c394ca in std::shared_ptr<dai::XLinkConnection> std::allocate_shared<dai::XLinkConnection, std::allocator<dai::XLinkConnection>, dai::DeviceInfo&, XLinkDeviceState_t>(std::allocator<dai::XLinkConnection> const&, dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr.h:702
    #12 0x55ce62c3517d in std::shared_ptr<dai::XLinkConnection> std::make_shared<dai::XLinkConnection, dai::DeviceInfo&, XLinkDeviceState_t>(dai::DeviceInfo&, XLinkDeviceState_t&&) /usr/include/c++/9/bits/shared_ptr.h:718
    #13 0x55ce62c2cddf in dai::DeviceBootloader::init(bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, tl::optional<dai::bootloader::Type>, bool) /usr/src/depthai-core/src/device/DeviceBootloader.cpp:258
    #14 0x55ce62c2bf43 in dai::DeviceBootloader::DeviceBootloader(dai::DeviceInfo const&, bool) /usr/src/depthai-core/src/device/DeviceBootloader.cpp:208
    #15 0x55ce62bab7c8 in dai::DeviceBase::init2(dai::DeviceBase::Config, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, tl::optional<dai::Pipeline const&>) /usr/src/depthai-core/src/device/DeviceBase.cpp:511
    #16 0x55ce62baa302 in dai::DeviceBase::init(dai::OpenVINO::Version, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/src/depthai-core/src/device/DeviceBase.cpp:425
    #17 0x55ce62ba7813 in dai::DeviceBase::DeviceBase(dai::OpenVINO::Version, dai::DeviceInfo const&, bool) /usr/src/depthai-core/src/device/DeviceBase.cpp:233
    #18 0x55ce62b97420 in dai::Device::Device(dai::Pipeline const&, dai::DeviceInfo const&) /usr/src/depthai-core/src/device/Device.cpp:46
    [... entrypoint ...]
```

jayv, 2022-01-24:
I can confirm ASAN is happy with the same code running against the `develop` branch.

closed 2022-01-24 by jayv