luxonis/depthai-core #349: macro error when using in external project

pierrextardif, 2022-01-24:
Hi,

I' trying to use depthai-core in a seperate c++ project, but it seems like the [serialize](https://github.com/luxonis/depthai-shared/blob/main/include/depthai-shared/utility/Serialization.hpp#L158) and seialize ext macros are not working for me.

I have tested it with the master branch on ubuntu 20.04, with manual build, as well as on windows 10.

I have then downloaded the release for windows10 (depthai-core-v2.14.1-win64 from the release section.), and the same error persists.

I have followed the steps to get depthai-core to work in a [non-cmake project](https://github.com/luxonis/depthai-core#non-cmake-integration-visual-studio-xcode-codeblocks-) ( in my case visual studio).

I have added the `.vcxproj` of my project [here](https://drive.google.com/file/d/13XDslAc1Kl-1zKSQobto2SSYqMsk-yg6/view?usp=sharing), and the whole project [here](https://drive.google.com/drive/folders/1sINHl27P8en_6QOnJIb9tc2O-AuzeqK5?usp=sharing).

In particular I have added the following for the Debug x64 (which is the one I am trying to work with):

```
<ItemDefinitionGroup Condition="'$(Configuration)|$(Platform)'=='Debug|x64'">
    <ClCompile>
      <PrecompiledHeader>Use</PrecompiledHeader>
      <WarningLevel>Level3</WarningLevel>
      <Optimization>Disabled</Optimization>
      <SDLCheck>true</SDLCheck>
      <PreprocessorDefinitions>_DEBUG;_CONSOLE;XLINK_USE_MX_ID_NAME=ON;__PC__=ON;%(PreprocessorDefinitions)</PreprocessorDefinitions>
      <ConformanceMode>true</ConformanceMode>
      <PrecompiledHeaderFile>pch.h</PrecompiledHeaderFile>
      <AdditionalIncludeDirectories>C:\Users\Jonathan\Documents\DEV\depthAI\depthai-core-v2.14.1-win64\include;C:\Users\Jonathan\Documents\DEV\depthAI\depthai-core-v2.14.1-win64\include\depthai-shared\3rdparty;C:\Users\Jonathan\Documents\DEV\depthAI\depthai-core-v2.14.1-win64\lib\cmake\depthai\dependencies\include;%(AdditionalIncludeDirectories)</AdditionalIncludeDirectories>
    </ClCompile>
    <Link>
      <SubSystem>Console</SubSystem>
      <GenerateDebugInformation>true</GenerateDebugInformation>
      <AdditionalLibraryDirectories>C:\Users\Jonathan\Documents\DEV\depthAI\depthai-core-v2.14.1-win64\lib;%(AdditionalLibraryDirectories)</AdditionalLibraryDirectories>
      <AdditionalDependencies>depthai-cored.lib;%(AdditionalDependencies)</AdditionalDependencies>
    </Link>
  </ItemDefinitionGroup>

```

The way I have created my new project is by creating a new C++ console project in Visual Studio, then modifying the properties of the solution (ALT+ENTER), and adding the above.

The error I'm getting is the following:

[image]

I feel like since this error is consistent on 2 platforms, it might be coming coming from your end ?
I can provide a project I have used last night on ubuntu, although on ubuntu, I'm using CMAKE.

Let me know if you can help with this, :)

Best,

P

themarpe, 2022-01-25:
Hi @pierrextardif 
Sorry for the inconvenience here, with CMake the required flag gets passed along automatically - you have to also set the `/Zc:preprocessor` flag to your solution. I'll update the readme accordingly.

Let me know if that works for you

pierrextardif, 2022-01-25:
Hi @themarpe ,

Ok, I've followed your advise and tried a `cmake` project on windows10, it worked!
So thank you.

However the method isn't what I'm used to, so will keep the issue opened and report further issues with using a visual studio solution / or close it within a few days if this was enough to get it sorted.

For further reference this what I've done to get it working on windows 10 with visual studio and CMake:
* 1) Open a new project in visual studio -> select `CMake project` ( which i called `depthAICmakeVS`)
* 2) Edited the `CMakeLists.txt` and added this (as described in the `readme` ):
```sh
find_package(depthai CONFIG REQUIRED)
target_link_libraries(depthAICmakeVS PRIVATE depthai::core)
``` 
* 3) Edited the `CMakeSettings.json` to point to the correct `depthai_DIR` :
`Project` -> `CMake Settings for depthAICmakeVS` -> `CMake Variables and Cache`  -> `depthai_DIR` -> `C:/Users/Jonathan/Documents/DEV/depthAI/depthai-core-v2.14.1-win64/lib/cmake/depthai` 

* 4) Added a `Pipeline` to the cpp:
```cpp
#include "depthai/depthai.hpp"

using namespace std;

int main()
{

	dai::Pipeline pipeline;
[...]
return 0;
}
```
et voila:

[image]

Thanks for your support so far.

pierrextardif, 2022-01-25:
No luck on linux using cmake so far:

Here is the modifications I have made to the `CMakelists.txt` of my project :

```sh

set(THREADS_PREFER_PTHREAD_FLAG ON)
find_package(Threads REQUIRED)
target_link_libraries(${APP_NAME} PRIVATE Threads::Threads)

find_package(depthai CONFIG REQUIRED)
target_link_libraries(${APP_NAME} PRIVATE depthai::opencv)

```
I have also passed every other linked library as `PUBLIC` , where `target_link_libraries` was used.

Here is the commad I have used:

```sh
cmake .. -D'depthai_DIR=/home/pierre/Documents/DEV/depthai-core/build/install/lib/cmake/depthai' >> cmakeLog.log 
```
cmakeLog.log:
[cmakeLog.log](https://github.com/luxonis/depthai-core/files/7937305/cmakeLog.log)
Then :
```sh
make -j12 &> makeLog.log
```
makeLog.log:
[makeLog.log](https://github.com/luxonis/depthai-core/files/7937316/makeLog.log)

I must detail I'm using Openframeworks with [ofxCMake](https://github.com/bahorn/ofxCMake) (this fork in particular). 

Let me know if you see something missing in my logs.
I can also try without using any framework with depthai-core, and just have plain C/C++ code to see how that works, :) .

themarpe, 2022-01-25:
@pierrextardif 
Maybe check the flags that you are using to compile your app - should be at least C++14
(If you linked as above, those should be set automatically though - unless overriden?)
(`make -j1 --trace` or `VERBOSE=1 make -j1` - Note: huge prints, but will display all compilation settings)

pierrextardif, 2022-01-25:
ok found the line :

```sh
/usr/bin/c++  \
-DDEPTHAI_TARGET_CORE \
-DDEPTHAI_TARGET_OPENCV \
-DJSON_USE_IMPLICIT_CONVERSIONS=1 \
-DXLINK_USE_MX_ID_NAME=ON \
-D__PC__ \
-I/usr/include/cairo \
-I/usr/include/glib-2.0 \
-I/usr/lib/x86_64-linux-gnu/glib-2.0/include \
-I/usr/include/pixman-1 \
-I/usr/include/uuid \
-I/usr/include/freetype2 \
-I/usr/include/libpng16 \
-I/usr/include/gstreamer-1.0 \
-I/usr/include/orc-0.4 \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/3d \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/app \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/communication \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/events \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/gl \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/graphics \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/math \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/sound \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/types \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/utils \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/openFrameworks/video \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/freetype/include \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/freetype/include/freetype2 \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/fmod/include \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/FreeImage/include \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/kiss/include \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/tess2/include \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/glfw/include \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/utf8/include \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/json/include \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/glm/include \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/pugixml/include \
-I/home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/libs/uriparser/include \
-isystem /usr/local/include/opencv4 \
-isystem /home/pierre/Documents/DEV/depthai-core/build/install/include \
-isystem /home/pierre/Documents/DEV/depthai-core/build/install/include/depthai-shared/3rdparty \
-isystem /home/pierre/Documents/DEV/depthai-core/build/install/lib/cmake/depthai/dependencies/include  \
-pthread -std=c++14 \
-o CMakeFiles/myOfApp.dir/src/main.cpp.o \
-c /home/pierre/Documents/OF/of_v0.11.2_linux64gcc6_release/apps/myApps/cmakeExample/src/main.cpp
```
is causing the issue, but :

```sh
/usr/bin/c++ \
-DDEPTHAI_TARGET_CORE \
-DDEPTHAI_TARGET_OPENCV \
-DJSON_USE_IMPLICIT_CONVERSIONS=1 \
-DXLINK_USE_MX_ID_NAME=ON \
-D__PC__ \
-isystem /usr/local/include/opencv4 \
-isystem /home/pierre/Documents/DEV/depthai-core/build/install/include \
-isystem /home/pierre/Documents/DEV/depthai-core/build/install/include/depthai-shared/3rdparty \
-isystem /home/pierre/Documents/DEV/depthai-core/build/install/lib/cmake/depthai/dependencies/include  \
-MD -MT CMakeFiles/CppTest.dir/main.cpp.o \
-MF CMakeFiles/CppTest.dir/main.cpp.o.d \
-o CMakeFiles/CppTest.dir/main.cpp.o \
-c /home/pierre/Documents/DEV/cppTest/src/main.cpp
```
works fine ( from CMake test without Openframeworks).
Seems to me that the `-MD -MT CMakeFiles/CppTest.dir/main.cpp.o` and `-MF CMakeFiles/CppTest.dir/main.cpp.o.d` are the problem?

closed 2022-01-26 by pierrextardif

themarpe, 2022-01-26:
> Seems to me that the imports are the problem?

Might be that some includes bring in some macros which are clashing...
You can check by taking the line which compiles eg. main.cpp and adding "preprocess only" flag `-E` to it - then feel free to paste the file here (it'll be huge, but should tell us what is going on).
(most likely the macros aren't clashing as that would output a warning, but there might be a macro redifining some keyword / name which messes things up...)

To try solving (workarounding) this easily try including `#include <depthai/depthai.hpp>` before anything else (I see that its currently included in 2nd line of `main.cpp` and 4th line of `ofApp.h`, maybe add it to the top of the `ofApp.h`)
Reason why it might work is that it won't clash with any overriden things by macros as preprocessor will do a pass through the source code and only later redefine (if thats the case). 

Let me know how that goes

pierrextardif, 2022-01-26:
> Let me know how that goes

Yeah i sorted it with your tip, :).
I've progressively added more `include`, and found out that the error was coming from `ofMain.h` importing `ofJson.h`, importing `json.hpp`, which version was 3.6.
I changed the content of `json.hpp` to match the depthai-core one and it worked ( hopefully that didn't break anything).
I will use the [apothecary](https://github.com/openframeworks/apothecary) next to rebuild OF with it in a more systematic way.

themarpe, 2022-01-26:
Great!

We might be able to hide json & libnop dependencies, but we use them in one public interface...
One thing to do is to add a check for version, and make sure its higher than the minimum supported. 
Both in CMake `cmake/depthaiDependencies.cmake:55` `find_package(nlohmann_json ...` and in the `Serialization.hpp`

themarpe, 2022-01-26:
@pierrextardif 
Created a PR that will address this - when used from CMake it will try finding the package with the specified version and when building from non-CMake case it will assert and stop compilation with a user friendly error message if there is a nlohmann json library clash (as in your case)

PR #351 (2022-01-26): Specify minimum nlohmann version
Shared PR: https://github.com/luxonis/depthai-shared/pull/80

Specifies minimum supported nlohmann json version for both CMake and non-CMake usage (in shared PR)

Fixes: #349 #265