luxonis/depthai-core #355: `CalibrationHandler` missing `const` on member functions - cascades to apps

diablodale, 2022-01-29:
Throughout `CalibrationHandler` the member functions are not declared `const` which prevents apps from using const calibration handlers. This cascades up to app classes containing a CalibrationHandler to be non-const, and further up the chain. Or cascades sideways to force app member functions to be non-const so they can call these handler functions. It metastasizes.

The hope of related https://github.com/luxonis/depthai-core/issues/10 didn't happen. 😢

### Setup

* all

### Repro

Found while writing my app. And then `depthai-core` code review.
Here are four examples of errant functions in CalibrationHandler... there are substantially more.

```c++
dai::CameraBoardSocket CalibrationHandler::getStereoRightCameraId();
float CalibrationHandler::getBaselineDistance(...);
std::vector<std::vector<float>> CalibrationHandler::getCameraIntrinsics(...);
float CalibrationHandler::getFov(...);
...
```

### Fix

Often just adding `const` to the end of the member function signatures. There maybe a small subset which need minor changes. This is less than a day's work.

### Workaround

None pretty. It might include making things `mutable` and removing `const` from member functions from apps, which then makes the app's code harder to maintain. :-/

None pretty. `mutable` doesn't help because

chengguizi, 2022-10-18:
This would be very helpful! Personally encountered this issue as well.

themarpe, 2022-10-18:
Addressed in latest `develop`:)

commit fe71aa24 (2022-10-18): Closes: #355
commit 0dc56eeb (2023-04-03): Ros release 2.21.0 (#795)

* Several FW updates:
- !!! For all sockets/cameras to be scanned, the EEPROM calib data boardName/productName should contain "FFC" or an FFC-related board name: 1090/2090/2088. Otherwise the behavior is to scan only for standard OAK devices sockets/cameras
- Some more checks for the LCM48 (on MAX) supported resolutions
- Fixes for RGB-mono SW-sync for standard OAK-D / OAK-D-Lite camera configs
- AR0234 monochrome support, however needs an OV9282 enabled at the same time for now

* Color/MonoCamera: handle some more cases for newly added resolutions/sockets

* Update [... cut]
commit 8839575b (2023-04-06): Ros release 2.21.2 (#798)

* Closes: #355

* Bump version to 2.18.0

* Updated FW with some Color/MonoCamera fixes

* stability test

* Update FW:
- fix for high system load crash, especially on PoE devices
- for for OV9282 as MonoCamera on RGB/CAM_A socket

* FW: OV9282 720p/800p max FPS: 60 -> 120

* stability test

* Added a convinience setTimesync function

* Added isUserBootloaderSupported to DeviceBootloader

* Updated XLink with Windows improvements when scanning for already booted USB devices

* Updated FW with device time reset on boot

* Bump to v2.19.0

* Improved wording on unavail [... cut]
closed 2022-10-18 by themarpe