luxonis/depthai-core #360: DepthAlign(RGB) breaks running pipeline if send StereoDepthConfig.setLeftRightCheckThreshold() or setConfidenceThreshold()

diablodale, 2022-01-31:
During a running Pipeline, sending `StereoDepthConfig` containing `setLeftRightCheckThreshold()` or `setConfidenceThreshold()` breaks `StereoDepth::depth` output when previously `StereoDepth::setDepthAlign(dai::CameraBoardSocket::RGB)`

When either of those settings are sent to a running pipeline, depthai-core sends warnings to the console and the depth data is forever errant until the pipeline is stopped and recreated.

The depth data is errant when viewed in a render of it as a 3D pointcloud -- hard to describe but it is definitely errant. The warnings are... 
```
[14442C10C1A1C6D200] [572.045] [StereoDepth(5)] [warning] Aligning disparity/depth frame to RGB perspective requires CENTER aligment! Defaulting to CENTER aligment.
```

### Setup

* Microsoft Windows [Version 10.0.19044.1469]
* VS2019 v16.11.9
* depthai-core `main` + xlink move/thread fixes

### Repro

1. Write an app that creates a pointcloud by combining `colorCamera::video`, `stereoDepth::depth`, `stereoDepth::setLeftRightCheck(true)`, and `stereoDepth::setDepthAlign(dai::CameraBoardSocket::RGB)`
2. Setup a xlink IN and attach it to `StereoDepth::inputConfig`
3. Start your Pipeline and render the aligned depth and color data as voxels in your favorite 3D rendering platform.
4. Confirm your 3D render is correct and as expected, e.g.
   [image]
5. Send a `setConfidenceThreshold()` or `setLeftRightCheckThreshold()` config change to the queue for the `StereoDepth::inputConfig` from step 2. e.g.
   ```c++
   dai::StereoDepthConfig depthConfig;
   depthConfig.setConfidenceThreshold(200);
   depthConfigQueue->send(depthConfig);
   ```

### Result

The 3D render is no longer correct. It looks ?quantized? and something more is wrong.
[image]

Warnings appear in the console...

```
[14442C10C1A1C6D200] [572.045] [StereoDepth(5)] [warning] Aligning disparity/depth frame to RGB perspective requires CENTER aligment! Defaulting to CENTER aligment.
```

### Expected

Good and expected 3D render. No warnings.

### Workaround and Notes

A workaround is to never set StereoDepth confidence or LRCheck threshold while the pipeline is running. Only set it as initialConfig before starting the pipeline.

I do not get these warnings if I use `StereoDepth::setDepthAlign(dai::StereoDepthProperties::DepthAlign::RECTIFIED_RIGHT);`. But that won't work for pointcloud use cases because DepthAI does not support aligning color camera to a depth perspective.

szabi-luxonis, 2022-01-31:
@diablodale 

You need to modify the `initialConfig` configuration.
`dai::StereoDepthConfig depthConfig;` is a separate instance, with default values : `right` alignment, subpixel off etc.

Something like:
```
auto initialConfig = stereo->initialConfig.get();
dai::StereoDepthConfig depthConfig(initialConfig);
depthConfig.setConfidenceThreshold(200);
depthConfigQueue->send(depthConfig);
```

A python example for full control:
https://github.com/luxonis/depthai-python/blob/9623e6ad8f74227d6bffbcd03b894c873edf316f/examples/StereoDepth/stereo_depth_from_host.py#L425

diablodale, 2022-01-31:
Ok, I'll try that.
How does that relate to `examples\ColorCamera\rgb_camera_control.cpp`?
In that example I see `ImageManipConfig` and `CameraControl` created with no relation to any initial settings.

diablodale, 2022-01-31:
Improvement, but not fully fixed. In my code I...
```
// all config and settings above, then...
depthConfigBaseline.set(stereoDepth->initialConfig.get());
// linking, pipeline startup, get queues, etc.
depthConfigBaseline.setConfidenceThreshold(200);
depthConfigQueue->send(depthConfigBaseline);
```

That still generates the warnings
```
E: [global] [   [14442C10C1A1C6D200] [46.961] [StereoDepth(5)] [warning] Aligning disparity/depth frame to RGB perspective requires CENTER aligment! Defaulting to CENTER aligment.
[14442C10C1A1C6D200] [64.237] [StereoDepth(5)] [warning] Aligning disparity/depth frame to RGB perspective requires CENTER aligment! Defaulting to CENTER aligment.
```

The pointcloud visually appears very similar before and after the warning. Perhaps it is only the change due to the new setting sent to the device.

Unclear on how many of these xxxConfig, xxxControl, etc. classes need to be get() and persisted so they work as expected and not reset such settings/config.

Unclear how the need to persist (or not) those classes as it is inconsistent in the python and c++ examples.

szabi-luxonis, 2022-01-31:
> rgb_camera_control.cpp

Setting runtime config message to initial, then modifying only what is really changed is just for convenience, as it has a lot more config options than ImageManip, which usually sets only 1 config.

You can avoid setting to `initialConfig`, but then you have to re-set all the previously set configs. e.g.:

```
dai::StereoDepthConfig depthConfig;
depthConfig.setLeftRightCheck(true);
depthConfig.setLeftRightCheckThreshold(x);
depthConfig.setDepthAlign(dai::StereoDepthConfig::AlgorithmControl::DepthAlign::CENTER);
depthConfig.setConfidenceThreshold(200);
....
depthConfigQueue->send(depthConfig);
```

diablodale, 2022-01-31:
I don't understand what you are trying to convey directly above. It doesn't clarify any of:

1. still get warnings. This seems a bug.
2. do all xxxConfig and xxxControl classes need to be either: a) initialConfig.get() and persisted  -or-  b) new instance and remember the initial settings somehow and duplicate them all again in the new instance
3. Are the examples like `rgb_camera_control` misleading or incorrect because they don't do (a) or (b) from 2?

szabi-luxonis, 2022-01-31:
> 1. still get warnings. This seems a bug.

The warning appears because of device side logic.
If `stereoDepth.setDepthAlign(dai::CameraBoardSocket::RGB)` is set then RGB alignment requires `CENTER` alignment of disparity/depth map. By default this is not set on host, but on device. I agree that we should change for transparency, since it's implementation detail, but might not work with multi cam support.

```
void StereoDepth::setDepthAlign(CameraBoardSocket camera) {
    properties.depthAlignCamera = camera;
    **if(camera == RGB) then setDepthAlign(CENTER)**
}
```

**Workaround:**
Set `stereo.setDepthAlign(dai::StereoDepthConfig::AlgorithmControl::DepthAlign::CENTER);`
when creating pipeline, you will get rid of warnings.

> 2. do all xxxConfig and xxxControl classes need to be either: a) initialConfig.get() and persisted  -or-  b) new instance and remember the initial settings somehow and duplicate them all again in the new instance

No, `CameraControl` is different, it applies config options separately, only what is sent based on an internal mask, on-device each command is sent separately to a component that handles 3A algorithms.

> 3. Are the examples like `rgb_camera_control` misleading or incorrect because they don't do (a) or (b) from 2?
No. Different approach. ImageManip doesn't have to know the previous config, the logic is handled on host, e.g. setting crop zone with `wasd` keys. The `initialConig` doesn't matter for the current one.

szabi-luxonis, 2022-02-01:
Stereo has 40+ configurable settings, taking the same approach as `CameraControl` would be unfeasible, having 40 setters+getters for each config or wrapping all in `std::optional` ... If you have better suggestion I will listen to it.
`CameraControl` is an exception, sending the same config every time would slow down the device (internal implementation detail) so it uses a mask for each command that is set by setters.
`Stereo` doesn't have this limitation, nor `ImageManip`, only `CameraControl` has a different behavior, all other nodes that support runtime config have the same behavior from this point of view. `EdgeDetector`, `StereoDepth`, `ImageManip`, `AprilTag`, `FeatureTracker` except `CameraControl`

When we add host-side nodes (meta nodes, combining multiple nodes in one for convenience) then we will be able to hide this logic behind it.

diablodale, 2022-02-01:
Good news. Your suggestion works to eliminate the warning if code calls BOTH functions.
```
// approach 1 -- works 🙂
// make both these calls during initial setup before the pipeline is started
// pointcloud looks ok and I get no warnings
stereoDepth->setDepthAlign(dai::StereoDepthConfig::AlgorithmControl::DepthAlign::CENTER);
stereoDepth->setDepthAlign(dai::CameraBoardSocket::RGB);

// approach 2 -- works with warnings 😐
// pointcloud looks ok but I get warnings on runtime config changes
stereoDepth->setDepthAlign(dai::CameraBoardSocket::RGB);

// approach 3 -- fails 😟
// depth data is errant
stereoDepth->setDepthAlign(dai::StereoDepthConfig::AlgorithmControl::DepthAlign::CENTER);
```

Below I've summarized my understanding. Did I get it right? 🧐

🏫 depthai has 6 messages to send runtime config/control changes to the device
* `CameraControl` only changes those settings explicitly set with the `setXXX()` member functions on a given class instance. Settings not set using those functions do not change state on the device. Therefore it is safe to create new instances and only set what is needed. You will not lose any previous settings.
* `EdgeDetectorConfig`, `FeatureTracker`, `ImageManipConfig`, `SpatialLocationCalculatorConfig`, and `StereoDepthConfig` change the device's state to match those settings stored in that class instance sent with `.send(instance)`. If the app needs to change only one setting, then it must also set *on that instance* all other settings to the previously known state. Otherwise, those settings not set *on that instance* will be reset to the depthai defaults. Persisting an `xxxxConfig` instance with the initial state before starting the Pipeline is convenient. Later runtime changes can then be applied to that instance and it be sent to the device.
* `AprilTag` ? 

This kind of info would be good in the docs as a tutorial as it applies widely. For weeks I have been silently losing my stereodepth settings for filters, conf, LR, etc. as I send() new changes for one setting. It was only today when I aligned and pointcloud projected and saw the prob and warning that this surfaced.

szabi-luxonis, 2022-02-01:
Yes, that's right. 
(`AprilTag` is a new addition, on develop, not main).
There is also an option to get `outConfig`, which outputs the config associated with the output frame.

Thanks for the feedback about improving docs.

diablodale, 2022-02-01:
Workaround is holding so far 😉

There is a design/bug issue above "we should change for transparency, since it's implementation detail, but might not work".
Leave this issue open for that? Or I can close with the workaround.

szabi-luxonis, 2022-02-02:
> Workaround is holding so far wink
> 
> There is a design/bug issue above "we should change for transparency, since it's implementation detail, but might not work". Leave this issue open for that? Or I can close with the workaround.

Instance number for disparity/depth frames, warnings are addressed on https://github.com/luxonis/depthai-core/commit/bb69f2c81cf38685d1ea2a6ca48886a7d6f56a93

diablodale, 2022-02-03:
good news. with depthai-core develop 04d5149e7cfddc4dc871d55d1a3d9ee270b76835...

* I do not get the errant warnings of "Approach 2" above https://github.com/luxonis/depthai-core/issues/360#issuecomment-1026374422
* now Approaches 1 and 2 both work as expected

I consider the code bug surfaced in this issue resolved. 👍
This issue can serve as documentation until the web/code docs are enhanced.

What remains is your comment of a design/bug issue above "we should change for transparency, since it's implementation detail, but might not work". Please do what you prefer: leave this issue open for the design, or close this resolved and track the design issue elsewhere, or...

closed 2022-02-04 by szabi-luxonis