luxonis/depthai-core #364: [system] [critical] Fatal error. Please report to developers. Log: 'OsDrvCpr' '1181'

diablodale, 2022-02-03:
While running ctest on the test+example suite, the `examples/feature_detector.exe` failed with the error

```
16: [14442C10C1A1C6D200] [3.620] [system] [critical] Fatal error. Please report to developers. Log: 'OsDrvCpr' '1181'
```

### Setup

* Microsoft Windows [Version 10.0.19044.1469]
* VS2019 v16.11.9
* depthai-core `develop` bb69f2c81cf38685d1ea2a6ca48886a7d6f56a93
* OAK-D early production (BW1098OBC)

### Repro

1. cmake config windows, 64, shared lib, all examples and tests
2. cmake build debug
3. run in a forever loop `ctest -j1 -C Debug -T test --output-on-failure -VV --stop-on-failure`

### Result

Runs one or two full ctest loops, then...

```
test 16
      Start 16: feature_detector

16: Test command: "C:\Program Files\CMake\bin\cmake.exe" "-DTIMEOUT_SECONDS=10" "-P" "C:/njs/depthai-core/examples/cmake/ExecuteTestTimeout.cmake" "C:/njs/depthai-core/build/examples/feature_detector.exe"
16: Environment variables:
16:  UBSAN_OPTIONS=halt_on_error=1
16: Test timeout computed to be: 1500
16: -- found -P
16: -- arguments: C:/njs/depthai-core/build/examples/feature_detector.exe;
16: [14442C10C1A1C6D200] [3.620] [system] [critical] Fatal error. Please report to developers. Log: 'OsDrvCpr' '1181'
16: Press 's' to switch between Harris and Shi-Thomasi corner detector!
16: Stack trace (most recent call last):
16: #8    Object "", at 00007FFC5D4A7034, in BaseThreadInitThunk
16: #7    Source "d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_main.cpp", line 17, in mainCRTStartup 
[00007FF61691219E]
16: #6    Source "d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 331, in __scrt_common_main [00007FF616911E6E]
16: #5    Source "d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 288, in __scrt_common_main_seh [00007FF616911FAE]
16: #4    Source "d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 79, in invoke_main [00007FF616912109]
16: #3    Source "C:\njs\depthai-core\examples\FeatureTracker\feature_detector.cpp", line 81, in main [00007FF616097D8F]
16:          78:     const auto rightWindowName = "right";16:          79:
16:          80:     while(true) {
16:       >  81:         auto inPassthroughFrameLeft = passthroughImageLeftQueue->get<dai::ImgFrame>();
16:          82:         cv::Mat passthroughFrameLeft = inPassthroughFrameLeft->getFrame();
16:          83:         cv::Mat leftFrame;
16:          84:         cv::cvtColor(passthroughFrameLeft, leftFrame, cv::COLOR_GRAY2BGR);
16: #2    Source "C:\njs\depthai-core\include\depthai\device\DataQueue.hpp", line 177, in dai::DataOutputQueue::get<dai::ImgFrame> [00007FF6160A26F1]
16:         174:         if(!queue.waitAndPop(val)) {     
16:         175:             throw std::runtime_error(exceptionMessage.c_str());
16:         176:         }
16:       > 177:         return std::dynamic_pointer_cast<T>(val);
16:         178:     }
16:         179:
16:         180:     /**
16: #1    Object "", at 00007FFC55ADB460, in CxxThrowException
16: #0    Object "", at 00007FFC5BD54F69, in RaiseException
16: -- After process executed,  produced the following exit code: -529697949
16: CMake Error at C:/njs/depthai-core/examples/cmake/ExecuteTestTimeout.cmake:42 (message):
16:    produced an error (-529697949) while running       
16:
16:
16/69 Test #16: feature_detector ....................***Failed  Error regular expression found in output. Regex=[\[critical\]]  9.68 sec
-- found -P
-- arguments: C:/njs/depthai-core/build/examples/feature_detector.exe;
[14442C10C1A1C6D200] [3.620] [system] [critical] Fatal error. Please report to developers. Log: 'OsDrvCpr' '1181'   
Press 's' to switch between Harris and Shi-Thomasi corner 
detector!
Stack trace (most recent call last):
#8    Object "", at 00007FFC5D4A7034, in BaseThreadInitThunk
#7    Source "d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_main.cpp", line 17, in mainCRTStartup [00007FF61691219E]
#6    Source "d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 331, in __scrt_common_main [00007FF616911E6E]
#5    Source "d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 288, in __scrt_common_main_seh [00007FF616911FAE]
#4    Source "d:\a01\_work\12\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 79, in invoke_main [00007FF616912109]
#3    Source "C:\njs\depthai-core\examples\FeatureTracker\feature_detector.cpp", line 81, in main [00007FF616097D8F]         78:     const auto rightWindowName = "right";    
         79:
         80:     while(true) {
      >  81:         auto inPassthroughFrameLeft = passthroughImageLeftQueue->get<dai::ImgFrame>();
         82:         cv::Mat passthroughFrameLeft = inPassthroughFrameLeft->getFrame();
         83:         cv::Mat leftFrame;
         84:         cv::cvtColor(passthroughFrameLeft, leftFrame, cv::COLOR_GRAY2BGR);
#2    Source "C:\njs\depthai-core\include\depthai\device\DataQueue.hpp", line 177, in dai::DataOutputQueue::get<dai::ImgFrame> [00007FF6160A26F1]
        174:         if(!queue.waitAndPop(val)) {
        175:             throw std::runtime_error(exceptionMessage.c_str());
        176:         }
      > 177:         return std::dynamic_pointer_cast<T>(val);
        178:     }
        179:
        180:     /**
#1    Object "", at 00007FFC55ADB460, in CxxThrowException#0    Object "", at 00007FFC5BD54F69, in RaiseException
-- After process executed,  produced the following exit code: -529697949
CMake Error at C:/njs/depthai-core/examples/cmake/ExecuteTestTimeout.cmake:42 (message):
   produced an error (-529697949) while running

94% tests passed, 1 tests failed out of 16

Total Test time (real) = 148.88 sec

The following tests FAILED:
         16 - feature_detector (Failed)
Errors while running CTest
```

### Expected

No errors and to loop forever until I explicitely stop it.

szabi-luxonis, 2022-02-03:
@diablodale can you try latest 04d5149e7cfddc4dc871d55d1a3d9ee270b76835 ?

diablodale, 2022-02-03:
Good news. No device fatal errors with new commit 04d5149e7cfddc4dc871d55d1a3d9ee270b76835
Note, I changed constexpr `DEFAULT_SEARCH_TIME` to 5 seconds due to issue #291.

I looped ctest for the full test+example suite for 2 hours.
Each run is about 700 seconds = 11.6 minutes --> 10 to 11 full runs of the suite
No fatal errors, all tests passed.

I also ran in a loop the one test `feature_detector.exe` for 10 minutes
Each run is ~10 seconds --> 60 runs of that single test
No fatal errors, test passed.

I consider this issue resolved 🙂

closed 2022-02-04 by szabi-luxonis