luxonis/depthai-core #371: 'lpb.ImgFrame' object has no attribute 'getTimestampDevice' `node::Script`

diablodale, 2022-02-09:
On device in script node python, need to get the device timestamp of the moment a device-side ImgFrame was created.
Unfortunately, creating a `node::Script` and giving it a python script with `rgbImgFrame.getTimestampDevice()` fails with error
```
[14442C10C1A1C6D200] [103.187] [Script(10)] [critical] AttributeError: 'lpb.ImgFrame' object has no attribute 'getTimestampDevice'
At:
  <script>(90): <module>
```

### Setup

* depthai-core `develop` at 04d5149e7cfddc4dc871d55d1a3d9ee270b76835
* Microsoft Windows [Version 10.0.19044.1526]
* VS2019 v16.11.9

### Repro

1. cmake config, build, install depthai-core as shared lib, x64, debug
2. Write your main c++ app with a `node::ColorCamera`, `node::NeuralNetwork`, `node::Script`
3. Link color camera -> NN input
4. Link NN passthrough -> Script
5. Make Script node python similar to
   ```python
   passrgb = node.io['from_passthrough'].get()
   devicetimestamp = passrgb.getTimestampDevice() // timedelta(microseconds=1)
   ```
6. Compile and run your main app

### Result

Console errors appear and the device-side python fails with error...

```
[14442C10C1A1C6D200] [103.187] [Script(10)] [critical] AttributeError: 'lpb.ImgFrame' object has no attribute 'getTimestampDevice'
At:
  <script>(90): <module>
```

### Expected

No console errors and python api imgframe.getTimestampDevice() works
https://docs.luxonis.com/projects/api/en/latest/components/messages/img_frame/

### Notes

The non-device timestamp works. Seems to be a missing API in the device-side python

```python
# works
othertimestamp = passrgb.getTimestamp() // timedelta(microseconds=1)
```

alex-luxonis, 2022-02-09:
Missed adding this to device side Python bindings back then when it was added on host. 
Will fix, thanks!

alex-luxonis, 2022-02-14:
@diablodale Actually I re-checked the FW implementation, and on device side `getTimestamp()` would return the device timestamp (what the host returns with `getTimestampDevice()`).

We store the ImgFrame timestamp in the device-side pipeline as captured from the device monotonic timer, and only before being serialized to the host, we do :
- `RawImgFrame::tsDevice` gets assigned the value as-is
- `RawImgFrame::ts` gets modified by adding the host-sync time diff (this time diff is also captured at the same time the image is captured from camera)

CC @themarpe on how should we get this more standardized, should we keep device `getTimestamp()` behavior as-is, and add a `getTimestampHost()`?
We also need to double-check if e.g Python's `time.monotonic()` on device side uses the same timer.

themarpe, 2022-02-14:
I think we should make these two equivalent as on host, by having `getTimestamp` return same time as on host would.
And adding `getTimestampDevice`.

We should also add a binding for `Clock.now` (eg: `std::steady_clock::now`) as on host.