luxonis/depthai-core #383: Cmake find_package don't work

dongxuanlb, 2022-02-16:
an@xavier:~/workspace/depthai-core$ cmake -S. -Bbuild -D'DEPTHAI_BUILD_EXAMPLES=ON' -D'BUILD_SHARED_LIBS=ON'
an@xavier:~/workspace/depthai-core$ cmake --build build --target install
an@xavier:~/workspace/depthai-core$ cmake --build build -j7

the build process is ok.

in my project cmakelist.txt:

# depthai
find_package(depthai CONFIG REQUIRED PATHS "/home/an/workspace/depthai-core/build/install/lib/cmake/depthai")
if(depthai_FOUND)
message(STATUS "depthai FOUND, Version:" ${depthai_VERSION} " directory: " ${depthai_DIR})
add_library(vision_oakd SHARED
src/input/oak/oakd.cpp
)
target_link_libraries(vision_oakd PRIVATE depthai::opencv)
endif()

when set PRIVATE, the error is :

‘
[build] In file included from /home/an/workspace/iRobot/vision/src/main.cpp:7:0:
[build] /home/an/workspace/iRobot/vision/src/input/oak/oakd.hpp:13:10: fatal error: depthai/depthai.hpp: No such file or directory
[build] #include <depthai/depthai.hpp>
[build] ^~~~~~~~~~~~~~~~~~~~~
‘
not set PRIVATE，but add my spdlog， seems confict：

‘
[build] CMakeFiles/visionApp.dir/src/main.cpp.o: In function void spdlog::info<char [3]>(char const (&) [3])': [build] /home/an/.hunter/_Base/9b4e732/08f9145/90457df/Install/include/spdlog/logger.h:168: undefined reference to spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view, spdlog::level::level_enum, fmt::v7::basic_string_view)'
[build] CMakeFiles/visionApp.dir/src/main.cpp.o: In function void spdlog::logger::log_<fmt::v7::basic_string_view<char>, unsigned long&>(spdlog::source_loc, spdlog::level::level_enum, fmt::v7::basic_string_view<char> const&, unsigned long&)': [build] /home/an/.hunter/_Base/9b4e732/08f9145/90457df/Install/include/spdlog/fmt/bundled/format.h:3821: undefined reference to void fmt::v7::detail::vformat_to(fmt::v7::detail::buffer&, fmt::v7::basic_string_view, fmt::v7::basic_format_args<fmt::v7::basic_format_context<fmt::v7::detail::buffer_appender<fmt::v7::type_identity::type>, fmt::v7::type_identity::type> >, fmt::v7::detail::locale_ref)'
[build] CMakeFiles/visionApp.dir/src/main.cpp.o: In function void spdlog::logger::log_<fmt::v7::basic_string_view<char>, unsigned long&>(spdlog::source_loc, spdlog::level::level_enum, fmt::v7::basic_string_view<char> const&, unsigned long&)': [build] /home/an/.hunter/_Base/9b4e732/08f9145/90457df/Install/include/spdlog/logger.h:341: undefined reference to spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view, spdlog::level::level_enum, fmt::v7::basic_string_view)'
[build] collect2: error: ld returned 1 exit status
‘

themarpe, 2022-02-16:
Hi @dongxuanlb 

The private error is okay - from your application, you won't be able to see depthai through `vision_oakd` library, as it links to it in private manner.

Conflict with spdlog I think comes from the fact that your project finds depthai's spdlog installation (under Hunter).
Can you try calling find_package(spdlog ...) before depthai, in your project CMakeLists.txt?

Also, please share the full (or atleast partially full) CMakeLists.txt file of your project

dongxuanlb, 2022-02-17:
> Hi @dongxuanlb
>
> The private error is okay - from your application, you won't be able to see depthai through `vision_oakd` library, as it links to it in private manner.
>
> Conflict with spdlog I think [...]

Thank you for your quick reply. for now, the easiest way to solve the error is change my spdlog version into 1.8.2 ^^.

here is my CMakeLists.txt

`cmake_minimum_required(VERSION 3.8)

project(iRobotVision)

add_definitions(-std=c++14)
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_FLAGS "-march=native")
set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g")
set(CMAKE_WARN_DEPRECATED OFF CACHE BOOL "" FORCE)

if (WIN32)
else()
    set(CMAKE_CXX_FLAGS "-fPIE -fPIC -D_FORTIFY_SOURCE=2 -fstack-protector -Wformat -Wformat-security -Wall ${CMAKE_CXX_FLAGS}")
endif()

set(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake)
set(CMAKE_LIBRARY_OUTPUT_DIRECTORY ${PROJECT_SOURCE_DIR}/lib)

\# cuda and tensorrt
find_package(CUDA 10.2 REQUIRED)
find_package(CUDNN REQUIRED)
include_directories(${CUDA_INCLUDE_DIRS} ${CUDNN_INCLUDE_DIR} ${NVINFER_INCLUDES})
\# opencv4 and eigen3
find_package(OpenCV REQUIRED)
find_package(Eigen3 REQUIRED)
include_directories(
  ${OpenCV_INCLUDE_DIRS}
  ${EIGEN3_INCLUDE_DIR}
)
\# vpi and visionworks
find_package(vpi 1.1 REQUIRED)
find_package(VisionWorks REQUIRED)
include_directories("/usr/share/visionworks/sources/nvxio/include")
\# log
find_package(spdlog REQUIRED)

include_directories(src)

...

\## vision core
add_library(vision_core SHARED
    src/core/parameters.cpp
    src/core/FrameStorage.cpp
    src/core/utils/print.cpp
)
target_link_libraries(vision_core ${LIBS} ${OpenCV_LIBS})

\## Env Pipeline
add_library(environment_pipeline SHARED
    src/proc/pipeline/EnvironmentPipeline.cpp
)
target_link_libraries(environment_pipeline 
    ${LIBS} ${OpenCV_LIBS} vision_core 
)

\## input start

\# realsense sdk
find_package(realsense2 2.49.0 REQUIRED)
if(NOT realsense2_FOUND)
    message(FATAL_ERROR "\n\n Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases\n\n")
else()
    add_library(vision_realsense SHARED
    src/input/realsense/d455.cpp
    )
    target_link_libraries(vision_realsense ${LIBS} ${OpenCV_LIBS} ${realsense2_LIBRARY} vision_core)
endif()

\# usb webcam by opencv
add_library(vision_usbcam SHARED
    src/input/webcam/OpencvUsbCam.cpp
)
target_link_libraries(vision_usbcam ${LIBS} ${OpenCV_LIBS} vision_core)

\# depthai
find_package(depthai CONFIG REQUIRED PATHS "/home/an/workspace/depthai-core/build/install/lib/cmake/depthai")
if(depthai_FOUND)
    message(STATUS "depthai FOUND, Version:" ${depthai_VERSION} " directory: " ${depthai_DIR})
    add_library(vision_oakd SHARED
        src/input/oak/oakd.cpp
    )
    target_link_libraries(vision_oakd depthai::opencv)
endif()

\## input end

add_executable(visionApp src/main.cpp)
target_link_libraries(visionApp 
PUBLIC
    vision_core 
    vision_realsense vision_usbcam vision_oakd
    environment_pipeline 
    track_perception
)
`

themarpe, 2022-02-17:
@dongxuanlb 
Thanks for sharing back how you've resolved the issue.

If you want to dig in deeper - can you print the path at which the `find_package(spdlog REQUIRED)` found the `spdlog` package? It seems like that will point to `~/.hunter/...` directory, but shouldn't... Do you have any additional CMake variables set like `IMPORT_PREFIX`, `spdlog_DIR`, ... ?

(Actually, it might resolve that through "registry" although unlikely.) Where is your installation of spdlog? Can you specify it manually using `-D spdlog_DIR=/path/to/you/spdlog/of/any/version`.
I think even having different spdlog version in you app should work as expected, as we hide away (visibility=hidden) our spdlog library we use internally.

dongxuanlb, 2022-02-17:
> @dongxuanlb Thanks for sharing back how you've resolved the issue.
>
> If you want to dig in deeper - can you print the path at which the `find_package(spdlog REQUIRED)` found the `spdlog` package? It [...]

find_package(spdlog REQUIRED)
message(STATUS "the spdlog path ${spdlog_DIR}")
The path is：
[cmake] -- the spdlog path /usr/local/lib/cmake/spdlog

`
an@xavier:/usr/local/include/spdlog$ cat version.h 
// Copyright(c) 2015-present, Gabi Melman & spdlog contributors.
// Distributed under the MIT License (http://opensource.org/licenses/MIT)

#pragma once

#define SPDLOG_VER_MAJOR 1
#define SPDLOG_VER_MINOR 8
#define SPDLOG_VER_PATCH 2

#define SPDLOG_VERSION (SPDLOG_VER_MAJOR * 10000 + SPDLOG_VER_MINOR * 100 + SPDLOG_VER_PATCH)
`

closed 2022-02-18 by dongxuanlb