luxonis/depthai-core #390: `DataOutputQueue::addCallback()` signature causes many copies, prevents mutable lambdas

diablodale, 2022-02-19:
The three `addCallback()` overloads do repeated copies inside and between themselves. There is no emplace, and no move semantics.

* Causes unneeded multiple copies of `std::function` itself and its data e.g. lambda captures.
* Depending on what is capture, this could be big data.
* Prevents mutable lambdas to be used as callbacks.

### Setup

* all
* depthai-core `develop`

### Repro

Found while writing app code to watch OAK sensor temperature. 

1. Write an app that uses depthai-core and wants to add a callback on an output queue. The callback will be a lambda that captures a filestream and writes to that file on each data the Queue delivers. The lambda must be `mutable` so the `std::ofstream` can be written into; otherwise the capture is a `const std::ofstream` which is unwritable.
Pseudocode...

```c++
std::ofstream fileStream;
fileStream.open("/mydir/myfile.csv", std::ios::out | std::ios::trunc);
mySysInfoQueue->addCallback([fileStream = std::move(fileStream)](std::shared_ptr<dai::ADatatype> data) mutable {
    const auto sysInfo = std::dynamic_pointer_cast<dai::SystemInformation>(data);
    const auto& t = sysInfo->chipTemperature;
    fileStream << now().time_since_epoch().count() << ", " << t.average << std::endl;
});
```

2. Attempt to compile

### Result

```
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\functional(896): error C2280: 'dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>::<lambda_37738337c0172f94b73042ce25286061>(const dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061> &)': attempting to reference a deleted function
[build] ..\dp.oak\hardware.cpp(675): note: see declaration of 'dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>::<lambda_37738337c0172f94b73042ce25286061>'
[build] ..\dp.oak\hardware.cpp(675): note: 'dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>::<lambda_37738337c0172f94b73042ce25286061>(const dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061> &)': function was implicitly deleted because a data member invokes a deleted or inaccessible function 'std::basic_ofstream<char,std::char_traits<char>>::basic_ofstream(const std::basic_ofstream<char,std::char_traits<char>> &)'
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\fstream(1076): note: 'std::basic_ofstream<char,std::char_traits<char>>::basic_ofstream(const std::basic_ofstream<char,std::char_traits<char>> &)': function was explicitly deleted
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\xmemory(272): note: see reference to function template instantiation 'std::_Func_impl_no_alloc<dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>,_Ret,std::shared_ptr<dai::ADatatype>>::_Func_impl_no_alloc<const _Callable&,0>(_Other)' being compiled
[build]         with
[build]         [
[build]             _Ret=void,
[build]             _Callable=dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>,
[build]             _Other=const dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061> &
[build]         ]
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\xmemory(283): note: see reference to function template instantiation 'std::_Func_impl_no_alloc<dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>,_Ret,std::shared_ptr<dai::ADatatype>>::_Func_impl_no_alloc<const _Callable&,0>(_Other)' being compiled
[build]         with
[build]         [
[build]             _Ret=void,
[build]             _Callable=dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>,
[build]             _Other=const dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061> &
[build]         ]
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\functional(901): note: see reference to function template instantiation '_Ty *std::_Global_new<std::_Func_impl_no_alloc<dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>,_Ret,std::shared_ptr<dai::ADatatype>>,const _Callable&>(const _Callable &)' being compiled
[build]         with
[build]         [
[build]             _Ty=std::_Func_impl_no_alloc<dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>,void,std::shared_ptr<dai::ADatatype>>,
[build]             _Ret=void,
[build]             _Callable=dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>
[build]         ]
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\functional(901): note: while compiling class template member function 'std::_Func_base<_Ret,std::shared_ptr<dai::ADatatype>> *std::_Func_impl_no_alloc<dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>,_Ret,std::shared_ptr<dai::ADatatype>>::_Copy(void *) const'
[build]         with
[build]         [
[build]             _Ret=void
[build]         ]
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\functional(813): note: see reference to class template instantiation 'std::_Func_impl_no_alloc<dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>,_Ret,std::shared_ptr<dai::ADatatype>>' being compiled
[build]         with
[build]         [
[build]             _Ret=void
[build]         ]
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\functional(1010): note: see reference to variable template 'const bool _Is_large<std::_Func_impl_no_alloc<<lambda_37738337c0172f94b73042ce25286061>,void,std::shared_ptr<dai::ADatatype> > >' being compiled
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\functional(1143): note: see reference to function template instantiation 'void std::_Func_class<_Ret,std::shared_ptr<dai::ADatatype>>::_Reset<dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>>(_Fx &&)' being compiled
[build]         with
[build]         [
[build]             _Ret=void,
[build]             _Fx=dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>
[build]         ]
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\functional(1143): note: see reference to function template instantiation 'void std::_Func_class<_Ret,std::shared_ptr<dai::ADatatype>>::_Reset<dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>>(_Fx &&)' being compiled
[build]         with
[build]         [
[build]             _Ret=void,
[build]             _Fx=dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>
[build]         ]
[build] ..\dp.oak\hardware.cpp(687): note: see reference to function template instantiation 'std::function<void (std::shared_ptr<dai::ADatatype>)>::function<dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>,0>(_Fx)' being compiled
[build]         with
[build]         [
[build]             _Fx=dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>
[build]         ]
[build] ..\dp.oak\hardware.cpp(675): note: see reference to function template instantiation 'std::function<void (std::shared_ptr<dai::ADatatype>)>::function<dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>,0>(_Fx)' being compiled
[build]         with
[build]         [
[build]             _Fx=dp::Sensor::Oak::startCameras::<lambda_37738337c0172f94b73042ce25286061>
[build]         ]
[build] ninja: build stopped: subcommand failed.
[build] Build finished with exit code 1
```

### Expected

clean build.

### Notes

I think the issue is the lack of move semantics and repeated copies. The first issue in the chain of issues is that the `addCallback(std::function<void(std::shared_ptr<ADatatype>)> callback)` overload attempts to make a copy of the `callback`. In that attempt, it would need to call the copy assign/constructor for `std::ofstream` which is naturally a deleted function. Can't copy streams.

This copying of a callback continues into the full signature `addCallback(std::function<void(std::string, std::shared_ptr<ADatatype>)> callback)` where it does yet another copy with `callbacks[id] = callback;`.

I think this can all be sorted with some move semantics. Likely small code changes and isolated to these three `addCallback()` overloads.

diablodale, 2022-02-20:
Fixed the copies using move semantics. Also learned that `std::function` requires the thing it wraps to be copy constructable, so my capture had to shift to a shared_ptr<ofstream>.
Code is passing all tests. I'll test with it a few days more then submit a PR.

themarpe, 2022-02-22:
Thanks @diablodale !

closed 2022-03-16 by themarpe