luxonis/depthai-core #392: Access to device performance counters

winterismute, 2022-02-20:
I was one of the backers of the OAK-D-Lite project, so I got one not long ago and I started playing with it - nice product btw!

One thing I was wondering about is whether you expose, via this API or in some other one, somehow getting data about the device performance counters, which seem to be exposed on the vino platform, at least looking at this document that I could quickly find: https://docs.openvino.ai/latest/openvino_docs_MO_DG_Getting_Performance_Numbers.html

I think it could give a clearer picture of the performance on what is running on the OAK, whether to change network etc.
I could not find this functionality in the depth-ai python SDK, but to be fair I did not look if it was exposed in this C++ package, apologies in that case.

themarpe, 2022-02-22:
Hi @winterismute 

Glad to hear you like it! 

We currently don't expose NN per layer performance data like OpenVINO does, mentioned above. 
Right now best way to get this data is using OpenVINO to analyze your network. The number of shaves might change, so thats something to take into account, but otherwise the results should transfer more or less (there might be some differences, between on scenarios).

In the future we'll most likely address diagnostics of message flow in the pipeline, fps of various queues, etc... 
But for NN per layer performance, nothing on our roadmap as of right now.

ORippler, 2022-08-01:
> We currently don't expose NN per layer performance data like OpenVINO does, mentioned above.
Right now best way to get this data is using OpenVINO to analyze your network. The number of shaves might change, so thats something to take into account, but otherwise the results should transfer more or less (there might be some differences, between on scenarios).

Could you provide some tips/guide for this?

If I understand you correctly, we should use
 1. [OpenVINO runtime](https://docs.openvino.ai/latest/openvino_docs_OV_UG_OV_Runtime_User_Guide.html), and
 2. run our model on the MyriadX via the [VPU Plugin](https://docs.openvino.ai/latest/openvino_docs_OV_UG_supported_plugins_VPU.html#doxid-openvino-docs-o-v-u-g-supported-plugins-v-p-u), and then
 3. use profiling functionality provided by openvino via [vtune](https://www.intel.com/content/www/us/en/developer/tools/oneapi/vtune-profiler.html).

Unfortunately, I am currently stuck at step 2, where I have to install the VPU plugin. I followed this guide, but as soon as I install [HDDL](https://github.com/openvinotoolkit/docker_ci/blob/master/install_guide_vpu_hddl.md) all `depthai`-based scripts fail to run (Claiming that the Device is already busy/in use), and hddl itself fails to discover the Myriadx when connecting the OAK-D via USB.

Cheers.

themarpe, 2022-08-02:
@ORippler 

Sure (CC: @tersekmatija on any additional comments) - with this case you'd run the device using OpenVINO framework instead and not using DepthAI - those two might clash using them at the same "run" time.

My thoughts on steps I'd take would be as following:
 - Install OpenVINO 2021.4 / 2022.1 by usual means (apt, instead of docker)
 - Skip HDDL, afaik it isn't necessary
 - Use the before mentioned benchmark app either C++ or Python with the specified `-pc` parameters https://docs.openvino.ai/latest/openvino_docs_MO_DG_Getting_Performance_Numbers.html#internal-inference-performance-counters-and-execution-graphs

But will also leave Matija to comment on this more

tersekmatija, 2022-08-03:
You can get per-layer benchmarks the easiest using `benchmark_app`. You can install `openvino-dev` 21.4 or 22.1 Python package instead of installing locally, and then call something like below with `-pc` flag as @themarpe  mentioned.
```
benchmark_app -m model.xml -d MYRIAD -pc
```

This should give you some statistics and information about execution.