luxonis/depthai-core #396: APIs allow easy change/corruption of `boardName` stored in EEPROM -- easily breaks apps

diablodale, 2022-02-21:
The EEPROM read/write apis, e.g. in the calibration.py script, can trivially delete, change, or corrupt the boardName stored in EEPROM.
This can result in unpredictable problems across applications which use boardName for identification, model classification, etc.

It is my view that the `eeprom.boardName` of an OAK device is permanent and can not be changed except for new firmware itself...therefore no one but Luxonis can change `boardName`.

### Setup

* all OS and depthai-core
* sensor OAK-D-Pro-PoE

### Repro

1. checkout depthai repro at `oak-d-pro` brance
2. Run calibration using the python script. Give it the board `OAK-D-PRO` as there is none for PoE. e.g. `python calibrate.py -s 3.95 -brd OAK-D-PRO -db`
3. Complete a successful calibration which will write calibration values to EEPROM.
4. Run some app which identifies the attached OAK sensor.

### Result

The OAK sensor is now an `OAK-D-PRO`. This breaks apps as the sensor identifies itself as one model, yet it is actually another model, causing unpredictable failures.

### Expected

The sensor OAK-D-Pro-PoE to always identify itself with `eeprom.boardName = "OAK-D-PRO-POE"`

### Notes

I can think of nothing good to allow such easy changing of core identification of a sensor. I recommend the depthai-core APIs remove the ability to set/change `boardName`. If some user somewhere in some super-uber-custom solution needs to change their boardname...they are a level of customer that can recompile depthai-core to add/enable an API to do so.

Luxonis-Brandon, 2022-04-18:
Meant to say thanks for this.  We've been working internally to address this.  @themarpe has the latest plan.

themarpe, 2022-07-08:
New devices now come with "factory" calibration which cannot be changed unless certain permissions are applied.
Current devices also have certain fields protected, which disallows changes unless certain permissions are specified.

The tooling surrounding flashing the EEPROM are now also mostly updated and should take that into account, to prevent from unintended modifications and work well with the protected fields and v7 eeprom changes

Thanks for opening the issue on this topic!

closed 2022-07-08 by themarpe