luxonis/depthai-core #400: invalid StereoDepth mesh files does not stop Device start; instead invalid config should stop start

diablodale, 2022-02-22:
If mesh files are invalid, the depthai-core ignores them and continues to startup the device/pipeline which will now serve invalid data.

My expectation is that when any (including mesh files) configuration is invalid, that the startup of the device/pipeline fails by throwing an exception.

### Setup

* depthai-core `multi_cam_support` at d9ffc562c7a9ab27ce6dccac6c9bef45edd601e7
* Microsoft Windows [Version 10.0.19044.1526]
* VS2019 v16.11.10

### Repro

* config, build, and install depthai-core above with: x64, debug, shared
* calibrate your OAK-D or OAK-D-PRO depth sensor so that you generate mesh files. 
* write an app that will digest StereoDepth.depth data at 400p resolution and calls `stereoDepth.loadMeshFiles("left.mesh", "right.mesh"). Yes...intentionally attempt to load mesh files with resolution not matching your StereoDepth resolution.
* Run your app

### Result

Your app runs and receives incorrect depth data. The data does not have mesh corrections.
If you collect logs, you will get a log entry
```
[14442C10C1A1C6D200] [38435.988] [StereoDepth(5)] [error] Mesh size 33048 not matching resolution 640x400 with step 16x16, expected 8528. Mesh won't be loaded
```

### Expected

An exception to be thrown, either at the call `stereoDepth.loadMeshFiles()` or the constructor `Device(pipelineConfig)` with exception.what() = "Mesh size 33048 not matching resolution 640x400 with step 16x16, expected 8528."

### Workaround

Messy and reliable. I could try to use `addLogCallback()`. Then on every construct Device() wait a few seconds 😬 and on another thread parse any log callbacks for the substring "Mesh won't be loaded" and then use a condition variable to signal back to my Device() construct thread to then check a variable if there was an error.

* race conditions and unknown amount of time to wait for enough log entries
* complex code
* substring matching on an error string which can change on any commit

alex-luxonis, 2022-03-09:
Thanks for the issue report.

We will change to a have an exception thrown here. It's evaluated on device when the pipeline is passed to, by:
```c++
Device(pipelineConfig);
```
or

```c++
d = Device(); 
...
d.startPipeline(pipelineConfig);
```

szabi-luxonis, 2022-04-20:
@diablodale Addressed in https://github.com/luxonis/depthai-core/commit/08be0a71b315c8fe4dbb58bf14bb0fe57e826e51

diablodale, 2022-04-23:
Unfortunately, no exception is thrown with v2.15.3

* No log warning or error. It is completely silent in v2.15.3. This is worse than previous.
* No exception is thrown
* The OAK device sends a stream of frames to the host with unknown mesh/rectification