luxonis/depthai-core #404: Is any example or api for get multi node frames synchronously?

dongxuanlb, 2022-02-23:
Now the API for get single MonoNode or ColorNode is using callback，but how two get three frame synchronously？

the realsense api is using frameset struct，it’s easy to synchronize access to all frames 。

now， I can only use three call and a blocking queue to synchronize manually，while I found they often cannot be strictly synchronized within a cycle（mono and rgb all in 30fps）。

themarpe, 2022-02-23:
Hi @dongxuanlb 
As of right now, we don't yet have any good synchronization mechanisms out of the box. You'd have to combine the frames using timestamp or sequence numbers by yourself. 

We are in the progress of adding such functionality, so doing so will be much easier and abstracted away.

There is an example with combining NN results and ColorCamera frames over at `examples/MobileNet/camera_mobilenet_sync_example.cpp` which might come handy

dongxuanlb, 2022-02-23:
> Hi @dongxuanlb As of right now, we don't yet have any good synchronization mechanisms out of the box. You'd have to combine the frames using timestamp or sequence numbers by yourself.
>
> We are in the [...]

Thanks.

And I found that if two 720p monocular +  one channel of 1080p NV12 are configed at the same time, the bandwidth seems to be insufficient and rgb will be lost. when the rgb down to 720p，will be no lost basically 。

themarpe, 2022-02-23:
> And I found that if two 720p monocular + one channel of 1080p NV12 are configed at the same time, the bandwidth seems to be insufficient and rgb will be lost. when the rgb down to 720p，will be no lost basically 。

What device do you use and at what speeds? PoE (100Mbit / 1Gbit) or USB (USB2 / USB3)?  The 1080p NV12 at 30FPS already consumes ~90MB/s which is too much for USB2 or 100Mbit speeds. Make sure you are running at adequate speeds for either device model.

dongxuanlb, 2022-02-24:
> 

I'm using oakd and the usbspeed is 3.0 （Super）.

themarpe, 2022-03-01:
@dongxuanlb in this case it might the the overhead of conversion + rendering - try commenting out the all processing on host and just retrieve the messages from the queue. Let me know what kind of results this creates.