luxonis/depthai-core #407: Windows app compiles fail in depthai headers on MSVC with c++20 and non-conformant preprocessor

diablodale, 2022-02-26:
WIndows app compiles fail in depthai-core headers on MSVC, c++20 standard, and the non-conformant preprocessor.
This is building a main app (e.g the tests)...not depthai-core itself. Depthai-core was built c++14 as usual.

when adjusting cmake to compile `tests/serialization_test` in c++14, 17, and 20...
14 and 17, with conformant and non-conformant preprocessor compiles
20 conformant preprocessor compiles
20 non-conformant preprocessor compile fails 👎

### Setup

* Microsoft Windows [Version 10.0.19044.1526]
* VS2019 v16.11.10
* depthai-core develop 3a2961de709faf44313d7af4f62314ad063acdec

### Repro

1. Adjust tests/CMakefile.txt so that you compile tests with 14, 17, and 20
2. Fix bug in tests/CMakefile.txt so it sets the preprocessor flags using `PRIVATE` (I recommend) or `PUBLIC` not the errant `INTERFACE`. e.g. `target_compile_options(${test_name}_conforming PRIVATE "/Zc:preprocessor")`
3. cmake config. Confirm (fix step 2) the preprocessor flags are on the compiles for conformant tests
4. cmake build

### Result

The compile of tests for C++20 non-conformant preprocessor compile fail with same errors. Here is attempt to build serialization_test...

```
[main] Building folder: depthai-core serialization_test_20
[build] Starting build
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.exe" --build c:/njs/depthai-core/build --config Debug --target serialization_test_20 -j 14 --
[build] [1/2  50% :: 5.209] Building CXX object tests\CMakeFiles\serialization_test_20.dir\src\serialization_test.cpp.obj
[build] FAILED: tests/CMakeFiles/serialization_test_20.dir/src/serialization_test.cpp.obj 
[build] C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\cl.exe  /nologo /TP -DDEPTHAI_TARGET_CORE -DJSON_USE_IMPLICIT_CONVERSIONS=1 -DXLINK_USE_MX_ID_NAME=ON -D__PC__ -Dserialization_test_20_EXPORTS -IC:\njs\vcpkg\installed\x64-windows-static-md-v142-sdk10\include -I..\include -I..\shared\depthai-shared\include -I..\shared\depthai-bootloader-shared\include -I..\shared\depthai-shared\3rdparty -IC:\.hunter\_Base\9b4e732\9daff0d\036e6a8\Install\include -DWINVER=0x0A00 -D_WIN32_WINNT=0x0A00 -DNTDDI_VERSION=0x0A000005 /arch:AVX /Gw /QIntel-jcc-erratum /guard:cf /GS  -D_CRT_SECURE_NO_WARNINGS /wd5030 /w15240 -D_CRT_SECURE_NO_WARNINGS /wd5030 /w15240   /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MDd /Zi /Ob0 /Od /RTC1 -DNOMINMAX -std:c++latest /showIncludes /Fotests\CMakeFiles\serialization_test_20.dir\src\serialization_test.cpp.obj /Fdtests\CMakeFiles\serialization_test_20.dir\ /FS -c ..\tests\src\serialization_test.cpp
[build] C:\.hunter\_Base\9b4e732\9daff0d\036e6a8\Install\include\nop/types/variant.h(294): error C2760: syntax error: unexpected token '...', expected ')'
[build] C:\.hunter\_Base\9b4e732\9daff0d\036e6a8\Install\include\nop/types/variant.h(292): note: This diagnostic occurred in the compiler generated function 'bool nop::IfAnyOf<ValidTypes...>::Call(nop::Variant<Types...> *,Op &&)'
[build] C:\.hunter\_Base\9b4e732\9daff0d\036e6a8\Install\include\nop/types/variant.h(351): note: see reference to class template instantiation 'nop::IfAnyOf<ValidTypes...>' being compiled
[build] C:\.hunter\_Base\9b4e732\9daff0d\036e6a8\Install\include\nop/types/variant.h(301): error C2760: syntax error: unexpected token '...', expected ')'
[build] C:\.hunter\_Base\9b4e732\9daff0d\036e6a8\Install\include\nop/types/variant.h(299): note: This diagnostic occurred in the compiler generated function 'bool nop::IfAnyOf<ValidTypes...>::Call(const nop::Variant<Types...> *,Op &&)'
[build] ninja: build stopped: subcommand failed.
[build] Build finished with exit code 1
```

### Expected

clean build, no errors, tests pass

diablodale, 2022-02-28:
I don't. The headers are used by depthai-core and by consuming apps. I believe the issue has to be investigated in those shared headers. C++20 seems to be off-limits for now.

themarpe, 2022-04-11:
Fixed in #448 @Bercon @diablodale

closed 2022-04-11 by themarpe