luxonis/depthai-core #411: Build depthai-core without hunter

lepalom, 2022-03-02:
I'm trying to build depthai-core but not using hunter. My idea it to package it (or try) for debian/ubuntu. 

I have already a version of depthai-core built without hunter and using its dependencies that are already packaged in debian (bzip2, fp16, spdlog, libarchive, zlib, libnop). However, I have several doubts that I would kindly ask because I don't understand.

 * In the libnop library I have seen that the upstream version and the version you use are practically the same. The only difference is that the library you use EncodingByte::Empty and the library in upstream has EncodingByte::Nil. I have changed it and it built. But, is it ok?
 * the cmake download a lot of files, especially blob files for the examples. Which are the license of that files? Can be redistributed?
 * Is it possible that you release a version with all the necessary components? At least the git submodules.

I have [read](https://github.com/luxonis/depthai-core/issues/72#issuecomment-929100355) that you haven't yet added any package manager support outside of Hunter. It's a pity, but I can understand it. Maybe you can help if you can give more information about the exact requirements to build depthai suite.

themarpe, 2022-03-02:
@lepalom 

> I'm trying to build depthai-core but not using hunter. My idea it to package it (or try) for debian/ubuntu.

I think you don't have to remove the Hunter to be able to do so. After you build the depthai-core and install it, there is no dependency on Hunter anymore. This could easily be packed, etc...

> In the libnop library I have seen that the upstream version and the version you use are practically the same. The only difference is that the library you use EncodingByte::Empty and the library in upstream has EncodingByte::Nil. I have changed it and it built. But, is it ok?

Use our fork (https://github.com/luxonis/libnop) instead of upstream version. Main intention for the change was that it doesn't cause issues with Objective-C.

> the cmake download a lot of files, especially blob files for the examples. Which are the license of that files? Can be redistributed?

AFAIK are all models used there from https://github.com/openvinotoolkit/open_model_zoo, whose license equals the one specified there after converting to blob (doesn't change the license I assume) - I'd say yes. IANAL. 

> Is it possible that you release a version with all the necessary components? At least the git submodules.

Already done, check Releases, eg: `depthai-core-v2.15.0.tar.gz`

> I have https://github.com/luxonis/depthai-core/issues/72#issuecomment-929100355 that you haven't yet added any package manager support outside of Hunter. It's a pity, but I can understand it. Maybe you can help if you can give more information about the exact requirements to build depthai suite.

Its mainly missing the specs on how to build from perspective of that package manager. In relation to the mentioned issue, #72 , that core issue should be fixed in latest release. 

One should be able to prepare a vcpkg "port", for DepthAI, by default, it might be easy to do without removing Hunter and just relying on the packaged `dependencies` that DepthAI brings with Hunter, but these might clash with vcpkg dependencies. One step further is for private dependencies to still continue using vcpkg dependencies, and one other option is to disable hunter and provide all dependencies by vcpkg (or manually).
 
If you wish, I'd be happy to guide on efforts of adding such "port" to vcpkg or others.

lepalom, 2022-03-02:
> I think you don't have to remove the Hunter to be able to do so. After you build the depthai-core and install it, there is no dependency on Hunter anymore. This could easily be packed, etc...

Well, I think that the first thing that I have to do is to build depthai-core, and AFAIK it needs Hunter. I'm in this first step.

> Use our fork (https://github.com/luxonis/libnop) instead of upstream version. Main intention for the change was that it doesn't cause issues with Objective-C.

I cannot. As I put my Debian Maintainer hat, I have to follow the policies and it is packaged in [Debian](https://packages.debian.org/search?suite=bullseye&searchon=names&keywords=libnop). If there are some issues in Objective-C, the maintainer should solve it. However, I have seen that now there are no [differences between your fork and the original one](https://github.com/google/libnop/compare/master...luxonis:master).

About the license, perfect. I will check what to do. 

And for depthai-device-fwp ? Do you have a release files?

The main problem is the dependencies and how to obtain it, but they will not be a problem because almost all are packaged. Think that with the current state of the release version it cannot be packaged. In Debian, when we prepare a package it has no network connection, so it cannot download anything. 

Checking the release version I found that:
 * NLOHMANN_JSON_ROOT: (ver.: 3.9.1) in Debian Stable 3.9.1-1 
 * XLINK_ROOT: not packaged, but can be released inside the sources?
 * BZIP2_ROOT:  (ver.: 1.0.8-p0) in Debian Stable 1.0.8-4
 * FP16_ROOT: your fork, in Debian Stable 0.0~git20200514.4dfe081-2
 * LIBARCHIVE-LUXONIS_ROOT: your fork ?? (ver.: 3.4.2-p2) in Debian Stable 3.4.3-2+b1
 * SPDLOG_ROOT: (ver.: 1.8.2)  in Debian Stable 1.8.1+ds-2.1
 * ZLIB_ROOT: (ver.: 1.2.11-p2)  in Debian Stable 1.2.11.dfsg-2
 * BACKWARD_ROOT:  (ver.: 1.6) I have no idea where it cames from ... I disabled it.
 * LIBNOP_ROOT: (ver.: 1.0-ec8f75a) in Debian Stable 0.0~git20200728.45dfe0f-3
Also, libdwarf-dev and libdw-dev are packaged.

I have built git master depthai-core with some modifications using the versions in Debian stable that it is my main platform and not using hunter. I have not finished yet because I'm patching the download part of the examples, but the library is built.

Is it possible to have some kind of IF in cmake to choose to use the hunter versions or the packaged versions? 

And yes, I wish that you guide me in this, but my target is a Debian package not a vcpkg. Something like:
```
$ apt install libdepthai-core-dev 
```

themarpe, 2022-03-03:
For libnop - the changes are on `develop` branch
https://github.com/google/libnop/compare/master...luxonis:develop

We've also had to fix a couple of things WRT to MSVC and adding CMake support. 
Not sure how projects are combined if they aren't supported out of the box... I've opened a PR against libnop for this, but haven't gotten any response from them... (https://github.com/google/libnop/pull/24)

> And for depthai-device-fwp ? Do you have a release files?

Those are downloaded by CMake upon configuration stage. But are included in the final binary by means of resource compilation, so they aren't required to be available separately.
Unless you think about something else by release files?

> The main problem is the dependencies and how to obtain it, but they will not be a problem because almost all are packaged. Think that with the current state of the release version it cannot be [...]

Does this include compilation from sources or only when supplying prebuilt binaries? As for prebuilt binaires, those are final, and won't require any additional network activity to be used.

> Is it possible to have some kind of IF in cmake to choose to use the hunter versions or the packaged versions?

Yes, but completely as of right now (its possible to do it more granuallary as well). Do `-D HUNTER_ENABLED=OFF` when configuring and all the `find_package` call will search for dependencies outside of Hunter. (Note, not tested as we don't target this usecase, but in theory should work)

> And yes, I wish that you guide me in this, but my target is a Debian package not a vcpkg. Something like:

For this usecase, I'd just compile depthai as is, and package the installation directory? (And assume no required dependencies). Then for `libdepthai-opencv-dev`, we assume opencv-dev dependency. Thoughts?

windelbouwman, 2022-04-01:
Thanks for the details of creating a debian package. I'm attempting to package depthai-core as an openembedded recipe, and am facing the same issues. It's good that hunter can be disabled, since I'm creating packages of the depdencies. Another problem as stated above is the patched libraries, it's best to use the unmodified upstream packages.

lepalom, 2023-03-19:
I have been working in this issue after some time stopped. 

It is very interesting, the work of @jwinarske for the yocto project. So, using their patches and some minor modifications, I have been able to build depthai-core without hunter in a Debian 11 (Bullseye), also with examples and using the system versions. 

However, it is not working because some firmware issues that I do not understand and I would like to ask @themarpe some guide. Think that my final target is to have something without hunter and without download any file.

I have found that CMake download two files: depthai-bootloader-fwp-XXXX and depthai-device-fwp-XXXX. That files are necessary and used in the Resources.cpp file. I have not seen how @jwinarske manages that topic in the yocto project.

That files are used in the build procedure? Because I have seen then only in the run time, but probably I have not understood quite well the process. 

Or, that files can be built in some way and how? 

All the blob files for the examples are necessary, right? Which license have they? 

And last question, because maybe I'm doing a wrong work. Do you have any tar.gz with ALL the necessary files (to be built in a computer without internet) to build dephai-core and use the OAK-D camera?

themarpe, 2023-03-19:
@lepalom

>That files are used in the build procedure? Because I have seen then only in the run time, but probably I have not understood quite well the process.

These files are used at the build time indeed - they are so called "resource compiled" into the final library. After the build, those files do NOT have to be present anywhere on the file system. Reason for this being that there is a tight coupling between the FW/BL & Host side library and checking if files are correct on FS would just bring additional complexity.

> All the blob files for the examples are necessary, right? Which license have they?

The FW/BL "blobs" (FW Packages) are proprietary (as is default - if not explicitly marked otherwise). The AI model blobs's licenses can be found by referencing the coresponding descriptions on their source. Example: https://docs.openvino.ai/latest/omz_models_model_mobilenet_v2.html - (note, specific version of zoo should be selected, this one is latest, just to paint the general idea)

> And last question, because maybe I'm doing a wrong work. Do you have any tar.gz with ALL the necessary files (to be built in a computer without internet) to build dephai-core and use the OAK-D camera?

No, this is an ongoing debate also over at #468

lepalom, 2023-03-19:
@themarpe 

Ok, tomorrow I will review my work. I have found that I didn't build depthai-resources library. Probably my fault. The CMake of depthai is really complex.

I will comment when I had more advances in this topic. In any case, it should be possible and recommended having a release with ALL necessary files to build depthai-core.

About the fw and bootloader, should I understand that depthai-bootloader-fwp-XXXX and depthai-device-fwp-XXXX are free software, right?

The others, I will check the license for every file.

lepalom, 2023-03-20:
@themarpe 

I have checked the patches and I have to admit that I don't understand the logic of the resources part of the CMakeLists.  I have copied the files in a resource directory, I have modified the CMakeList and it seems that works. It compiles, but it I try to run any example, the program crashes. The first time with an exception and the second time with an exception but showing that:

- depthai-bootloader-fwp-0.0.24.tar.xz doesn't exists.

When CMake is run I got this message:

-- Compiling depthai-core resources in PATCH_ONLY mode

The dephai-resources library  is not build (or I don't find libcmrc-base).

Please, could you give me some idea about what is going on?

jwinarske, 2023-03-20:
@lepalom The way I refactored things for CMake isolates all downloaded components external to the CMake.  Meaning Yocto is responsible for downloading all the artifacts, not CMake.  This is required for Long Term Support, where the configure phase is not allowed to access the network, and files are pulled from the global download directory.  The association is this:

```
DEPTHAI_BOOTLOADER_VERSION = "0.0.24"
DEPTHAI_DEVICE_SIDE_COMMIT = "8c3d6ac1c77b0bf7f9ea6fd4d962af37663d2fbd"

DEPTHAI_FWP_BOOTLOADER_FILENAME = "depthai-bootloader-fwp-${DEPTHAI_BOOTLOADER_VERSION}.tar.xz"
DEPTHAI_FWP_DEVICE_SIDE_FILENAME = "depthai-device-fwp-${DEPTHAI_DEVICE_SIDE_COMMIT}.tar.xz"
DEPTHAI_ARTIFACTORY_URL = "https://artifacts.luxonis.com/artifactory"
...
    ${DEPTHAI_ARTIFACTORY_URL}/luxonis-myriad-release-local/depthai-bootloader/${DEPTHAI_BOOTLOADER_VERSION}/${DEPTHAI_FWP_BOOTLOADER_FILENAME};unpack=0;name=bootloader-fwp \
    ${DEPTHAI_ARTIFACTORY_URL}/luxonis-myriad-snapshot-local/depthai-device-side/${DEPTHAI_DEVICE_SIDE_COMMIT}/${DEPTHAI_FWP_DEVICE_SIDE_FILENAME};unpack=0;name=device-fwp \
 ...
 SRC_URI[bootloader-fwp.sha256sum]         = "c913c636c86652ea96eda2135a517620f7d528d6357ddc697edfa20762a7cb30"
SRC_URI[device-fwp.sha256sum]             = "7b036b9ab1b5c20d37dffd4977ccbe1c43b54dbf770a2abbcdcfd3d4ffdf04b1"
...
    -D DEPTHAI_FWP_BOOTLOADER_FILENAME="${DEPTHAI_FWP_BOOTLOADER_FILENAME}" \
    -D DEPTHAI_FWP_DEVICE_SIDE_FILENAME="${DEPTHAI_FWP_DEVICE_SIDE_FILENAME}" \
    ```

Yocto downloads the files and validates sha256sum in recipe "fetch" phase.  The files are stored into the default working directory.  They are also cached in the general download directory for no-network use case.

The CMake patch uses `DEPTHAI_FWP_BOOTLOADER_FILENAME` and `DEPTHAI_FWP_DEVICE_SIDE_FILENAME` to set/specify the FWP files.  It doesn't matter the location of the file, as long as these two variables point to an absolute or relative file path.

Also the Yocto recipe defaults building depthai-core as a shared library.  Beyond setting `BUILD_SHARED_LIBS=ON` to depthai-core it also requires passing `CMAKE_POSITION_INDEPENDENT_CODE=ON` to both spdlog and XLink.  This turns into -fPIC in the project.

lepalom, 2023-03-21:
@jwinarske  @themarpe 

Now it is working. I had a stupid mistake with the DEPTHAI_BOOTLOADER_VERSION and the DEPTHAI_DEVICE_VERSION. I was using the @jwinarske patch version from March 15th. Now, I'm trying to prepare a debian package.

lepalom, 2023-03-22:
I have been working in this issue an I have a prototype of Debian package of depthai-core. It is built, it is installed, but there are some issues that I have to manage. Probably because I have the Debian Maintainer Hat it is difficult to me understand several things that I have found in depthai-core.

The hunter-free version packaged needs three repositories more:

* https://github.com/lepalom/XLink
* https://github.com/lepalom/libnop
* https://github.com/lepalom/depthai-resources

`XLink` and `libnop` are the same of the cloned version and the only thing that I have added is a tag v2.20.2 as a reference to create the package.

`depthai-resources` is a repo with the blob files, depthai-bootloader-fwp-0.0.24.tar.xz and depthai-device-fwp-8c3d6ac1c77b0bf7f9ea6fd4d962af37663d2fbd.tar.xz, that are the versions used in the 2.20.2 version.

The package _only_ install depthai includes that I have found later that is a mistake because what you need also:
- depthai-shared
- depthai-bootloader-shared

Trying to use the package I have found that it is a mess. I cannot understand why the CMake Config add the XLink flag to link against it. XLink is built as static library so, the user that use libdepthai-core (or libdepthai-opencv) don't have to link against XLink. Is it right? The libdepthai-core users needs anything from XLink?

OTOH, libnop provides another interesting problem to solve. Luxunis has its own patched version with the "EncodingByte::Empty" patch. libnop is a header _only_ library. No links against it. I don't know if the users of depthai-core needs to have their .h or needs to know anything about nop.

The link references also can be applied to libarchive and libbzip2, and all the references to 3rdparty directory in the includes.

If you are curious and have some experience in debian packaging, you can check it here:

https://salsa.debian.org/lepalom/depthai-core

The last point to say is that depthai-core doesn't compiles with gcc-12.

jwinarske, 2023-03-22:
@lepalom cool.  Yeah I think for a long term strategy it would be better to use the upstream package and patch per platform as needed.  Not all patches apply to all platforms.  Building against a "dirty" library isn't ideal.

In my cmake_json project I implemented a patching scheme.  This could be extended to differentiate platform types. It's a working proposal to replace the hunter implementation:
https://github.com/jwinarske/cmake_json/

For packaging I like to use CMake.  It has all the support to generate TGZ, RPM, Debian, and Bundles.  Requires minimal setup.  This is the pattern:

Checks if host (not cross-compile build), if so include `packaging.cmake`
https://github.com/jwinarske/cmake_json/blob/main/CMakeLists.txt#L35

Builds TGZ, RPM, DEB
https://github.com/jwinarske/cmake_json/blob/main/cmake/packaging.cmake

Nightly CI job that uses this pattern:
https://github.com/toyota-connected/ivi-homescreen/blob/main/.github/workflows/ivi-homescreen-linux.yml
https://github.com/toyota-connected/ivi-homescreen/blob/main/cmake/packaging.cmake

Just requires

    make package -j

lepalom, 2023-03-22:
@jwinarske , I think that the problem is that depthai had cherry picked other projects for it own proposals and patched them. To put it in main distros as Debian, or Fedora in the Linux world it is not the best option. And I personally don't like approaches like snap.

I have seen your cmake_json and it seems a nice solution. And yes, I know the Cpack possibilities of CMake. But, it was a debate in the debian-devel list and it had not many acceptances. The Debian packages has their quality because it is an accurate process and manually checked to, from the same source, to be built in all the archs (currently officially [10](https://wiki.debian.org/SupportedArchitectures).

Another thing is that from my point of view, depthai-core is a mesh of header files, making a nightmare or carrying a lot of code that you don't use but you need to compile or link against it.

In any case, I will write some issues. Maybe I'm wrong and everything is ok.