luxonis/depthai-core #414: Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 80128FF0' '0'

diablodale, 2022-03-06:
Device critical trap error running tests on my OAK-D with depthai-core v2.15.0
```
[14442C10C1A1C6D200] [4.048] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 80128FF0' '0'
```
and threw exception with text `Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)`

### Setup

* Microsoft Windows [Version 10.0.19044.1526]
* depthai-core v2.15.0
* VS v16.11.10
* OAK-D

### Repro

1. Write a app for testing that creates a pipeline with permutations of settings and opens a Device.
2. Iterate through them. Here is one that I found with a device error fatal trap.
```
color=off
depth=400p
imu=off
subpixel=off
align=tocolor
lrcheck=off
filter=median5x5
fps=30.0
usb
```

### Result

Open fails with exception `Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)` and critical device fail trap via spdlog

```
[14442C10C1A1C6D200] [4.048] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 80128FF0' '0'
```

### Expected

No traps or critical failures like this. If some combination of settings is not allowed, then a normal exception with specific incompatibility text.

szabi-luxonis, 2022-04-20:
@diablodale addressed in https://github.com/luxonis/depthai-core/commit/08be0a71b315c8fe4dbb58bf14bb0fe57e826e51

diablodale, 2022-04-23:
Unfortunately, a similar errors still occur
```
[14442C10C1A1C6D200] [3.337] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 80136744' '0'

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
perf_hw_cputemp.exe is a Catch v2.13.7 host application.
Run with -? for options

-------------------------------------------------------------------------------
Hardware CPU and Temp
-------------------------------------------------------------------------------
..\tests\src\hardware_cputemp_perf.cpp(107)
...............................................................................

..\tests\src\hardware_cputemp_perf.cpp(272): FAILED:
  REQUIRE_NOTHROW( [&]() { const auto targetDevice = getAvailableDevice(std::chrono::seconds(protocol == X_LINK_TCP_IP ? 60 : 
15), protocol); do { ; Catch::AssertionHandler catchAssertionHandler( "REQUIRE"_catch_sr, ::Catch::SourceLineInfo( "..\\tests\\src\\hardware_cputemp_perf.cpp", static_cast<std::size_t>( 272 ) ), "std::get<bool>(targetDevice)", Catch::ResultDisposition::Normal ); try { __pragma( warning(push) ) catchAssertionHandler.handleExpr( Catch::Decomposer() <= std::get<bool>(targetDevice) ); __pragma( warning(pop) ) } catch(...) { catchAssertionHandler.handleUnexpectedInflightException(); } catchAssertionHandler.complete(); } while( (void)0, (false) && static_cast<bool>( !!(std::get<bool>(targetDevice)) ) ); device = std::make_unique<dai::Device>(p, std::get<dai::DeviceInfo>(targetDevice)); }() )
due to unexpected exception with messages:
  epoch,color,depth,mono,imu,subpixel,align,lrcheck,filter_median,fps,protocol
  1650729462,0,400,0,false,false,true,false,5,30.000000,usb,
  Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)

[14442C10C1A1C6D200] [3.368] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 80136744' '0'
-------------------------------------------------------------------------------
Hardware CPU and Temp
-------------------------------------------------------------------------------
..\tests\src\hardware_cputemp_perf.cpp(107)
...............................................................................

..\tests\src\hardware_cputemp_perf.cpp(272): FAILED:
  REQUIRE_NOTHROW( [&]() { const auto targetDevice = getAvailableDevice(std::chrono::seconds(protocol == X_LINK_TCP_IP ? 60 : 
15), protocol); do { ; Catch::AssertionHandler catchAssertionHandler( "REQUIRE"_catch_sr, ::Catch::SourceLineInfo( "..\\tests\\src\\hardware_cputemp_perf.cpp", static_cast<std::size_t>( 272 ) ), "std::get<bool>(targetDevice)", Catch::ResultDisposition::Normal ); try { __pragma( warning(push) ) catchAssertionHandler.handleExpr( Catch::Decomposer() <= std::get<bool>(targetDevice) ); __pragma( warning(pop) ) } catch(...) { catchAssertionHandler.handleUnexpectedInflightException(); } catchAssertionHandler.complete(); } while( (void)0, (false) && static_cast<bool>( !!(std::get<bool>(targetDevice)) ) ); device = std::make_unique<dai::Device>(p, std::get<dai::DeviceInfo>(targetDevice)); }() )
due to unexpected exception with messages:
  epoch,color,depth,mono,imu,subpixel,align,lrcheck,filter_median,fps,protocol
  1650729468,0,400,0,false,false,true,true,5,30.000000,usb,
  Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)

[14442C10C1A1C6D200] [3.404] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 80136744' '0'
-------------------------------------------------------------------------------
Hardware CPU and Temp
-------------------------------------------------------------------------------
..\tests\src\hardware_cputemp_perf.cpp(107)
...............................................................................

..\tests\src\hardware_cputemp_perf.cpp(272): FAILED:
  REQUIRE_NOTHROW( [&]() { const auto targetDevice = getAvailableDevice(std::chrono::seconds(protocol == X_LINK_TCP_IP ? 60 : 
15), protocol); do { ; Catch::AssertionHandler catchAssertionHandler( "REQUIRE"_catch_sr, ::Catch::SourceLineInfo( "..\\tests\\src\\hardware_cputemp_perf.cpp", static_cast<std::size_t>( 272 ) ), "std::get<bool>(targetDevice)", Catch::ResultDisposition::Normal ); try { __pragma( warning(push) ) catchAssertionHandler.handleExpr( Catch::Decomposer() <= std::get<bool>(targetDevice) ); __pragma( warning(pop) ) } catch(...) { catchAssertionHandler.handleUnexpectedInflightException(); } catchAssertionHandler.complete(); } while( (void)0, (false) && static_cast<bool>( !!(std::get<bool>(targetDevice)) ) ); device = std::make_unique<dai::Device>(p, std::get<dai::DeviceInfo>(targetDevice)); }() )
due to unexpected exception with messages:
  epoch,color,depth,mono,imu,subpixel,align,lrcheck,filter_median,fps,protocol
  1650729499,0,400,0,false,true,true,false,5,30.000000,usb,
  Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)

[14442C10C1A1C6D200] [3.300] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 80136744' '0'
-------------------------------------------------------------------------------
Hardware CPU and Temp
-------------------------------------------------------------------------------
..\tests\src\hardware_cputemp_perf.cpp(107)
...............................................................................

..\tests\src\hardware_cputemp_perf.cpp(272): FAILED:
[... cut ...] dler.handleUnexpectedInflightException(); } catchAssertionHandler.complete(); } while( (void)0, (false) && static_cast<bool>( !!(std::get<bool>(targetDevice)) ) ); device = std::make_unique<dai::Device>(p, std::get<dai::DeviceInfo>(targetDevice)); }() )
due to unexpected exception with messages:
  epoch,color,depth,mono,imu,subpixel,align,lrcheck,filter_median,fps,protocol
  1650729505,0,400,0,false,true,true,true,5,30.000000,usb,
  Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)
```

MAVProxyUser, 2023-03-19:
I get this same error trying to use the ROS nodes on OAK-1 MAX 
https://github.com/luxonis/depthai-python/issues/790