luxonis/depthai-core #415: OAK-D-Pro-PoE fails after 6 connections and requires power-cycle

diablodale, 2022-03-06:
OAK-D-Pro-PoE fails after 6 connections and requires power-cycle. I now have repro code. 🙃
When OAK enters this dead state, it does not respond to pings or appear active on the ethernet network (though my LAN diagnostic tools are limited)

A PR is forthcoming that holds the repro code. It is a new "performance" test which I would like to eventually merge so that Luxonis and others can instrument/evaluate the OAK hardware CPU+temperature stats with different builds and different permutations of Pipeline settings.

### Setup

* OAK-D-Pro-PoE preproduction
* Ubiquiti POE Injector PoE+ Gigabit 802.3af -- a full Ubi switch is forever not available in Europe 😢
* TP-Link 1gpb ethernet switch
* Microsoft Windows [Version 10.0.19044.1526]
* depthai-core v2.15.0
* VS2019 v16.11.10  (a c++17 compiler is required for the repro code)

### Repro

1. git clone from branch in forthcoming PR. The PR will hold the repro code
2. config, build for: x64, Debug, shared libs, with all examples and tests
3. run the new test `build\tests\perf_hw_cputemp.exe`

### Result

Only 6 iterations will be successful. The 7th will fail since the OAK is dead.

### Expected

784 test iterations to run which will likely be a day or more of time. But I've never gotten past 6.

### Notes

I can use my OAK-D for this same test and it is successful running all 784. I have lots of data that I want to compare against PoE.
If you want to use a USB sensor, you will need to change the test code `auto protocol = GENERATE(X_LINK_TCP_IP);`.

This performance test is written for C++17 -- can't be bothered to make it work with earlier. The PR also includes updates so that tests can specify the c++ standard they need and the cmake harness will do the right thing.

alex-luxonis, 2022-03-06:
@diablodale Could you check what bootloader version is installed, and upgrade it to the latest `0.0.17`, it should have some fixes for this problem. Can use this app to upgrade:
https://github.com/luxonis/depthai-core/blob/main/examples/bootloader/flash_bootloader.cpp

diablodale, 2022-03-06:
My device was `Version: 0.0.15` and using that tool is now `Version: 0.0.17`
Unfortunately, that didn't help. It continues to die between the 6 and 7th iteration. Which is a good thing...it is a consistent repro of a fail.

diablodale, 2022-03-07:
PR is posted. I'm interested to see what you experience with your similar hardware.

alex-luxonis, 2022-03-07:
Thanks @diablodale.

I'm able to reproduce the failure on OAK-D-Pro-PoE, with the same behavior as on your side: fails at the 7th iteration, and the device no longer responds to pings until power-cycled.

I also tested on [OAK-FFC-PoE-3P](https://shop.luxonis.com/products/oak-ffc-poe-3p) and it failed, but after a longer time: 54 mins, 141 iterations. 
Logs: https://gist.github.com/alex-luxonis/366bfd117cc0bb25632eacd3edecdb01/9a1c7f3510dbd779a3033c6aaeeda5c567149686
It may be the same issue observed on https://github.com/luxonis/depthai-core/issues/406 (CC @chkarl)

We are investigating.

Luxonis-Brandon, 2022-03-10:
@diablodale - do you happen to have the exact part number for the `TP-Link 1gpb ethernet switch` so that we can order the same one (if it's still available)?

This may just be advantageous for debugging any other future issues as well, as likely it is better at bringing out the crash(es) we are seeing here, faster.

Thanks again for the help!

diablodale, 2022-03-10:
TP-Link 5-port Gigabit Desktop Switch
Model:  TL-SG1005D  Ver: 4.0

I bought this in 2011, I do not see the exact model avail today. The model# is the same but the look of the device is radically different. Here is mine...
[image]

Ubiquiti POE Injector PoE+ Gigabit 802.3af
PN:  U-POE-af

Some of my customers are artists making performances/exhibits; probably will not have an IT dept with managed racks of PoE switches. They will use instead injectors and a friendly consumer ethernet switch like mine.
And, there will be some set that do have IT depts + racks...for them I'll eventually use a Ubi switch for my testing (when they are again avail in Germany).

diablodale, 2022-03-10:
My pre-production oak-d-pro-poe
[image]

themarpe, 2022-10-21:
Resolved by: https://github.com/luxonis/depthai-core/releases/tag/v2.17.1 onwards
(At least the exact specific scenario as described in OP)

closed 2022-10-21 by themarpe

diablodale, 2022-10-21:
Yay! Will test when I again have access to my OAK PoE in November. :-)