luxonis/depthai-core #431: OAK-D-Pro-PoE sends no IMU data to host

diablodale, 2022-03-24:
I just today noticed, my OAK-D-PRO-POE sends no IMU data to the host.

### Setup

* depthai-core v2.15.1
* Microsoft Windows [Version 10.0.19044.1586]
* VS2019 v16.11.11
* oak-d-pro-poe sensor

### Repro

1. clone, config, build depthai-core for x64, shared lib, Debug, all examples+tests
2. set env var `DEPTHAI_LEVEL=info`
3. Run `examples/imu_gyroscope_accelerometer.exe`

### Result

No gyro/acc output to console. Only the device info reports

```
[192.168.2.23] [412.203] [system] [info] Temperatures - Average: 40.62 ┬░C, CSS: 41.89 ┬░C, MSS 40.04 ┬░C, UPA: 38.88 ┬░C, DSS: 41.66 ┬░C
[192.168.2.23] [412.203] [system] [info] Cpu Usage - LeonOS 16.90%, LeonRT: 0.04%
[192.168.2.23] [413.204] [system] [info] Memory Usage - DDR: 0.13 / 339.99 MiB, CMX: 2.06 / 2.50 MiB, LeonOS Heap: 25.89 / 78.27 MiB, LeonRT Heap: 2.95 / 41.54 MiB
[192.168.2.23] [413.204] [system] [info] Temperatures - Average: 40.45 ┬░C, CSS: 41.66 ┬░C, MSS 39.81 ┬░C, UPA: 38.65 ┬░C, DSS: 41.66 ┬░C
...
```

### Expected

gyro/accel data output to console, and the info reports

### Notes

My OAK-D has no problems with this.

This problem occurs in my own app, and then I saw the problem also occurs with the examples.

If I pause the example exe, I see that it is waiting in the call `imuData = imuQueue->get<dai::IMUData>();`
[image]

szabi-luxonis, 2022-03-24:
@diablodale please check out this PR: https://github.com/luxonis/depthai-core/pull/399
IIRC you have a pre-production unit which probably didn't go through IMU FW update

diablodale, 2022-03-24:
🤹‍♀️ can I also use https://github.com/luxonis/depthai-core/issues/319#issuecomment-1003446759 ? Would be somewhat easier for me.  fyi, the test script you also provided in that issue outputs nothing from this poe IMU. No imu part or firmware numbers.

szabi-luxonis, 2022-03-24:
> woman_juggling can I also use [#319 (comment)](https://github.com/luxonis/depthai-core/issues/319#issuecomment-1003446759) ? Would be somewhat easier for me. fyi, the test script you also provided in that issue outputs nothing from this poe IMU. No imu part or firmware numbers.

Yes, you can. Or better use this: https://github.com/luxonis/depthai-python/pull/515 and run any of the IMU examples. Easier to install a python lib.

szabi-luxonis, 2022-03-24:
@diablodale you can install the library from `imu_firmware_update` directly as:

```
python3 -m pip install -U --prefer-binary --user --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local "depthai==2.14.1.0.dev+682598dde411486e6fc7cb4f6222a1da2606fce0"
```

0.4.0 didn't have POE support back then.

diablodale, 2022-03-25:
No luck. :-/
1. checked out the current `imu_firmware_update` branch.
2. created a venv
3. `python -m pip install -U --prefer-binary --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local "depthai==2.14.1.0.dev+682598dde411486e6fc7cb4f6222a1da2606fce0"`
4. `pip install opencv-python`
5. `cd examples\imu`
6. `python imu_rotation_vector.py`
7. Wait 5 minutes
8. Nothing happens, so I press ctrl-c

It seems to be blocking on `get()` like my C++ app did.

```
(.venv) C:\njs\depthai-python>python -m pip install -U --prefer-binary --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local "depthai==2.14.1.0.dev+682598dde411486e6fc7cb4f6222a1da2606fce0"
Looking in indexes: https://pypi.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local
Collecting depthai==2.14.1.0.dev+682598dde411486e6fc7cb4f6222a1da2606fce0
  Downloading https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/depthai/depthai-2.14.1.0.dev0%2B682598dde411486e6fc7cb4f6222a1da2606fce0-cp39-cp39-win_amd64.whl (10.2 MB)
     |████████████████████████████████| 10.2 MB 345 kB/s
Installing collected packages: depthai
Successfully installed depthai-2.14.1.0.dev0+682598dde411486e6fc7cb4f6222a1da2606fce0
WARNING: You are using pip version 21.1.1; however, version 22.0.4 is available.
You should consider upgrading via the 'C:\njs\depthai-python\.venv\Scripts\python.exe -m pip install --upgrade pip' command.

(.venv) C:\njs\depthai-python>pip install opencv-python
Collecting opencv-python
  Downloading opencv_python-4.5.5.64-cp36-abi3-win_amd64.whl (35.4 MB)
     |████████████████████████████████| 35.4 MB 3.3 MB/s
Collecting numpy>=1.17.3
  Using cached numpy-1.22.3-cp39-cp39-win_amd64.whl (14.7 MB)
Installing collected packages: numpy, opencv-python
Successfully installed numpy-1.22.3 opencv-python-4.5.5.64
WARNING: You are using pip version 21.1.1; however, version 22.0.4 is available.
You should consider upgrading via the 'c:\njs\depthai-python\.venv\scripts\python.exe -m pip install --upgrade pip' command.

(.venv) C:\njs\depthai-python\examples\IMU>python imu_rotation_vector.py
Traceback (most recent call last):
  File "C:\njs\depthai-python\examples\IMU\imu_rotation_vector.py", line 39, in <module>
    imuData = imuQueue.get()  # blocking call, will wait until a new data has arrived
KeyboardInterrupt
^C
```

szabi-luxonis, 2022-03-26:
> No luck. :-/
>
> 1. checked out the current `imu_firmware_update` branch.
> 2. created a venv
> 3. `python -m pip install -U --prefer-binary --extra-index-url [...]

Can you check out latest https://github.com/luxonis/depthai-core/pull/399 run examples/install_requirements.py and copy the logs from device if still fails?

diablodale, 2022-03-26:
No success. :-/

1. same checkout above and venv
2. git merged in your firmware sha update of yesterday
3. `python install_requirements.py` (I noticed it downloaded a whl with a 2.14 version `depthai-2.14.1.0.dev...`)

```
(.venv) C:\njs\depthai-python\examples\IMU>git pull --ff-only upstream imu_firmware_update
From https://github.com/luxonis/depthai-python
 * branch            imu_firmware_update -> FETCH_HEAD
Already up to date.

(.venv) C:\njs\depthai-python\examples\IMU>cd ..

(.venv) C:\njs\depthai-python\examples>python install_requirements.py
pip 22.0.4 from C:\njs\depthai-python\.venv\lib\site-packages\pip (python 3.9)
Requirement already satisfied: pip in c:\njs\depthai-python\.venv\lib\site-packages (22.0.4)
Requirement already satisfied: pyyaml in c:\njs\depthai-python\.venv\lib\site-packages (6.0)
Requirement already satisfied: requests in c:\njs\depthai-python\.venv\lib\site-packages (2.27.1)
Requirement already satisfied: numpy in c:\njs\depthai-python\.venv\lib\site-packages (1.22.3)
Requirement already satisfied: opencv-python!=4.5.4.58 in c:\njs\depthai-python\.venv\lib\site-packages (4.5.5.64)
Requirement already satisfied: certifi>=2017.4.17 in c:\njs\depthai-python\.venv\lib\site-packages (from requests) (2021.10.8)
Requirement already satisfied: idna<4,>=2.5 in c:\njs\depthai-python\.venv\lib\site-packages (from requests) (3.3)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\njs\depthai-python\.venv\lib\site-packages (from requests) (1.26.9)
Requirement already satisfied: charset-normalizer~=2.0.0 in c:\njs\depthai-python\.venv\lib\site-packages (from requests) (2.0.12)
From https://github.com/luxonis/depthai-core
 * branch              748667d35bc088e150f0fea881f7fa9aabb8b68a -> FETCH_HEAD
Looking in indexes: https://pypi.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local
Collecting depthai==2.14.1.0.dev+86cc4f83aa285e2a32fc2144cbfee20418d60695
  Downloading https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/depthai/depthai-2.14.1.0.dev0%2B86cc4f83aa285e2a32fc2144cbfee20418d60695-cp39-cp39-win_amd64.whl (10.2 MB)
     ---------------------------------------- 10.2/10.2 MB 9.9 MB/s eta 0:00:00
Installing collected packages: depthai
  Attempting uninstall: depthai
    Found existing installation: depthai 2.14.1.0.dev0+682598dde411486e6fc7cb4f6222a1da2606fce0
    Uninstalling depthai-2.14.1.0.dev0+682598dde411486e6fc7cb4f6222a1da2606fce0:
      Successfully uninstalled depthai-2.14.1.0.dev0+682598dde411486e6fc7cb4f6222a1da2606fce0
Successfully installed depthai-2.14.1.0.dev0+86cc4f83aa285e2a32fc2144cbfee20418d60695
################|| Downloading calibration-sample ||################

========== Retrieving C:\njs\depthai-python\examples\models\depthai_calib.json from the cache

========== Retrieving C:\njs\depthai-python\examples\models\depthai_v5.calib from the cache

========== Retrieving C:\njs\depthai-python\examples\models\BW1098OBC.json from the cache

################|| Downloading concat-model ||################

========== Retrieving C:\njs\depthai-python\examples\models\concat_openvino_2021.4_6shave.blob from the cache

################|| Downloading mobilenet-ssd ||################

========== Retrieving C:\njs\depthai-python\examples\models\mobilenet-ssd_openvino_2021.4_6shave.blob from the cache

========== Retrieving C:\njs\depthai-python\examples\models\mobilenet-ssd_openvino_2021.4_5shave.blob from the cache

========== Retrieving C:\njs\depthai-python\examples\models\mobilenet-ssd_openvino_2021.4_8shave.blob from the cache

################|| Downloading normalization-model ||################

========== Retrieving C:\njs\depthai-python\examples\models\normalize_openvino_2021.4_4shave.blob from the cache

################|| Downloading person-detection-retail-0013 ||################

========== Retrieving C:\njs\depthai-python\examples\models\person-detection-retail-0013_openvino_2021.4_7shave.blob from the cache

################|| Downloading stereo-dataset ||################

========== Retrieving C:\njs\depthai-python\examples\models\dataset.zip from the cache

========== Unpacking C:\njs\depthai-python\examples\models\dataset.zip
[... 43 lines cut ...]
[192.168.2.23] [40.381] [IMU(0)] [info] Part 10003254 : Version 4.4.3 Build 485
[192.168.2.23] [40.381] [IMU(0)] [info] Part 10003171 : Version 4.2.10 Build 548
[192.168.2.23] [40.381] [IMU(0)] [warning] IMU firmware version detected: 3.2.13 differs from latest available: 3.9.7. Use 'imu.doFirmwareUpdate()' to update firmware to latest.
[192.168.2.23] [40.418] [system] [info] Memory Usage - DDR: 0.13 / 339.80 MiB, CMX: 2.06 / 2.50 MiB, LeonOS Heap: 24.84 / 78.27 MiB, LeonRT Heap: 2.92 / 41.54 MiB
[192.168.2.23] [40.418] [system] [info] Temperatures - Average: 37.94 ┬░C, CSS: 38.65 ┬░C, MSS 37.24 ┬░C, UPA: 37.01 ┬░C, DSS: 38.88 ┬░C
[192.168.2.23] [40.418] [system] [info] Cpu Usage - LeonOS 3.28%, LeonRT: 3.92%
[192.168.2.23] [41.419] [system] [info] Memory Usage - DDR: 0.13 / 339.80 MiB, CMX: 2.06 / 2.50 MiB, LeonOS Heap: 25.89 / 78.27 MiB, LeonRT Heap: 2.92 / 41.54 MiB
[192.168.2.23] [41.419] [system] [info] Temperatures - Average: 38.47 ┬░C, CSS: 39.81 ┬░C, MSS 38.18 ┬░C, UPA: 37.01 ┬░C, DSS: 38.88 ┬░C
[192.168.2.23] [41.419] [system] [info] Cpu Usage - LeonOS 12.35%, LeonRT: 0.05%
[192.168.2.23] [42.420] [system] [info] Memory Usage - DDR: 0.13 / 339.80 MiB, CMX: 2.06 / 2.50 MiB, LeonOS Heap: 25.89 / 78.27 MiB, LeonRT Heap: 2.92 / 41.54 MiB
[... cut ...]
[2022-03-26 16:06:31.525] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[2022-03-26 16:06:31.525] [debug] XLinkResetRemote of linkId: (0)
[2022-03-26 16:06:31.525] [debug] DataOutputQueue (imu) closed
[2022-03-26 16:06:31.695] [debug] Device closed, 267
Traceback (most recent call last):
  File "C:\njs\depthai-python\examples\IMU\imu_rotation_vector.py", line 39, in <module>
    imuData = imuQueue.get()  # blocking call, will wait until a new data has arrived
KeyboardInterrupt
^C
```

szabi-luxonis, 2022-03-26:
@diablodale add `imu.doFirmwareUpdate()` to the script, it will do IMU firmware update. I will make this behavior automatic.
Looks like we didn't test your beta board before shipping it out, sorry about that.

diablodale, 2022-03-26:
No problem on my beta board. I fully expect all types of odd behaviors from a non-production unit.
Yes, I will add that to the python script.

I am interested in the length of time needed for the firmware update. I remember on my usb oak-d it was not "fast". It needed many seconds. That test utility I used has a UI and I could see progress so I was patient. Given this sensor is PoE, I suspect it will also be many seconds or more. The startup time (if automatic) will be substantially more than a user/customer expects. And there is no progress UI so they know to be patient. The user/customer ?might? think this is an error condition and ctrl-c, stop, kill, or disconnect the sensor....in the middle of the firmware update. Which I suspect is a bad thing.

Is that firmware update delay being considered and have a reliable/good solution for those customers that don't know to wait and interrupt the firmware update? I understand that my unit is beta and needs the firmware. I can also image a future where you need to make a firmware update for the production models which introduces the potential of interrupted firmware updates. 🤔

szabi-luxonis, 2022-03-26:
Firmware update is performed during factory calibration/test, so for production units it's not needed.
It handles interrupt in the middle of firmware update, so it won't get bricked, and it also has progress report.
It takes about 30 seconds.

diablodale, 2022-03-26:
I have low-confidence that Luxonis will never in the life of your sensors update the IMU firmware. This is based on the last ~20 years of tech industry that introduced a practice of updating their hardware with firmware/drivers.

I see no progress update callback API. Perhaps there is console logging (what log level?) but that leads to similar problems as #400 and #418. Since the update is considerable in length and the unit *does not function correctly*, I consider this an error condition like in those two related issues. The `Device` startup would throw with the "IMU firmware version detected: 3.2.13 differs from latest available: 3.9.7. Use 'imu.doFirmwareUpdate()' to update firmware to latest." message.

This allows the developer/app/IT to alert the user/customer of the situation and to prepare for that firmware update. Or...rollback to an earlier depthai version that does not need that firmware.

diablodale, 2024-03-19:
OAK-D-Pro-PoE now sends imu data. Unfortunatey, the accelerometer data is incorrect.
Getting any data required upgrading the OAK-D-Pro-PoE firmware to v3.9.9. And it was tested using depthai-core v2.25.0
I'm closing this issue since (incorrect) data is now being sent by PoE and received by depthai-core. This issue does not focus on bad data

closed 2024-03-19 by diablodale