luxonis/depthai-core #433: [Feature-Request] Configure ISO Independently of Exposure Time

justin-larking-pk, 2022-03-25:


alex-luxonis, 2022-03-25:
I agree this would be very valuable, but unfortunately it's not easy to implement, as the 3A library on device controls both exposure and ISO, and the rule of combining them is described in the camera tuning database.
I added more info here (related): https://github.com/luxonis/depthai-core/issues/304#issuecomment-997121371

We'll check what's possible to do.

wdimmit, 2022-11-22:
Has there been any progress on this issue (or the related #304)?

priyajakhar, 2023-05-26:
Having this feature would solve our problems too in high vibration environment where we want to put a cap on shutter speed while still allowing exposure to be adjusted automatically.

alex-luxonis, 2023-11-15:
It's possible now to set an upper limit for auto-exposure: https://github.com/luxonis/depthai-python/pull/921. Once the limit is reached, AE will increase ISO as needed.
Example: `cam.initialControl.setAutoExposureLimit(5000)`  # microseconds. Can also be updated at runtime