luxonis/depthai-core #434: feature req: set mesh data after Device created, or get calibration before Device started, etc.

diablodale, 2022-03-25:
I have need to calculate and set the undistortion mesh data for the OAK device *after* `Device` is created.
Why?
1. Calibration values are needed for the specific OAK device to create the mesh at a specific resolution and camera(s)
2. Calibration values can only be gotten *after* that specific device is started
3. But then depthai-core doesn't support calling `StereoDepth::loadMeshData()`. I don't see another api on `dai::RawStereoDepthConfig` or `dai::StereoDepthConfig`

### Setup

* all

### Scenario

My app works dynamically and can control any number and type of OAK sensors. I want to undistort the sensor data and one approach is to use the OAK "mesh files". These files are unique to an OAK device, the specific sensor resolution on that OAK device, and the mesh step size. For example, an OAK-D with stepsize=16 would have 8 mesh files (4 resolutions * 2 (left and right)).

Given the easy of creating the mesh data and the variations, I prefer to generate the data myself in memory and then load it into the depthai pipeline. But to do that, I need to know:

1. customers choice of resolution. I easily have this information.
2. specific calibration data for that sensor and that chosen resolution. I only know this after the `Device` is started and I can get the calibration handler to call the cali APIs.
3. specific parameters to the `cv::initUndistortRectifyMap()` api used to create the mesh data. One of those params can accept the call `cv::getOptimalNewCameraMatrix()` which takes dynamic runtime parameters. I use this with my Kinect solutions and would like to also have it on my OAK solutions.

Number (3) means that my customers can at runtime adjust the params to those APIs, which creates a new set of mesh data, which then needs to be uploaded at runtime to the OAK sensor which is continuing to send data to the host.

### Workaround

Nothing pretty that I've found.

* Iterating and caching all possible combinations of resolutions for a specific sensor.
* Or starting the device to read the calibration, then stopping to generate the data, then restarting it again to load that data. This would be the death of PoE and isn't realistic.
* I could have a degraded feature if I could get the calibration before `Device` start. Perhaps via your newer rpc mechanism. This allows me to generate the mesh data for a specific sensor+resolution but not the runtime adjustments.

szabi-luxonis, 2022-03-25:
Porting the mesh generator to on device is WIP so generating it on host won't be needed.

diablodale, 2022-03-25:
Does the current plan allow c++ code to update at runtime the undistortion parameters with something like `getOptimalNewCameraMatrix()`. Or is the mesh calculated "once" and can not be changed while the `Device` is started?

diablodale, 2022-04-27:
closing: the on-device mesh generation was released with https://github.com/luxonis/depthai-core/releases/tag/v2.15.3 and I have private code that meets my other needs.

closed 2022-04-27 by diablodale