luxonis/depthai-core #438: [system] [critical] Fatal error. Please report to developers. Log: 'PlgWarpHW' '741'

diablodale, 2022-03-28:
Hi. While running a pipeline, I got

```
[1844301051F4691200] [2571.232] [system] [critical] Fatal error. Please report to developers. Log: 'PlgWarpHW' '741'
```

I was trying to isolate the cause of stereodepth frames having the same seq number as color frames yet when visually inspected they are definitely not in sync

### Setup

* depthai-core v2.15.1, built for debug
* VS2019 v16.11.11
* Microsoft Windows [Version 10.0.19044.1586]
* OAK-D-Lite. This sensor was calibrated using the `lite-calibration` branch

### Repro

It is in my C++ app, so the setup is complex. I have been trying to isolate the lag issue using python unsuccessfully. So, here is a general description.

Build a c++ application that visually outputs on the host StereoDepth::depth and ColorCamera::video with config...

* all sensors fps=30
* color sensor=1080p and setIspScale(4,9) and setVideoSize(832,480)
* stereodepth sensor=400p and setDepthAlign(dai::CameraBoardSocket::RGB)
* leftrightcheck=true
* subpixel=true
* setConfidenceThreshold=240
* setRectifyEdgeFillColor=0
* setMedianFilter=5x5
* thresholdFilter = 700 -> 10000

### Result

The visual inspection of the depth and color frames starts in sync. But in less than a minute, the depth is visually lagging. The depth will *visually* progressively lag more and more behind the color. And sometimes I get the fatal error...

```
[1844301051F4691200] [2571.232] [system] [critical] Fatal error. Please report to developers. Log: 'PlgWarpHW' '741'
```

### Expected

No lag. No fatal error.

### Notes

I can not reproduce this lag on my 2nd OAK-D-Lite. 
I can run the two Lite sensors side-by-side with the same code in memory. Only one sensor lags (and sometimes this fatal error).
I can use the same cable. It is still just one sensor that lags.
The only thing that is different that I cause...is the one that lags was calibrated with the `lite-calibration` branch.
The sensor that does *not* lag has its original factory calibration.

diablodale, 2022-03-28:
firmware fatal seems related to https://github.com/luxonis/depthai-core/pull/427
could the lag also?

Luxonis-Brandon, 2022-03-28:
Thanks for the details!  Triaging.

szabi-luxonis, 2022-03-29:
> firmware fatal seems related to #427
> could the lag also?

Can you reproduce in latest develop which contains that proposed fix? 2.15.1 doesn't contain. It's an old BUG that appears with RGB alignment.

diablodale, 2022-03-29:
Using current `develop` d5349cada8c94b766b50c6f0f22b25f753e7cabf I have not reproduced the fatal `'PlgWarpHW' '741'`.

I have evidence of progressive lag, but that is a separate issue.
closing.