luxonis/depthai-core #440: OAK-D-Lite StereoDepth is progressively lagging color yet has the same sequence number

diablodale, 2022-03-29:
I have evidence of one of my two OAK-D-Lite sensors sending frames of data to the host where the timestamps are progressively out-of-sync (even more than 1 second) yet have the same sequence number.

I received these two sensors in the same kickstarter pledge. The only thing I know is different, is that the sensor with the progressive lag was calibrated with the script from `lite-calibration` branch. My other D-Lite has its factory calibration.

### Setup

* depthai-core v2.15.1 and the newer develop d5349cada8c94b766b50c6f0f22b25f753e7cabf
* VS2019 v16.11.11
* Microsoft Windows [Version 10.0.19044.1586]

### Repro

It is in my C++ app, so the setup is complex. I have not been able to isolate this lag issue using python.
Here is a general description.

1. Build a c++ application that visually outputs on the host StereoDepth::depth and ColorCamera::video with config...

* all sensors fps=30
* color sensor=1080p and setIspScale(4,9) and setVideoSize(832,480)
* stereodepth sensor=400p and setDepthAlign(dai::CameraBoardSocket::RGB)
* leftrightcheck=true
* subpixel=true
* setConfidenceThreshold=240
* setRectifyEdgeFillColor=0
* setMedianFilter=5x5
* thresholdFilter = 700 -> 10000

2. Use the dict approach to synchronize frames on the host with their sequence number.
3. create loop which gets from the dict a sync'd pair, verifies it, and displays them
4. The verification should confirm the sequence numbers of the pair are equal.
5. Output the chrono diff of the dai::imgframe timestamps of the sync'd pair to the console in units of floating point seconds

### Result

My 1st OAK-D-Lite that was calibrated has problems. The visual inspection of the depth and color frames starts in sync. But in less than a minute, the depth is visually lagging. The depth will *visually* progressively lag more and more behind the color. Video at https://www.youtube.com/watch?v=ITx32-ZfTak

The console output shows the time delta of the sync'd frames progressively increase. Yet these depth+color paired frames have the same sequence number.

```
seqnum 8 depth->color offset sec 0.0369401
seqnum 9 depth->color offset sec 0.0368821
seqnum 10 depth->color offset sec 0.0368217
seqnum 11 depth->color offset sec 0.0367634
seqnum 12 depth->color offset sec 0.0367051
seqnum 13 depth->color offset sec 0.0366456
seqnum 14 depth->color offset sec 0.0365862
...
seqnum 99 depth->color offset sec 0.00312969
seqnum 100 depth->color offset sec 0.00208995
seqnum 101 depth->color offset sec 0.00105084
seqnum 102 depth->color offset sec -1.0585e-05
seqnum 103 depth->color offset sec -0.00102366
seqnum 104 depth->color offset sec -0.0020622
seqnum 105 depth->color offset sec -0.00310604
seqnum 106 depth->color offset sec -0.00414251
seqnum 107 depth->color offset sec -0.00518231
seqnum 108 depth->color offset sec -0.00622197
seqnum 109 depth->color offset sec -0.00726578
seqnum 110 depth->color offset sec -0.00830556
seqnum 111 depth->color offset sec -0.00934533
...
seqnum 185 depth->color offset sec -0.041171
seqnum 186 depth->color offset sec -0.0412308
seqnum 187 depth->color offset sec -0.0412891
seqnum 188 depth->color offset sec -0.0413485
...
seqnum 3166 depth->color offset sec -0.217051
seqnum 3167 depth->color offset sec -0.217111
seqnum 3168 depth->color offset sec -0.21717
seqnum 3169 depth->color offset sec -0.217228
seqnum 3170 depth->color offset sec -0.217289
seqnum 3171 depth->color offset sec -0.217347
seqnum 3172 depth->color offset sec -0.217405
...
seqnum 6249 depth->color offset sec -0.398948
seqnum 6250 depth->color offset sec -0.399008
seqnum 6251 depth->color offset sec -0.399067
seqnum 6252 depth->color offset sec -0.399125
seqnum 6253 depth->color offset sec -0.399184
...
seqnum 8666 depth->color offset sec -0.54155
seqnum 8667 depth->color offset sec -0.541612
seqnum 8668 depth->color offset sec -0.541669
seqnum 8669 depth->color offset sec -0.541728
seqnum 8670 depth->color offset sec -0.541787
seqnum 8671 depth->color offset sec -0.541847
reject newseq 8689 type 1   buffer full    // this is due to my sync code only keeping 15 frames to find a sync (15/30 = 0.5 sec)
seqnum 8674 depth->color offset sec -0.542024
...
```

My 2nd OAK-D-Lite which was not calibrated does *not* have problems. Visually the frames are synchronized and the time diff shows slight oscillation arounds zero. e.g.

```
...
seqnum 28750 depth->color offset sec 0.000787798
seqnum 28752 depth->color offset sec 0.00112704
seqnum 28753 depth->color offset sec 0.0010706
seqnum 28754 depth->color offset sec 0.00101098
seqnum 28755 depth->color offset sec 0.000952764
seqnum 28756 depth->color offset sec 0.000892497
seqnum 28757 depth->color offset sec 0.00083487
seqnum 28758 depth->color offset sec 0.000774007
seqnum 28760 depth->color offset sec 0.000655067
seqnum 28761 depth->color offset sec 0.000201294
seqnum 28762 depth->color offset sec -0.000247756
seqnum 28763 depth->color offset sec -0.000704759
seqnum 28764 depth->color offset sec -0.00116026
seqnum 28765 depth->color offset sec -0.0016157
seqnum 28766 depth->color offset sec -0.00207358
seqnum 28768 depth->color offset sec -0.00298673
seqnum 28769 depth->color offset sec -0.00304437
seqnum 28770 depth->color offset sec -0.00309899
seqnum 28771 depth->color offset sec -0.00315955
seqnum 28772 depth->color offset sec -0.00321959
seqnum 28773 depth->color offset sec -0.003277
...
```

### Expected

No lag.

### Notes

* I can not reproduce this lag on my 2nd OAK-D-Lite.
* I can run the two Lite sensors side-by-side; the same code paths in memory. Only one sensor lags. See video.
* I can use the same cable and test each sensor. It is still just the one sensor that lags.
* I don't see a progressive increase in memory on the host. I do not think frames are queued in memory on the host.

Luxonis-Brandon, 2022-03-29:
Odd.  Thanks for writing up.  Asking.

tosa-no-onchan, 2022-04-04:
How about, use 1 queue buffer count.
Inside deptahi-core, use many queues and threads.

    auto qRgb =device.getOutputQueue("rgb", 1, false);
    auto qDepth =device.getOutputQueue("depth", 1, false);

    std::shared_ptr<dai::ImgFrame> rgb_sp = qRgb->get<dai::ImgFrame>();
    std::shared_ptr<dai::ImgFrame> depth_sp = qDepth->get<dai::ImgFrame>();

diablodale, 2022-04-09:
I've been thinking...in firmware and hw...what is the generator/incrementor of the sequence numbers?
If these numbers are somehow generated "virtually" by the sensors themselves, then if the output rate of multiple sensors are not exactly the same, then there will eventually be drift.

StereoDepth node...does it always wait for matching seq num frames?
Or what happens if one mono camera is 29.8 fps and the second mono is 30.0fps?
I programmatically set everything to 30fps. But....the 3A firmware can override this and change fps in certain situations.

In my repro above, there are three sensors...each perhaps slightly at a different rate and therefore will drift over time. The specific method by which sequence numbers are generated and assigned to the output of these three sensors may lead to OP behavior. Perhaps one (or two) of the mono sensors is not as light sensitive as the other, so 3A adjusts the exposure time which leads to slightly different fps.

I did not test my two OAK-D-lite sensors in this way before I calibrated one of them. So this lag (or exposure time?) may have always been there. Or perhaps the calibration code in the `lite-calibration` branch changed some setting in that one OAK-D-Lite which is exposing this lag.

Thoughts? 🤔

diablodale, 2022-04-11:
### Experiment 1

I changed my code to use `setQueueSize(1)` and `setBlocking(false)` on all node inputs (nn, imgmanp, stereodepth, xout, etc.) as https://discord.com/channels/790680891252932659/924798503270625290/961835869352886292

No improvement. ☹ One OAK-D-Lite continues to drift drastically out of sync. I also noticed...
Bad sync sensor sends seq num matched frames at almost 30fps
Correct sync sensor sends seq num matched frames at ~25.5fps
Yet both sensors run the same codepath in memory with exactly the same api calls and settings.🤪

### Experiment 2

I think 3A might be somehow involved in this seq num lag problem. To test this....
I set mono exposure and sensitivity to 28000us 1000iso, 28000us 1200iso, and 28000us 1600iso with color always `setAutoExposureEnable(); setAutoExposureLock(false)`.
All of those quickly demonstrated the sync/lag problems as in the video.

Alternatively, I set color to 28000us 1600iso with mono always `setAutoExposureEnable(); setAutoExposureLock(false)`.
🤯 No lag or sync problems with 10 minutes of testing.

### Questions

1. How does Experiment-2 fixing ***color*** camera exposure and sensitivity lead to depth frames not stamped with old sequence numbers?
2. When `StereoDepth::setDepthAlign(RECTIFIED_RIGHT)` is there any possibility that the firmware/hw needs or tracks color frames, dependent on their timing, etc.? 
3. Two mono cameras feed into StereoDepth. If one of those two mono cameras doesn't give frames to StereoDepth at the same rate as its sibling, then how does StereoDepth behave?
4. Does it wait on matching seq numbers and discard newer until it finally gets the matching older frame?
5. Does stereodepth process with a pair (one old, one new)? Will it substitute or duplicate inputs in any situation?
6. And when (and if) StereoDepth processes any pair...which Sequence Number does it stamp on the depth and disparity frames? Newest, oldest, left, right? And if the two inputs are ever not exactly the same in seqnum and time, then the stamped seq number for output will be "incorrect" for a significant portion of situations.

The video above suggests there is a 1->16 frame buffer/queue somewhere. Even though I have set queue=1/noblock, it seems that there is still a queue somewhere that is holding old frames and writing incorrect sequence numbers on them. The video doesn't seem to have dropped frames until my console errors start happening (which indicates my host-side queue is now dropping). Up until that point...the video shows the progressive lag...frames that are all there...with a progressively larger amount of time old. I don't visibly see out-of-order frames or stuttering frames.

szabi-luxonis, 2022-04-11:
AFAIK: all 3 cameras are started at the same time, RGB camera frame time is adjusted periodically to do a soft synchronization compared to stereo cameras, only if RGB FPS matches stereo FPS. Stereo cameras are started through the broadcast address and kept in sync. The FPS of the mono cameras are determined by the FPS of the first MonoCamera node (limitation in current version). (E.g. setting first to 10, the second to 40 means that both will run at 10).

> ### Questions
> 1. How does Experiment-2 fixing _**color**_ camera exposure and sensitivity lead to depth frames not stamped with old sequence numbers?

RGB frame time is adjusted by at most 2ms each time in the direction of mono timestamp (either up if lagging behind, or down if it's ahead), when the difference between RGB and mono frame is more than 70 us.
This 2 ms maximum could explain the slow/inconsistent convergence together with exposure adjustment which alters frame time.

> 2. When `StereoDepth::setDepthAlign(RECTIFIED_RIGHT)` is there any possibility that the firmware/hw needs or tracks color frames, dependent on their timing, etc.?

No. Stereo depth alignment doesn't depend at all on RGB camera, it depends only on RGB calibration, if RGB alignment is enabled.

> 3. Two mono cameras feed into StereoDepth. If one of those two mono cameras doesn't give frames to StereoDepth at the same rate as its sibling, then how does StereoDepth behave?

They are forced to the same FPS as described above.

> 4. Does it wait on matching seq numbers and discard newer until it finally gets the matching older frame?

There's an internal sync node based on timestamps, not sequence numbers.

> 5. Does stereodepth process with a pair (one old, one new)? Will it substitute or duplicate inputs in any situation?

No, it always check left and right timestamp, if the difference between them is less than 16.666 ms (hardcoded in current implementation) then it's considered synced.

> 6. And when (and if) StereoDepth processes any pair...which Sequence Number does it stamp on the depth and disparity frames? Newest, oldest, left, right? And if the two inputs are ever not exactly [...]

If it's aligned to CENTER/RIGHT then right frame, else left frame. In case of RGB alignment it doesn't inherit the sequence number of RGB frame AFAIK.

@alex-luxonis can correct me on any of the above questions.

diablodale, 2022-06-13:
Hi. My Lite sensors continue to be problematic, and behave significantly different than all my other OAK sensors.
I can easily repro this time/lag issue with my app.
This seems to be something complex or hw so I've been trying to simplify....now I have something. 🧐

I've modified `gen2-syncing\host-frame-sync.py` at https://github.com/diablodale/depthai-experiments/tree/repro-440 to reproduce a failure where the fps drops from its full rate almost 30fps down to ~3 fps.
This failure occurs 80% of the time on my OAK-D-Lite sensors.
It never occurs on my OAK-D.

Here is a video of the failure. The stream of right-mono frames starts ok for a few seconds then quickly reduces to ~3fps and never recovers.

Luxonis-Brandon, 2022-06-13:
Thanks for the heads up on this and detailed report.  I"m not sure if you intended to have a video linked as well.  Just mentioning in case that was missed.

@alex-luxonis - do we need to enable hardware sync on OV7251 maybe?

diablodale, 2022-06-14:
Oops, added video link to above post.

Luxonis-Brandon, 2022-06-15:
Thanks.  We're fighting the good fight on some other underlying things (namely some firmware flashing regressions) but we will be digging into this ASAP.

jasonm1892, 2022-11-11:
Any update on this @alex-luxonis ? `multi_cam_support` ?