luxonis/depthai-core #445: Syntax Error in variant.h

Monopolymer, 2022-04-06:
Hello,
I am Trying to compile the rgb_preview.cpp on VS2019 I am getting a syntax error unexpected Token "..." expected :) in variant.h line 294 and line 301

detail::Set<Types...>::NOP_TEMPLATE IsSubset<ValidTypes...>::value,

I hope someone can help

Best regards
Robert
[image]

1>G:\depthai-core-v2.15.1-win64\lib\cmake\depthai\dependencies\include\nop\types\variant.h(294,64): error C2760: Syntaxfehler: Unerwartetes Token "...". Erwartet: )
1>G:\depthai-core-v2.15.1-win64\lib\cmake\depthai\dependencies\include\nop\types\variant.h(292,15): message : Diese Diagnose trat in der vom Compiler generierten Funktion "bool nop::IfAnyOf<ValidTypes...>::Call(nop::Variant<Types...> *,Op &&)" auf.
1>G:\depthai-core-v2.15.1-win64\lib\cmake\depthai\dependencies\include\nop\types\variant.h(351): message : Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "nop::IfAnyOf<ValidTypes...>".
1>G:\depthai-core-v2.15.1-win64\lib\cmake\depthai\dependencies\include\nop\types\variant.h(301,64): error C2760: Syntaxfehler: Unerwartetes Token "...". Erwartet: )
1>G:\depthai-core-v2.15.1-win64\lib\cmake\depthai\dependencies\include\nop\types\variant.h(299,15): message : Diese Diagnose trat in der vom Compiler generierten Funktion "bool nop::IfAnyOf<ValidTypes...>::Call(const nop::Variant<Types...> *,Op &&)" auf.

themarpe, 2022-04-06:
Hi @Monopolymer 

Make sure you have the correct C++ standard set (C++14) on your project.

Monopolymer, 2022-04-06:
HI @themarpe I do Have C++14 set on my project
[image]

themarpe, 2022-04-06:
@Monopolymer 

Can you compile `rgb_preview.cpp` from `depthai-core examples (using the provided CMake file) (See Building Examples) successfully?

Monopolymer, 2022-04-06:
@themarpe I will try, i was using depthai-core-v2.15.1-win64 i have not tried to compile from source

themarpe, 2022-04-07:
@Monopolymer 

Seems like its connected to `NOP_TEMPLATE` and specifying a `template ...` keyword in front of calling such functions.

I'll recheck on when this define can be set for MSVC

Monopolymer, 2022-04-08:
@themarpe I have tried to build using the provided Cmake file, I seem to be doing something wrong
[image]

themarpe, 2022-04-08:
@Monopolymer 
Which version of opencv are you linking against?

Can you post the whole CMake output when configuring and after you build as a file / codeblock.

Also, please rather copy text instead of doing screenshots as its easier to search for and use in debugging (eg. to search & translate the error messages, etc...)

Monopolymer, 2022-04-09:
@themarpe
I am using Opencv 4.5.5 
[Output.txt](https://github.com/luxonis/depthai-core/files/8456359/Output.txt)

themarpe, 2022-04-09:
@Monopolymer 
Can you try by pointing DepthAI to this opencv https://github.com/opencv/opencv/releases/download/4.5.5/opencv-4.5.5-vc14_vc15.exe library? (using the following argument added to cmake `-D OpenCV_DIR=[path/to/installed/opencv]`)

Monopolymer, 2022-04-11:
@themarpe it builds using that library.
When running the examples I get opencv_world455d.dll missing.
After adding the missing DLLs the example starts to run but I get runtime errors
[Exampleserror.txt](https://github.com/luxonis/depthai-core/files/8466767/Exampleserror.txt)

themarpe, 2022-04-11:
Thats odd - just to double check, have you set the required defines: https://github.com/luxonis/depthai-core#non-cmake-integration-visual-studio-xcode-codeblocks- ?

Monopolymer, 2022-04-12:
@themarpe 
i was just running the examples that are build with -D'DEPTHAI_BUILD_EXAMPLES=ON' I havent tried running my own Project.

Monopolymer, 2022-04-12:
This makes me wonder how I would go about compiling it so I dont need opencv_world455d.dll. because I usually dont use it in my OpenCV Projects and I have my Opencv wich I use with Cuda. The Opencv Library you pointed me to doesnt seem to work with my own OpenCV Projects. I will Compile the Opencv Debug Librarys and will try to Compile Depthai-core again.
I would prefer to just use depthai-core-v2.15.1-win64.

Monopolymer, 2022-04-12:
I got it working it seems I had another issue with one of my USB Ports. So I guess this can be closed.

themarpe, 2022-04-14:
@Monopolymer 

Can you post more information about your system. Given the initial error discovering the device (`[2022-04-11 20:19:25.057] [warning] skipping X_LINK_UNBOOTED device having name "<error>"`) I think there might be some subtle differences system wise that might be causing this.

Monopolymer, 2022-04-16:
@themarpe I Solved the USB Issue by not using the Front USB. 
With The Self Compiled DepthAI-Core Library i still get the Syntax Error in variant.h.

themarpe, 2022-04-16:
Great, thanks for the heads up @Monopolymer 

Regarding syntax error, use branch `develop`, where this issue was addressed (a release will be made soon)

Monopolymer, 2022-04-17:
Just for anyone haveing he same issue, NOP_TEMPLATE  was defined as nothing "#define NOP_TEMPLATE", adding the Keyword Template fixed it. "#define NOP_TEMPLATE template"

closed 2022-04-17 by Monopolymer

themarpe, 2022-04-17:
@Monopolymer 
Was this required on `develop` branch for you? Also, NOP_TEMPLATE is defined explicitly before used in the variant.h header in libnop, which should override any previous definition.

Monopolymer, 2022-04-19:
@themarpe I did not try the develop branch, but i will as sonn as I can. I fixed it for me in the latest release i was using. I addet Template to the first define of NOP_Template in Compiler.h.

 * Copyright 2017 The Native Object Protocols Authors
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *      http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

#ifndef LIBNOP_INCLUDE_NOP_UTILITY_COMPILER_H_
#define LIBNOP_INCLUDE_NOP_UTILITY_COMPILER_H_

#define NOP_GCC_VERSION \
  (__GNUC__ * 10000 + __GNUC_MINOR__ * 100 + __GNUC_PATCHLEVEL__)

// Compatability with non-clang compilers.
#ifndef __has_cpp_attribute
#define __has_cpp_attribute(x) 0
#endif

#ifdef _MSC_VER

// MSVC case
#define NOP_FALLTHROUGH
#define NOP_GNU_USED
#define NOP_TEMPLATE template

#else

#define NOP_GNU_USED [[gnu::used]]
// Test for fallthrough support.
#if __has_cpp_attribute(clang::fallthrough)
#define NOP_FALLTHROUGH [[clang::fallthrough]]
#elif NOP_GCC_VERSION >= 70000
#define NOP_FALLTHROUGH [[fallthrough]]
#else
#define NOP_FALLTHROUGH
#endif
#define NOP_TEMPLATE template

#endif

#endif  // LIBNOP_INCLUDE_NOP_UTILITY_COMPILER_H_

themarpe, 2022-04-20:
@Monopolymer 
Ohh got it. I think that the `develop` (and the upcoming release) should address this without any required changes

PR #448 (2022-04-11): Updated libnop 
Fixes #445 #407 

Updated libnop with C++20 fixes (MSVC) and added filesystem test targeting C++20