luxonis/depthai-core #447: Error during build using Yocto

oto313, 2022-04-07:
I am using Yocto build system to integrate depthai into linux. It fails with error

The first command that fails:
```
FAILED: CMakeFiles/depthai-core.dir/src/device/Device.cpp.o
/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++ -DBACKWARD_HAS_BACKTRACE=0 -DBACKWARD_HAS_BACKTRACE_SYMBOL=1 -DBACKWARD_HAS_BFD=0 -DBACKWARD_HAS_DW=0 -DBACKWARD_HAS_DWARF=0 -DBACKWARD_HAS_LIBUNWIND=0 -DBACKWARD_HAS_UNWIND=1 -DDEPTHAI_BOOTLOADER_VERSION=\"0.0.17\" -DDEPTHAI_DEVICE_VERSION=\"afe8d00c868344c73ee897e26278f80a444b9763\" -DDEPTHAI_ENABLE_BACKWARD -DDEPTHAI_PATCH_ONLY_MODE -DDEPTHAI_RESOURCE_COMPILED_BINARIES -DDEPTHAI_TARGET_CORE -DJSON_USE_IMPLICIT_CONVERSIONS=1 -DLIBARCHIVE_STATIC -DSPDLOG_COMPILED_LIB -DXLINK_USE_MX_ID_NAME=ON -D__PC__ -I/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/include -I/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/shared/depthai-shared/include -I/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/shared/depthai-bootloader-shared/include -I/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/include/depthai -I/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/src -I/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/shared/depthai-shared/src -I/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/shared/depthai-bootloader-shared/src -I/home/ysoft-ci/.hunter/_Base/cb0ea1f/32ad177/c6f42ce/Install/include -I/home/ysoft-ci/.hunter/_Base/cb0ea1f/32ad177/c6f42ce/Install/lib/backward -isystem /base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/shared/depthai-shared/3rdparty -isystem /base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/build/_cmrc/include -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time  --sysroot=/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0=/usr/src/debug/depthai/2.15.2-r0                      -fdebug-prefix-map=/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0=/usr/src/debug/depthai/2.15.2-r0                      -fdebug-prefix-map=/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/recipe-sysroot=                      -fdebug-prefix-map=/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time  --sysroot=/base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/recipe-sysroot -g -Wall -Wextra -Woverloaded-virtual -Wformat=2 -Wmisleading-indentation -Wduplicated-cond -Wduplicated-branches -Wnull-dereference -Wdouble-promotion -Wsign-compare -Wtype-limits -Werror=return-type -Werror=non-virtual-dtor -Werror=sign-compare -Werror=reorder -Werror=switch-enum -std=c++14 -MD -MT CMakeFiles/depthai-core.dir/src/device/Device.cpp.o -MF CMakeFiles/depthai-core.dir/src/device/Device.cpp.o.d -o CMakeFiles/depthai-core.dir/src/device/Device.cpp.o -c /base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/src/device/Device.cpp
```

Error:
```
/home/ysoft-ci/.hunter/_Base/cb0ea1f/32ad177/c6f42ce/Install/include/nlohmann/json.hpp:21496:9: error: enumeration value 'string' not handled in switch [-Werror=switch-enum]
21496 |         switch (m_type)
      |         ^~~~~~
/home/ysoft-ci/.hunter/_Base/cb0ea1f/32ad177/c6f42ce/Install/include/nlohmann/json.hpp:21496:9: error: enumeration value 'boolean' not handled in switch [-Werror=switch-enum]
/home/ysoft-ci/.hunter/_Base/cb0ea1f/32ad177/c6f42ce/Install/include/nlohmann/json.hpp:21496:9: error: enumeration value 'number_integer' not handled in switch [-Werror=switch-enum]
/home/ysoft-ci/.hunter/_Base/cb0ea1f/32ad177/c6f42ce/Install/include/nlohmann/json.hpp:21496:9: error: enumeration value 'number_unsigned' not handled in switch [-Werror=switch-enum]
/home/ysoft-ci/.hunter/_Base/cb0ea1f/32ad177/c6f42ce/Install/include/nlohmann/json.hpp:21496:9: error: enumeration value 'number_float' not handled in switch [-Werror=switch-enum]
/home/ysoft-ci/.hunter/_Base/cb0ea1f/32ad177/c6f42ce/Install/include/nlohmann/json.hpp:21496:9: error: enumeration value 'binary' not handled in switch [-Werror=switch-enum]
/home/ysoft-ci/.hunter/_Base/cb0ea1f/32ad177/c6f42ce/Install/include/nlohmann/json.hpp:21496:9: error: enumeration value 'discarded' not handled in switch [-Werror=switch-enum]
In file included from /base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/shared/depthai-shared/include/depthai-shared/utility/Serialization.hpp:14,
                 from /base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/shared/depthai-shared/include/depthai-shared/datatype/RawBuffer.hpp:6,
                 from /base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/include/depthai/pipeline/datatype/ADatatype.hpp:6,
                 from /base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/include/depthai/device/DataQueue.hpp:9,
                 from /base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/include/depthai/device/Device.hpp:13,
                 from /base/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/depthai/2.15.2-r0/git/src/device/Device.cpp:1:

```

It looks like that you are setting flag compiler -Werror=switch-enum . But these flags are used also in 3rdparty libraries which are used through header files.

I tried also build it on Ubuntu and it works. But I do not understand why it works on native linux. When I look into included header files (eg. json.hpp) I see that not all cases in switch are handled. So it should definitely fail.

for example here at json.hpp
```

  21494     size_type size() const noexcept
  21495     {
  21496         switch (m_type)
  21497         {
  21498             case value_t::null:
  21499             {
  21500                 // null values are empty
  21501                 return 0;
  21502             }
  21503
  21504             case value_t::array:
  21505             {
  21506                 // delegate call to array_t::size()
  21507                 return m_value.array->size();
  21508             }
  21509
  21510             case value_t::object:
  21511             {
  21512                 // delegate call to object_t::size()
  21513                 return m_value.object->size();
  21514             }
  21515
  21516             default:
  21517             {
  21518                 // all other types have size 1
  21519                 return 1;
  21520             }
  21521         }
  21522     }
```
enum definition at value_t.hpp
```
enum class value_t : std::uint8_t
{
    null,             ///< null value
    object,           ///< object (unordered set of name/value pairs)
    array,            ///< array (ordered collection of values)
    string,           ///< string value
    boolean,          ///< boolean value
    number_integer,   ///< number value (signed integer)
    number_unsigned,  ///< number value (unsigned integer)
    number_float,     ///< number value (floating-point)
    binary,           ///< binary array (ordered collection of bytes)
    discarded         ///< discarded by the parser callback function
};
```

So I propose to remove -Werror=switch-enum flag from cmake/Flags.cmake
When I remove this flag it works as expected.

Complete error output of command:
[error.txt](https://github.com/luxonis/depthai-core/files/8441561/error.txt)

themarpe, 2022-04-07:
Hi @oto313 - do you mind checking why there is such a difference in this two scenarios?

Also, what version of GCC are you using to compile the sources?

We'll have to check how this could be mitigated, as this flag is quite useful in our codebase

oto313, 2022-04-07:
I also do not understand why there is such difference. But in Yocto it make sense that it turn switch warnings into errors. Maybe some fix in a newer GCC

Compiler: GCC 11.2

themarpe, 2022-04-07:
@oto313 

Temporary I'd then suggest adding a patch file and according bb layer to build without issues.

Will have to dig more to resolve this.

oto313, 2022-04-07:
Ok Great to hear that you plan to address this issue. Thanks

windelbouwman, 2022-04-08:
Hi @oto313 ,

Glad somebody else tries to package depthai-core in yocto. I made some bitbake recipes for this, and faced a couple of issues. Step one for me was to disable the hunter package manager.

See also this issue: #411 .

Is there already a layer of some sort? Or a repository for yocto bitbake recipes?

themarpe, 2022-04-08:
For Yocto, I'd just compile & install depthai-core as usual (into some working directory), and then specify the rules in layer to install that folder to sysroot (eg `/usr` or `/usr/local`, ...)

That should work as intended

themarpe, 2022-04-08:
@oto313 

I assume that the difference could be in `-isystem` vs `-I` means of specifying include directories. Hunter dependencies are under `-isystem` which might be the reason it doesn't apply the flag in some cases.

Maybe check if that is so in this case as well (verbose build and copy back the whole build command that is executed)

oto313, 2022-04-10:
@windelbouwman  I successfully built depthai after apply patch to Flags.cmake. I can share recipe:

```
SUMMARY = "DepthAI API allows users to connect to, configure and communicate with their OAK devices"
HOMEPAGE = "https://github.com/luxonis/depthai-core"
LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://LICENSE;md5=39a6484daa66eaf84c7ee9c45e34db02"
DEPENDS = "libusb1"

SRC_URI = "gitsm://github.com/luxonis/depthai-core.git;nobranch=1;tag=v2.15.2;protocol=git \
           file://fix-enum-error.patch \
           "

S = "${WORKDIR}/git"

inherit cmake pkgconfig

OECMAKE_C_FLAGS += " -fPIC "
OECMAKE_CXX_FLAGS += " -fPIC "

PACKAGECONFIG ?= ""
PACKAGECONFIG[opencv] = "-DDEPTHAI_OPENCV_SUPPORT=TRUE, -DDEPTHAI_OPENCV_SUPPORT=FALSE, opencv"

PACKAGES = "${PN} ${PN}-dev ${PN}-staticdev"

FILES:${PN}-staticdev = "${libdir}/lib*.a ${libdir}/cmake/depthai"
FILES:${PN}-dev = "${includedir}/* ${libdir}/cmake/depthai/*.cmake"

```
Note that I am not Yocto expert. Somebody maybe could improve the recipe. The fPIC switch is required on ARM because some static library needs to be build with relocatable code.

windelbouwman, 2022-04-11:
Below is my attempted bitbake recipe. Couple of notes:

- the "fwp" resources required are downloaded using bitbake instead of cmake, this allows caching of these downloads from the artifactory server
- hunter is disabled, I used packages existing in the yocto environment.
- I disabled clang/tests/docs/examples to improve compilation speed
- the .so file has no version, this gives an error in bitbake when splitting packages into -dev versions and other packages.
- I created seperate recipes for xlink, libnop and fp16, since those packages are custom forks, and no recipes exist already to re-use.

```

SUMMARY = "DepthAI C++ Library"
HOMEPAGE = "https://github.com/luxonis/depthai-core"
LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://LICENSE;md5=39a6484daa66eaf84c7ee9c45e34db02"

# These resources are required as resources, note that these are very coupled to the version of depthai-core:
# See: cmake/Depthai/DepthaiBootloaderConfig.cmake
#   depthai bootloader: release-0.0.17
#   https://artifacts.luxonis.com/artifactory/luxonis-myriad-release-local/depthai-bootloader/0.0.17/depthai-bootloader-fwp-0.0.17.tar.xz
# See: cmake/Depthai/DepthaiDownloader.cmake
#   depthai device side
#   https://artifacts.luxonis.com/artifactory/luxonis-myriad-snapshot-local/depthai-device-side/afe8d00c868344c73ee897e26278f80a444b9763/depthai-device-fwp-afe8d00c868344c73ee897e26278f80a444b9763.tar.xz

SRC_URI = "https://github.com/luxonis/${BPN}/releases/download/v${PV}/${BPN}-v${PV}.tar.gz;name=src \
           https://artifacts.luxonis.com/artifactory/luxonis-myriad-release-local/depthai-bootloader/0.0.17/depthai-bootloader-fwp-0.0.17.tar.xz;unpack=0;name=bootloader-fwp \
           https://artifacts.luxonis.com/artifactory/luxonis-myriad-snapshot-local/depthai-device-side/afe8d00c868344c73ee897e26278f80a444b9763/depthai-device-fwp-afe8d00c868344c73ee897e26278f80a444b9763.tar.xz;unpack=0;name=device-fwp \
           file://80-movidius.rules \
           file://0001-Adapt-cmake-configuration-to-be-compatible-with-open.patch \
           file://0002-Remove-downloading-from-CMakeLists.txt.patch \
           "

# These three downloads are a coupled set, when updating the src, update the other blobs as well!
SRC_URI[src.sha256sum] = "dea755d06ed19a1ea57cf6474a80c47302d7edd27060234447c17a93b9602877"
SRC_URI[bootloader-fwp.sha256sum] = "27ea0d797498f7fff1d82e9524ea6a244a5998424b523eab9b353944548bcbef"
SRC_URI[device-fwp.sha256sum] = "5cc6e63d5c10430c3cc73f74278c928a39d36b860160403962e898e63ef96046"

S = "${WORKDIR}/${BPN}-v${PV}"

inherit cmake

# DEPTHAI_BINARIES_RESOURCE_COMPILE=OFF should disable the inclusion of bootloader and application as a resource.
# HUNTER_ENABLED=OFF disables the hunter C++ package manager (since we use bitbake)
# DEPTHAI_ENABLE_BACKWARD=OFF disables the use of a fancy C++ backtrace printing library called 'backward'
# BUILD_SHARED_LIBS=ON ensures we build shared libraries
# using DEPTHAI_BOOTLOADER_SHARED_LOCAL prevents GIT submodule initialization
EXTRA_OECMAKE:append = " \
    -DHUNTER_ENABLED=OFF \
    -DCMAKE_BUILD_TYPE=Release \
    -DDEPTHAI_CLANG_FORMAT=OFF \
    -DDEPTHAI_CLANG_TIDY=OFF \
    -DDEPTHAI_ENABLE_BACKWARD=OFF \
    -DDEPTHAI_BINARIES_RESOURCE_COMPILE=ON \
    -DBUILD_SHARED_LIBS=ON \
    -DDEPTHAI_BOOTLOADER_SHARED_LOCAL=${S}/shared/depthai-bootloader-shared \
    -DDEPTHAI_SHARED_LOCAL=${S}/shared/depthai-shared \
    -DDEPTHAI_BUILD_EXAMPLES=OFF \
    -DDEPTHAI_BUILD_TESTS=OFF \
    -DDEPTHAI_BUILD_DOCS=OFF \
    -DDEPTHAI_BOOTLOADER_FWP=${WORKDIR}/depthai-bootloader-fwp-0.0.17.tar.xz \
    -DDEPTHAI_DEVICE_FWP=${WORKDIR}/depthai-device-fwp-afe8d00c868344c73ee897e26278f80a444b9763.tar.xz \
    "

DEPENDS = " \
    bzip2 \
    spdlog \
    libarchive \
    zlib \
    xz \
    nlohmann-json \
    luxonis-xlink \
    luxonis-libnop \
    luxonis-fp16 \
"

do_install:append() {
    # These permissions are required, otherwise the OAK-D device will report as X_LINK_UNBOOTED
    install -Dm 0644 ${WORKDIR}/80-movidius.rules ${D}${sysconfdir}/udev/rules.d/80-movidius.rules
}

# -dev package depthai-core-dev contains non-symlink .so '/usr/lib/libdepthai-core.so' [dev-elf]
FILES_SOLIBSDEV = ""
FILES:${PN} += " \
    ${libdir}/lib*${SOLIBSDEV} \
    ${sysconfdir}/udev/rules.d/80-movidius.rules \
"
```

windelbouwman, 2022-06-23:
Sure, this is my set of recipes.

It would be nice to store these in some public repo. Another note is that many patches done here should be somehow included in the depthai source code, such that the recipes become simpler.

Another note is that the git hashes must be manually determined by checking the hunter dependencies file, where specific commits are referred. This is a bit tricky.

## depthai-core_2.15.2.bb

```

SUMMARY = "DepthAI C++ Library"
HOMEPAGE = "https://github.com/luxonis/depthai-core"
LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://LICENSE;md5=39a6484daa66eaf84c7ee9c45e34db02"

# These resources are required as resources, note that these are very coupled to the version of depthai-core:
# See: cmake/Depthai/DepthaiBootloaderConfig.cmake
#   depthai bootloader: release-0.0.17
#   https://artifacts.luxonis.com/artifactory/luxonis-myriad-release-local/depthai-bootloader/0.0.17/depthai-bootloader-fwp-0.0.17.tar.xz
# See: cmake/Depthai/DepthaiDownloader.cmake
#   depthai device side
#   https://artifacts.luxonis.com/artifactory/luxonis-myriad-snapshot-local/depthai-device-side/afe8d00c868344c73ee897e26278f80a444b9763/depthai-device-fwp-afe8d00c868344c73ee897e26278f80a444b9763.tar.xz

SRC_URI = "https://github.com/luxonis/${BPN}/releases/download/v${PV}/${BPN}-v${PV}.tar.gz;name=src \
           https://artifacts.luxonis.com/artifactory/luxonis-myriad-release-local/depthai-bootloader/0.0.17/depthai-bootloader-fwp-0.0.17.tar.xz;unpack=0;name=bootloader-fwp \
           https://artifacts.luxonis.com/artifactory/luxonis-myriad-snapshot-local/depthai-device-side/afe8d00c868344c73ee897e26278f80a444b9763/depthai-device-fwp-afe8d00c868344c73ee897e26278f80a444b9763.tar.xz;unpack=0;name=device-fwp \
           file://80-movidius.rules \
           file://0001-Adapt-cmake-configuration-to-be-compatible-with-open.patch \
           file://0002-Remove-downloading-from-CMakeLists.txt.patch \
           "

# These three downloads are a coupled set, when updating the src, update the other blobs as well!
SRC_URI[src.sha256sum] = "dea755d06ed19a1ea57cf6474a80c47302d7edd27060234447c17a93b9602877"
SRC_URI[bootloader-fwp.sha256sum] = "27ea0d797498f7fff1d82e9524ea6a244a5998424b523eab9b353944548bcbef"
SRC_URI[device-fwp.sha256sum] = "5cc6e63d5c10430c3cc73f74278c928a39d36b860160403962e898e63ef96046"

S = "${WORKDIR}/${BPN}-v${PV}"

inherit cmake

# DEPTHAI_BINARIES_RESOURCE_COMPILE=OFF should disable the inclusion of bootloader and application as a resource.
# HUNTER_ENABLED=OFF disables the hunter C++ package manager (since we use bitbake)
# DEPTHAI_ENABLE_BACKWARD=OFF disables the use of a fancy C++ backtrace printing library called 'backward'
# BUILD_SHARED_LIBS=ON ensures we build shared libraries
# using DEPTHAI_BOOTLOADER_SHARED_LOCAL prevents GIT submodule initialization
EXTRA_OECMAKE:append = " \
    -DHUNTER_ENABLED=OFF \
    -DCMAKE_BUILD_TYPE=Release \
    -DDEPTHAI_CLANG_FORMAT=OFF \
    -DDEPTHAI_CLANG_TIDY=OFF \
    -DDEPTHAI_ENABLE_BACKWARD=OFF \
    -DDEPTHAI_BINARIES_RESOURCE_COMPILE=ON \
    -DBUILD_SHARED_LIBS=ON \
    -DDEPTHAI_BOOTLOADER_SHARED_LOCAL=${S}/shared/depthai-bootloader-shared \
    -DDEPTHAI_SHARED_LOCAL=${S}/shared/depthai-shared \
    -DDEPTHAI_BUILD_EXAMPLES=OFF \
    -DDEPTHAI_BUILD_TESTS=OFF \
    -DDEPTHAI_BUILD_DOCS=OFF \
    -DDEPTHAI_BOOTLOADER_FWP=${WORKDIR}/depthai-bootloader-fwp-0.0.17.tar.xz \
    -DDEPTHAI_DEVICE_FWP=${WORKDIR}/depthai-device-fwp-afe8d00c868344c73ee897e26278f80a444b9763.tar.xz \
    "

DEPENDS = " \
    bzip2 \
    spdlog \
    libarchive \
    zlib \
    xz \
    nlohmann-json \
    luxonis-xlink \
[... 422 lines cut ...]
 #This is for the Movidius team
-set(XLINK_INCLUDE_DIRECTORIES ${XLINK_INCLUDE} ${LIBUSB_INCLUDE_DIR})
+set(XLINK_INCLUDE_DIRECTORIES ${XLINK_INCLUDE})

```

## luxonis-libnop_git.bb

```

SUMMARY = "libnop: C++ Native Object Protocols"
HOMEPAGE = "https://github.com/luxonis/libnop"

LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENSE;md5=4e5640559f0849cf863b1dc47bac078b"

# See: https://github.com/luxonis/depthai-core/blob/v2.15.2/cmake/Hunter/config.cmake
SRC_URI = "git://github.com/luxonis/libnop.git;branch=develop"
SRCREV = "ec8f75aa4df3454f73b5d7a7fe0680f3701b1744"

S = "${WORKDIR}/git"

inherit cmake

```

bartoszptak, 2022-11-10:
Hi @windelbouwman ! Thank you for your code, it works well on VCK190 with Petalinux. Is the code anywhere available as a GitHub meta-repository? If not, can I create one with reference to your comment?