luxonis/depthai-core #460: Wrong spatial  location calculator output when rgb alignment is enabled.

developer-mayuan, 2022-04-23:
I just found a potential bug that spatial location calculator output when rgb alignment is enabled is wrong as shown in the picture below:
[image]

* Setup
Ubuntu 18.04
depthai-core develop branch with commit 7b60935b67299fae7fbf79c5bd3854d5cc74b8b6

* Reproduce
1. Build depthai-core with commit 7b60935b67299fae7fbf79c5bd3854d5cc74b8b6 using command below:
```
cmake -S. -Bbuild -D'DEPTHAI_BUILD_EXAMPLES=ON'
cmake --build build
```
2. Run spatial_mobilnet example:
```
./build/examples/spatial_mobilenet
```

* Expect result
The correct spatial location of detected object is generated and the roi region in depth frame is correct.

* Actual output
Wrong roi regions of detected objects are generated from spatial detection network and visualized in the depth image.

My diagnose of this issue is that with rgb alignment enabled, the depth image was re-scaled to rgb ISP image size after alignment, however the bounding box (ROI) generated from the preview rgb image are just mapped directly to the aligned depth image without modifying the bounding box coordinates which causes the spatial calculator generating wrong result.

szabi-luxonis, 2022-04-24:
@developer-mayuan which board are you using? There's no such issue on OAK-D or OAK-D Lite.
Did you modify the example before running?

developer-mayuan, 2022-04-24:
HI @szabi-luxonis I'm using OAK-D for running this test and I didn't modify the example, just checkout the commit and did a fresh build.

szabi-luxonis, 2022-04-25:
@developer-mayuan can you share your calibration info? Could be an older version that you are using.
Run the following example: https://github.com/luxonis/depthai-python/blob/main/examples/calibration/calibration_reader.py
and share the output `.json` file.

developer-mayuan, 2022-04-25:
@szabi-luxonis It seems like this is indeed a calibration issue since I found one of my oak-d camera doesn't have this issue. But other boards come with the error out of box. Here is a sample calibration info I got from one of oak-d camera having this issue:
```
[18443010E1DF1AF500] [60.719] [system] [info] PRINT:LeonCss: bss_init took 40.417 ms
BootloaderConfig.options1 checksum doesn't match. Is: 0x10304418 should be: 0xEB4AFD53
BootloaderConfig.options2 checksum doesn't match. Is: 0x00000000 should be: 0x2380F205
GPIO boot mode 0x16, interface USBD
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
--> brdInit ...
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: ff ff ff
eeprom_read_status 0!
Reading VERSION 6 --- ->
Failed to probe IR driver LM3644
Failed to probe IR driver LM3644
Failed to probe IR driver LM3644
Closing EEPROm!
Is booted from flash by bootloader: 0
Networking not available...
Called by: LOS, controller: LOS
Enumerating on socket: Cam_A / RGB / Center
Probe failed I2C1 0x1a reg 0016: expected 0214, read 0378
Probe failed I2C1 0x1a reg 0016: expected 0214, read 0378
  >> Registered camera A12N02A (imx378) as /dev/Camera_0
Enumerating on socket: Cam_B / Left
  >> Registered camera TG161B (ov9282) as /dev/Camera_1
Enumerating on socket: Cam_C / Right
  >> Registered camera TG161B (ov9282) as /dev/Camera_2
Enumerating on socket: CAM_D
initial keepalive, countdown: 9
Initializing XLink...
UsbPumpVscAppI_Event: 5 VSC2_EVENT_ATTACH
initial keepalive, countdown: 8
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 0 VSC2_EVENT_INTERFACE_UP
Done!
Usb connection speed: Super - USB 3.0
I: [Timesync] [   1905056] [main] startSync:116     Timesync | Callback not set
Temperature: Driver registered.
Temperature: Initialized driver.
Temperature: Sensor opened: CSS.
Temperature: Sensor opened: MSS.
Temperature: Sensor opened: UPA.
Temperature: Sensor opened: DSS.
[18443010E1DF1AF500] [60.719] [system] [info] PRINT:LeonMss: Called by: LRT, controller: LOS
RGB Camera Default intrinsics...
[[3110.83837890625, 0.0, 1921.591064453125], [0.0, 3110.83837890625, 1096.3291015625], [0.0, 0.0, 1.0]]
3840
2160
LEFT Camera resized intrinsics...
[[802.74401855   0.         644.60253906]
 [  0.         802.74401855 378.73175049]
 [  0.           0.           1.        ]]
LEFT Distortion Coefficients...
k1: -11.881447792053223
k2: 119.9446029663086
p1: 0.0002003137196879834
p2: -0.0004118149518035352
k3: -70.5708236694336
k4: -11.907358169555664
k5: 119.2244644165039
k6: -68.99750518798828
s1: 0.0
s2: 0.0
s3: 0.0
s4: 0.0
τx: 0.0
τy: 0.0
RIGHT Camera resized intrinsics...
[[805.98931885   0.         616.02502441]
 [  0.         805.98931885 356.56661987]
 [  0.           0.           1.        ]]
RIGHT Distortion Coefficients...
k1: -8.983304977416992
k2: 94.06720733642578
p1: 0.0005835810443386436
p2: 0.0006875866674818099
k3: -56.06025695800781
k4: -9.04328727722168
k5: 93.61856842041016
k6: -54.901123046875
s1: 0.0
s2: 0.0
s3: 0.0
s4: 0.0
τx: 0.0
τy: 0.0
RGB FOV 68.7938003540039, Mono FOV 71.86000061035156
LEFT Camera stereo rectification matrix...
[[ 1.00436654e+00  7.18781715e-03 -3.44817361e+01]
 [-6.62840836e-03  1.00422704e+00 -1.98676893e+01]
 [ 5.63491554e-07  5.82279554e-07  9.99416006e-01]]
RIGHT Camera stereo rectification matrix...
[[ 1.00032248e+00  7.15887554e-03 -3.11855436e+00]
 [-6.60171920e-03  1.00018354e+00  3.62704692e+00]
 [ 5.61222666e-07  5.79935016e-07  9.99447249e-01]]
Transformation matrix of where left Camera is W.R.T right Camera's optical center
[[ 9.99966621e-01  7.99766742e-03  1.66361686e-03 -7.52113533e+00]
 [-7.99612608e-03  9.99967575e-01 -9.30935843e-04  8.98113288e-03]
 [-1.67100830e-03  9.17602272e-04  9.99998212e-01  1.59466267e-02]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]
Transformation matrix of where left Camera is W.R.T RGB Camera's optical center
[[ 9.99976695e-01  5.96394110e-03 -3.31831700e-03 -3.76314974e+00]
 [-5.95916621e-03  9.99981165e-01  1.44668622e-03 -1.32785551e-02]
 [ 3.32688261e-03 -1.42687815e-03  9.99993443e-01 -2.45756328e-01]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]
```

saching13, 2022-04-25:
@developer-mayuan  
- Can you share the calibration info from the one that doesn't have issue ?
- Are all the devices were bought around the same time ?

developer-mayuan, 2022-04-25:
I investigated this issues more and found cameras with calibration info with EEPROM data version 5 work fine, but cameras with EEPROM data version 6 have this issue.

Here is the calibration info without this issue:

szabi-luxonis, 2022-04-25:
@developer-mayuan check out 1ee7cedee3899ecc670ce94944ced8928474da8c it was an RGB alignment related issue when configured RGB resolution (1080p) was different from calibrated one (4k in your case)

developer-mayuan, 2022-04-25:
Thanks for the quick fix! I can confirm that by checking out https://github.com/luxonis/depthai-core/commit/1ee7cedee3899ecc670ce94944ced8928474da8c and doing a fresh build, I cannot reproduce this issue anymore using my current oak-d cameras. Great work!

closed 2022-04-26 by szabi-luxonis