luxonis/depthai-core #462: Fatal Error when running depth_post_processing.cpp

Monopolymer, 2022-04-25:
[18443010E144E40F00] [132.805] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 2A, address: 801B1C4C' '0'
F: [global] [         0] [Scheduler00Thr] dispatcherResponseServe:925   no request for this response: XLINK_WRITE_RESP 1

szabi-luxonis, 2022-04-25:
@Monopolymer can you provide context?
Which board?
Any changes to the example?

Monopolymer, 2022-04-25:
@szabi-luxonis sorry for not Providing context.
I am using OAK-D-S2, I have not made any changes to the example.

Luxonis-Brandon, 2022-04-25:
Odd.  And what OS/host is being used here?

Monopolymer, 2022-04-26:
@Luxonis-Brandon its no Truoble I am glad you are taking the time to help, I am using windows 10, VS2019 on a Ryzen5 2700, Depthai-core 2.15.2

szabi-luxonis, 2022-04-26:
We also ran it on our side, with no issues on OAK-D-S2.
Is this the only example you are having issues with?
Can you try the python version: https://github.com/luxonis/depthai-python/blob/main/examples/StereoDepth/depth_post_processing.py
to rule out compilation related issues.