luxonis/depthai-core #464: depthai build failure (CMake,toolchain) ERROR LOG INCLUDED

Swazir9449, 2022-04-26:
Hi,

Running windows 10.
Visual studio 2019, C++
When I try to use cmake -S. -Bbuild -D'BUILD_SHARED_LIBS=ON' from https://github.com/luxonis/depthai-core , I get the following error: 

E:\depthaiLIB\depthai-core>cmake -S. -Bbuild -D'BUILD_SHARED_LIBS=ON'
-- Using specified toolchain file: E:/depthaiLIB/depthai-core/build/generated/toolchain.cmake combined into: E:/depthaiLIB/depthai-core/build/generated/toolchain.cmake
CMake Error at build/generated/toolchain.cmake:48 (message):
  Internal error
Call Stack (most recent call first):
  build/CMakeFiles/3.23.0/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:36 (project)

-- Configuring incomplete, errors occurred!
See also "E:/depthaiLIB/depthai-core/build/CMakeFiles/CMakeOutput.log".
See also "E:/depthaiLIB/depthai-core/build/CMakeFiles/CMakeError.log".

I have attached the log files. 
[CMakeError.log](https://github.com/luxonis/depthai-core/files/8567187/CMakeError.log)
[CMakeOutput.log](https://github.com/luxonis/depthai-core/files/8567188/CMakeOutput.log)

themarpe, 2022-04-26:
Hi @Swazir9449 
Can you try removing the build directory and try again? Toolchain merging procedure can cause such issues - we are yet to address this

Swazir9449, 2022-04-26:
Just to confirm, do you mean removing everything that is inside the build directory? and then running the command again?

Swazir9449, 2022-04-26:
It worked. I passed that stage of the install. Thank you.

closed 2022-04-26 by Swazir9449

iurii-provizio, 2022-05-10:
This is reproduced for me. I don't agree deleting the build directory can be considered a solution, as it forces me to do a clean rebuild every time I do a build of my project, which may take ages. Need a proper fix (not sure even where to look to fix it myself).

themarpe, 2022-05-10:
@iurii-provizio 
Addressed in `develop` with 4747c83733cf2bea07775c62d17dc9c18974de71

iurii-provizio, 2022-05-11:
> @iurii-provizio Addressed in `develop` with [4747c83](https://github.com/luxonis/depthai-core/commit/4747c83733cf2bea07775c62d17dc9c18974de71)
> 
> If error still occurs, remove build directory, or for a cleaner resolution: `rm build/generated/toolchain.cmake` and remove `CMAKE_TOOLCHAIN_PATH` cache variable from `build/CMakeCache.txt` file

Thank you, it seems to have solved the issue!