luxonis/depthai-core #465: OAK-D PoE: Delayed/missed IMU data in pipeline with multiple image streams

RR-BenSteer, 2022-04-27:
I am trying to create a visual-inertial SLAM + 3D reconstruction application using an OAK-D PoE unit and a Jetson TX2. This application requires low latency and fairly time-consistent data. When I create a pipeline with only an IMU node, I have no trouble and can read messages very fast. But when creating a pipeline with multiple image streams (stereo, depth and RGB all together) I experience 100-500ms IMU data delays and am missing messages.

I'm wondering if this is expected behaviour, or if there is a way to prioritise the IMU stream? The reason I ask if it is expected behaviour is [this issue from depthai-hardware](https://github.com/luxonis/depthai-hardware/issues/8#issue-643312958) which mentioned _"WRT the USB transfer of IMU data, XLink won't be a good option if we want low latency, as large transfers (from video streams) would delay the transmission."_ I'm not using a USB connection, but is this related to what I'm experiencing?

### Setup
- depthai-core v2.15.2
- NVIDIA Jetson TX2 with Ubuntu 18.04 LTS
- ethernet connection through gigabit switch + PoE injector

### Reproduce
C++ program which pulls left/right stereo, RGB, depth (not disparity) and IMU data streams:
- cameras set to FPS 30
- 1080p RGB, 400p stereo
- depth aligned to RGB
- gyro and accelero rates 400hz
- IMU setBatchReportThreshold 1, setMaxBatchReports 10
- output queues set to non-blocking
- output queue size 4 for images, size 50 for IMU
- use tryGetAll for image queues and tryGet for IMU queue

### Result
IMUData messages are only being received every 100-500 ms. When collating all IMU data received, there are chunks of high frequency data, separated by 100-500ms timestamp jumps, i.e. messages are being missed.

Thanks!

szabi-luxonis, 2022-04-27:
@RR-BenSteer 

POE is communicating through TCP, which is not suitable for low latency streaming.
On top of that, with the number of streams you have enabled, the bandwidth is saturated.
In this case I would suggest increasing the batch size of IMU packets, to not miss them. E.g. 
`IMU setBatchReportThreshold 30, setMaxBatchReports 30` or higher.

RR-BenSteer, 2022-04-28:
Thanks for the reply! We initially bought the PoE model because of the IP67 rating, but I will pick up a USB model for better bandwidth. Good to know that this is expected behaviour.

I tried your advice and that has indeed fixed the timestamp jumps (missing messages), although the delays are still present (presumably bandwidth-related). I used a value of 28 for both settings. I wasn't clear on the usage of setBatchReportThreshold. Thank you :)

closed 2022-04-28 by RR-BenSteer

szabi-luxonis, 2022-04-28:
@RR-BenSteer for POE latency we have a branch that brings improvements: https://github.com/luxonis/depthai-core/pull/428

From 100+ ms to ~30ms latency.