luxonis/depthai-core #466: [14442C10D17960D700] [197.047] [system] [critical] Fatal error. Please report to developers. Log: 'PlgWarpHW' '411'

friend045, 2022-04-27:
Hi. While running a c++, I got

`[14442C10D17960D700] [197.047] [system] [critical] Fatal error. Please report to developers. Log: 'PlgWarpHW' '411'`

 but python was a success.

**Setup**
- depthai-core v2.15.3, built for Release X64
- VS2019 v16.9.4
- Microsoft Windows [Version 10]
- OAK-D

**Repro**
I have a model, after getting the X, and Y coordinates, the world coordinates X, Y, and Z are taken through the depth map.

**Result**
`[14442C10D17960D700] [197.047] [system] [critical] Fatal error. Please report to developers. Log: 'PlgWarpHW' '411'`

szabi-luxonis, 2022-04-27:
@friend045 
Can you share some details about the pipeline about how stereo node is configured?
The error points to output depth size not multiple of 16 bytes (HW limitation) when doing RGB alignment. CC: @alex-luxonis 
Are you using `stereo->setOutputSize(width, height)` to explicitly set the depth map size?
If not, I suggest doing so, a good value would be the stereo cameras resolution. 
If the RGB camera is set to 12 MP (4056x3040) the output depth map is aligned to that resolution, 4056 is not a multiple of 16.

friend045, 2022-04-27:
@szabi-luxonis ,
Now I have another question. 
This happens every time in the first few frames of the program. 
How can I avoid it?
Thanks.

```
[14442C10D17960D700] [20.808] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
[14442C10D17960D700] [20.824] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
[14442C10D17960D700] [20.836] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
[14442C10D17960D700] [20.850] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
[14442C10D17960D700] [20.866] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
[14442C10D17960D700] [20.880] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
[14442C10D17960D700] [20.899] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
[14442C10D17960D700] [20.913] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
[14442C10D17960D700] [20.933] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
[14442C10D17960D700] [20.961] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
[14442C10D17960D700] [21.033] [SpatialLocationCalculator(6)] [error] No ROI specified for depth calculator! Skipping frame.
```

szabi-luxonis, 2022-04-27:
@friend045 
That error is because you didn't provide initial config, so there is no ROI to calculate spatial data. If you want to avoid the error you need to provide a initial config like here: https://github.com/luxonis/depthai-core/blob/main/examples/SpatialDetection/spatial_location_calculator.cpp#L55

friend045, 2022-04-27:
@szabi-luxonis ,
You can solve my problem quickly every time, great!
I have another small question. Hope you can help, thanks.

```
[14442C10D17960D700] [43.677] [XLinkOut(8)] [error] Message has too much metadata (97628B) to serialize. Maximum is 51200B. Dropping message
[14442C10D17960D700] [43.825] [XLinkOut(8)] [error] Message has too much metadata (98222B) to serialize. Maximum is 51200B. Dropping message
[14442C10D17960D700] [43.968] [XLinkOut(8)] [error] Message has too much metadata (98816B) to serialize. Maximum is 51200B. Dropping message
[14442C10D17960D700] [44.122] [XLinkOut(8)] [error] Message has too much metadata (99410B) to serialize. Maximum is 51200B. Dropping message
[14442C10D17960D700] [44.278] [XLinkOut(8)] [error] Message has too much metadata (100004B) to serialize. Maximum is 51200B. Dropping message
[14442C10D17960D700] [44.444] [XLinkOut(8)] [error] Message has too much metadata (100598B) to serialize. Maximum is 51200B. Dropping message
[14442C10D17960D700] [44.587] [XLinkOut(8)] [error] Message has too much metadata (101258B) to serialize. Maximum is 51200B. Dropping message
```

 Every time it runs for a while, there is a problem.
Is there a way to release buffer messages manually?

szabi-luxonis, 2022-04-27:
@friend045 it looks like you set a lot of ROIs for the spatial location calculator? In the order of 1000s?
There is a limit of how much you can set in a message. Don't have exact numbers, but It's well over 1000.
Is this an intended use case?
When sending a new config, make sure it's a new one, and doesn't contain the previous ROIs, otherwise it will accumulate and lead to the above error, due to sending 100s of ROI to spatial calculator at once.

What other nodes do you have in the pipeline? It can also come from mobilenet/yolo detection networks.
If the network is not set up correctly, the decoding process doesn't work as expected, which leads to invalid detections in the order of 100, 1000, which again leads to the above error.

friend045, 2022-04-28:
@szabi-luxonis ,
I'm using "while" in Script Node and that's causing the problem.
The problem is solved now ( has been bothering me for a week) ,great. 😀
Thanks.

closed 2022-04-28 by friend045