luxonis/depthai-core #467: [14442C10D17960D700] [606.356] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'

friend045, 2022-04-28:
Hi. While running a c++, I got

```[14442C10D17960D700] [606.356] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'```

The error may appear after about 30 minutes ~ 1 hour.

**Setup**
- depthai-core v2.15.3, built for Release X64
- VS2019 v16.9.4
- Microsoft Windows [Version 10]
- OAK-D

**Repro**
I have a model, after getting the X, and Y coordinates, the world coordinates X, Y, and Z are taken through the depth map.
The error may appear after about 30 minutes ~ 1 hour.

**Result**
```[14442C10D17960D700] [606.356] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'```

szabi-luxonis, 2022-04-28:
@friend045 
Can you share some details about your pipeline? (Or if possible the whole pipeline)
What nodes are you using etc.

friend045, 2022-04-29:
@szabi-luxonis , 

The whole process is probably like this...

    // Create pipeline

    // Create RGB Camera

    // Create Mono Camera

    // Creating Stereo Depth

    // LR-check is required for depth alignment

    // Linking (mono left and right)

    // Creating Point Neural Network

    // Creating ImageManip

    // Linking (RGB -> ImageManip) (ImageManip -> Point NN)
 
    // Create Spatial Location Calculator 

    // Set Spatial Location Calculator 

    // Linking (stereoDepth -> spatialLocationCalculator)

    // Create Script Node (Computational depth)

    // Linking (Point NN -> Script Node)

    // Linking (Script Node -> spatialLocationCalculator)

    // Create XLinkOut spatialLocationCalculator

Thanks.

szabi-luxonis, 2022-04-29:
@friend045 there may be issues with the script node. Possible to share that part?
Make sure the script node is configured to run on `CSS` CPU, like here: https://github.com/luxonis/depthai-experiments/blob/master/gen2-face-recognition/main.py#L156
By default it's set to `CSS`
There are some known issues when running on `MSS`.

diablodale, 2022-05-02:
@szabi-luxonis, where are the known issues of script node on MSS? If it is not public, I ask a new issue be created to track the unpublished issues so that I can subscribe to it and be notified when they are internally fixed.

friend045, 2022-05-03:
I've made sure the script node is configured to run on the CSS CPU. 
Like: https://github.com/luxonis/depthai-experiments/blob/master/gen2-face-recognition/main.py#L156
But also the same problem.

> @friend045 there may be issues with the script node. Possible to share that part?
https://docs.google.com/presentation/d/1MhCDnTTvUNZvOH-U1ckWjS4rILaDWW0NZhAEIVTCWtc/edit?usp=sharing

szabi-luxonis, 2022-05-06:
@friend045 

The script should run in a while loop

I will share a pseudo-code, given the confidentiality of the script (can you enable commenting in the document?)

e.g.
```
while True:
point=...get()
if face is not None and point is not None:
*create config and cfg objects here*
*add extra logic, check if ROI is inside of image*
node.send(cfg)
```

Make sure that config objects are not shared and modified between different runs, e.g. always create them temporary inside the while loop

friend045, 2022-05-09:
@szabi-luxonis ,
I have updated my code, thanks.

closed 2022-05-18 by friend045