luxonis/depthai-core #485: windows compilation in external project - not Cmake -

pierrextardif, 2022-05-20:
Hi,

I'm trying to compile depthai-core with a non Cmake project on windows.
Cmake works like a charm no problems, just need non Cmake project sturcture here if possible, :).

I've added the following to my Visual Studio project:

1) C/C++ Additional include directories:

```sh
C:/Users/pierr/Documents/DEV/DEPTHAI_DEV/depthai-core/include
C:/Users/pierr/Documents/DEV/DEPTHAI_DEV/depthai-core/shared/depthai-shared/include
C:/Users/pierr/Documents/DEV/DEPTHAI_DEV/depthai-core/shared/depthai-bootloader-shared/include
C:/Users/pierr/Documents/DEV/DEPTHAI_DEV/depthai-core/shared/depthai-shared/3rdparty
C:/.hunter/_Base/cb0ea1f/4a379ff/9850c45/Install/include
```

2) Linker Additionnal Dependencies

```sh
C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib\depthai-opencv.lib
C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib\depthai-resources.lib
```

3) Linker Command Line added options:

```sh
/DEPTHAI_TARGET_CORE /JSON_USE_IMPLICIT_CONVERSIONS=1 /__PC__ /XLINK_USE_MX_ID_NAME 
```

as detailed in the [integreation](https://github.com/luxonis/depthai-core#non-cmake-integration-visual-studio-xcode-codeblocks-) of the readme.

I get the following [LNK2001](https://docs.microsoft.com/en-us/cpp/error-messages/tool-errors/linker-tools-error-lnk2001?f1url=%3FappId%3DDev16IDEF1%26l%3DEN-US%26k%3Dk(LNK2001)%26rd%3Dtrue&view=msvc-170) errors on compilation:

[image]

[Here ](https://drive.google.com/file/d/1u8KRunSceku954dRFD8L_FNqPl2ouRao/view?usp=sharing)is the project, so you can test it yourself?
I'm using Visual studio 2019 with 1503d545d4caad91e0c1a8436c7951ff79b400ef master branch.

Thanks for your help.

Best,

P

themarpe, 2022-05-20:
Hi @pierrextardif 

We've observed a similar issue in some other case as well, wondering if non CMake integration isn't working currently on Windows.

Will take a look at this and get back to you

themarpe, 2022-05-20:
@pierrextardif 
Builds for me but with a couple of changes.
Build dynamic/shared library - using `cmake build -D BUILD_SHARED_LIBS=ON` and setting `depthai-core.lib` under Linker -> Input -> Additional Dependencies.
Also, this applies for a Release build of depthai library. When building, build depthai-opencv target. Will build core along as well.

pierrextardif, 2022-05-20:
@themarpe ,

Thanks for your answer.
It seems like it's not working for me.

Just to clarify:

> Build dynamic/shared library - using `cmake build -D BUILD_SHARED_LIBS=ON`

I'v added a bool flag in my Cmake GUI :
[image]

Then when building, I build the solution for Depthai-open, which already has the depthai-core.lib as a linked lib:

[image]

It is building a `.dll` : `[...]Release\depthai-opencv.dll`

Then when building my external Visual Studio project, you mean to have:

```sh
C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib\depthai-opencv.lib
C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib\depthai-core.lib
C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib\depthai-resources.lib
```

?

Does it matter which commit I'm on on the master branch?

themarpe, 2022-05-20:
> Then when building my external Visual Studio project, you mean to have:
C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib\depthai-opencv.lib
C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib\depthai-core.lib
C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib\depthai-resources.lib

Correct. But resources shouldn't be needed anymore. (make sure you do install again).

Also for consistency sake rely only on one. Either install and use the installed files, or use the files from build directory.
You seem to be mixing both now. (WRT include, library and runtime dependencies)

I guess we'll have to also provide a VS sln along in prebuilt binaries as a template to get up and running faster

> Does it matter which commit I'm on on the master branch?

Latest commit on `main` should do. Nothing new on `develop` wrt to this currently.

pierrextardif, 2022-05-23:
Hi,

I'm not sure i understand, and I have the same probelm still.

Here is what i have been doing :

### Method1 cmake-gui + visual studio

cmake-gui 3.23
Open depthai-core then set correct build folder, flags, etc:
[image]

Open project in Visual studio and build the depthai-opencv solution.
I get files in `C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib` which i then link in my external project.

Either i link :
```
C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib\depthai-core.lib
C:\Users\pierr\Documents\DEV\DEPTHAI_DEV\build_4\install\lib\depthai-opencv.lib
C:\dev\opencv\opencv\build\x64\vc15\lib\opencv_world455.lib
```

or just the `opencv.lib` one it is not working either. - iget the same linker error.

### Method2 cmake command 

from the `x64 Command Promp`  do `cmake -S. -Bbuild -D'BUILD_SHARED_LIBS=ON'` then `cmake --build build --target install ` .
Then link the created files from `depthai-core/build/installed/libs` .

What do you mean by : 

>  Either install and use the installed files, or use the files from build directory.
You seem to be mixing both now. (WRT include, library and runtime dependencies)

Best,

pierrextardif, 2022-05-24:
@themarpe seems like a quick one for you, any tips?

Best,

themarpe, 2022-05-24:
@pierrextardif 

Are you building for x64 or Win32? Also make sure you are building in same config (Release vs Debug)

If time permits, I'll prepare a template VS solution for integration.

pierrextardif, 2022-06-07:
@themarpe yes Release for x64.

Seems like i still had no luck. any chance you could provide a template please?
Or trouble shoot the answer above?

themarpe, 2022-06-14:
@pierrextardif 
[depthai-integration.zip](https://github.com/luxonis/depthai-core/files/8895743/depthai-integration.zip)

Extract into: `depthai-core/msvc/[depthai-integration.sln/...]`
Build depthai-core and install:
```
cmake -S depthai-core -B depthai-core/build -D BUILD_SHARED_LIBS=ON
cmake --build depthai-core/build --parallel --target install --config Debug
```

Copy libusb libraries from: `depthai-core/build/install/lib/cmake/depthai/dependencies/bin` to: `depthai-core\build\install\bin`

Make sure the OpenCV is installed under `C:/opencv/[build/x64/...]` otherwise modify the include, link and PATH (`Environment variables`, under Debugging) settings

closed 2022-07-11 by themarpe