luxonis/depthai-core #486: [multi_cam_support] Multiple declarations in functions shadows class members

oysstu, 2022-05-20:
In the branch multi_cam_support, there are numerous declarations that shadows variables defined in the class. I recommend enabling -Werror=shadow in order to catch these. Below are some of the most obvious cases:

https://github.com/luxonis/depthai-core/blob/a20d06d361ca19123e846f1641dc35297d187466/include/depthai/pipeline/Node.hpp#L72-L80

https://github.com/luxonis/depthai-core/blob/a20d06d361ca19123e846f1641dc35297d187466/include/depthai/pipeline/Node.hpp#L152-L192

https://github.com/luxonis/depthai-core/blob/a20d06d361ca19123e846f1641dc35297d187466/include/depthai/pipeline/node/SPIOut.hpp#L46

themarpe, 2022-05-20:
@oysstu 
What is the actual shadowing in this case? Can you share the error message?

oysstu, 2022-05-20:
Hi, in the first linked code snippet, the constructor parameter "group" shadows the class definition. For the second snippet, "group", "blocking", and "queueSize" are shadowed by some of the constructors. There's also some shadowing in LockingQueue.hpp and depthai-shared. Partial compilation log is attached below:
```
In file included from depthai-core/include/depthai/pipeline/datatype/ImgFrame.hpp:8,
                 from depthai-core/include/depthai/pipeline/datatype/ImageManipConfig.hpp:8,
                 from depthai-core/include/depthai/pipeline/node/ImageManip.hpp:4,
                 from depthai-core/include/depthai/pipeline/nodes.hpp:11,
                 from depthai-core/include/depthai/depthai.hpp:17
depthai-core/include/depthai/build/config.hpp:7: error: "DEPTHAI_HAVE_OPENCV_SUPPORT" redefined [-Werror]
    7 | #define DEPTHAI_HAVE_OPENCV_SUPPORT
      | 
<command-line>: note: this is the location of the previous definition
In file included from depthai-core/include/depthai/pipeline/node/XLinkOut.hpp:3
depthai-core/include/depthai/pipeline/Node.hpp: In constructor ‘dai::Node::Output::Output(dai::Node&, std::string, std::string, Type, std::vector<dai::Node::DatatypeHierarchy>)’:
depthai-core/include/depthai/pipeline/Node.hpp:79:39: error: declaration of ‘group’ shadows a member of ‘dai::Node::Output’ [-Werror=shadow]
   79 |         Output(Node& par, std::string group, std::string n, Type t, std::vector<DatatypeHierarchy> types)
      |                           ~~~~~~~~~~~~^~~~~
depthai-core/include/depthai/pipeline/Node.hpp:72:21: note: shadowed declaration is here
   72 |         std::string group = "";
      |                     ^~~~~
depthai-core/include/depthai/pipeline/Node.hpp: In constructor ‘dai::Node::Output::Output(dai::Node&, std::string, std::string, Type, std::vector<dai::Node::DatatypeHierarchy>)’:
depthai-core/include/depthai/pipeline/Node.hpp:79:39: error: declaration of ‘group’ shadows a member of ‘dai::Node::Output’ [-Werror=shadow]
   79 |         Output(Node& par, std::string group, std::string n, Type t, std::vector<DatatypeHierarchy> types)
      |                           ~~~~~~~~~~~~^~~~~
depthai-core/include/depthai/pipeline/Node.hpp:72:21: note: shadowed declaration is here
   72 |         std::string group = "";
      |                     ^~~~~
depthai-core/include/depthai/pipeline/Node.hpp: In constructor ‘dai::Node::Output::Output(dai::Node&, std::string, std::string, Type, std::vector<dai::Node::DatatypeHierarchy>)’:
depthai-core/include/depthai/pipeline/Node.hpp:79:39: error: declaration of ‘group’ shadows a member of ‘dai::Node::Output’ [-Werror=shadow]
   79 |         Output(Node& par, std::string group, std::string n, Type t, std::vector<DatatypeHierarchy> types)
      |                           ~~~~~~~~~~~~^~~~~
depthai-core/include/depthai/pipeline/Node.hpp:72:21: note: shadowed declaration is here
   72 |         std::string group = "";
      |                     ^~~~~
depthai-core/include/depthai/pipeline/Node.hpp: In constructor ‘dai::Node::Input::Input(dai::Node&, std::string, Type, bool, int, std::vector<dai::Node::DatatypeHierarchy>)’:
depthai-core/include/depthai/pipeline/Node.hpp:170:68: error: declaration of ‘queueSize’ shadows a member of ‘dai::Node::Input’ [-Werror=shadow]
  170 |         Input(Node& par, std::string n, Type t, bool blocking, int queueSize, std::vector<DatatypeHierarchy> types)
      |                                                                ~~~~^~~~~~~~~
depthai-core/include/depthai/pipeline/Node.hpp:158:27: note: shadowed declaration is here
  158 |         tl::optional<int> queueSize;
      |                           ^~~~~~~~~
depthai-core/include/depthai/pipeline/Node.hpp:170:54: error: declaration of ‘blocking’ shadows a member of ‘dai::Node::Input’ [-Werror=shadow]
  170 |         Input(Node& par, std::string n, Type t, bool blocking, int queueSize, std::vector<DatatypeHierarchy> types)
      |                                                 ~~~~~^~~~~~~~
depthai-core/include/depthai/pipeline/Node.hpp:157:28: note: shadowed declaration is here
  157 |         tl::optional<bool> blocking;
      |                            ^~~~~~~~
depthai-core/include/depthai/pipeline/Node.hpp: In constructor ‘dai::Node::Input::Input(dai::Node&, std::string, Type, bool, int, std::vector<dai::Node::DatatypeHierarchy>)’:
depthai-core/include/depthai/pipeline/Node.hpp:170:68: error: declaration of ‘queueSize’ shadows a member of ‘dai::Node::Input’ [-Werror=shadow]
  170 |         Input(Node& par, std::string n, Type t, bool blocking, int queueSize, std::vector<DatatypeHierarchy> types)
      |                                                                ~~~~^~~~~~~~~
depthai-core/include/depthai/pipeline/Node.hpp:158:27: note: shadowed declaration is here
  158 |         tl::optional<int> queueSize;
      |                           ^~~~~~~~~
depthai-core/include/depthai/pipeline/Node.hpp:170:54: error: declaration of ‘blocking’ shadows a member of ‘dai::Node::Input’ [-Werror=shadow]
  170 |         Input(Node& par, std::string n, Type t, bool blocking, int queueSize, std::vector<DatatypeHierarchy> types)
      |                                                 ~~~~~^~~~~~~~
depthai-core/include/depthai/pipeline/Node.hpp:157:28: note: shadowed declaration is here
  157 |         tl::optional<bool> blocking;
      |                            ^~~~~~~~
[... cut ...]
In file included from depthai-core/include/depthai/pipeline/nodes.hpp:16:
depthai-core/include/depthai/pipeline/node/SPIOut.hpp: In member function ‘void dai::node::SPIOut::setBusId(int)’:
depthai-core/include/depthai/pipeline/node/SPIOut.hpp:46:23: error: declaration of ‘id’ shadows a member of ‘dai::node::SPIOut’ [-Werror=shadow]
   46 |     void setBusId(int id) {
      |                   ~~~~^~
depthai-core/include/depthai/pipeline/Node.hpp:274:14: note: shadowed declaration is here
  274 |     const Id id;
      |              ^~
```

themarpe, 2022-05-20:
Upon a quick glance these are all okay (at least for the Node Input/Output ones)

oysstu, 2022-05-20:
Ok, looks like GCC 12.1 is too aggressive for reporting shadowing in constructors. The variable resolution should work fine for identical names provided that initialization lists are used. There is an upstream gcc bug regarding this:

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=78147

closed 2022-05-20 by oysstu