luxonis/depthai-core #494: Depth frame returns as 8U instead of 16U

jennib3, 2022-06-02:
When following the example here: https://github.com/luxonis/depthai-core/blob/main/examples/StereoDepth/stereo_depth_video.cpp I wanted to confirm that the returned image for "depth" was in fact 16 bit depth (my understanding is that this is mapped to 16 bits corresponding to the mm measurement of the depth of that pixel).

The value returned was an 8U instead of a 16U for the cv::Mat. Am I approaching this incorrectly or accessing the depth image in the wrong way?

Thanks

szabi-luxonis, 2022-06-02:
@jennib3 it's U16, perhaps you were looking at disparity?
Add the following to the [code](https://github.com/luxonis/depthai-core/blob/main/examples/StereoDepth/stereo_depth_video.cpp#L111):
```
                auto depth = depthQueue->get<dai::ImgFrame>()->getCvFrame();
                std::cout << "Depth type: " << depth.type() << std::endl;
                std::cout << "CV_16UC1: " << CV_16UC1 << std::endl;
                cv::imshow("depth", depth);
```         
Prints:
```
Depth type: 2
CV_16UC1: 2
```

jennib3, 2022-06-02:
@szabi-luxonis 

Good thought, but I'm not certain how it could be as I've removed "disp" from the streams (only pushing "rgb" and "depth").

```if(name == "depth") {
                    cv::Mat testFrame = latestPacket[name]->getFrame();
                    
                     cout << testFrame.type() << endl;
```
This results in a "0" for me.

szabi-luxonis, 2022-06-02:
In the above example you have to set outputDepth in order to obtain depth map.
See here: https://github.com/luxonis/depthai-core/blob/cedff3f6419e07dca5fd9c5a087432f665613db8/examples/StereoDepth/stereo_depth_video.cpp#L8

jennib3, 2022-06-02:
Ah, That got me on the right track. The disparity was being linked into the depthOut stream:

```
IS:
stereo->disparity.link(depthOut->input);
SB:
stereo->depth.link(depthOut->input);
```

Thanks!

closed 2022-06-02 by jennib3