luxonis/depthai-core #495: [Feature-Request] Get the Exposure Time and ISO value set after enabling Auto Exposure

manujtrehan, 2022-06-02:
Would it be possible to add in the ability to read/get the actual exposure time and ISO value set after auto exposure is enabled?

themarpe, 2022-06-20:
Hi @manujtrehan 

Sorry for the late reply - https://github.com/luxonis/depthai-core/pull/191 this branch adds this functionality. TBD when it'll be merged to main

manujtrehan, 2022-06-20:
@themarpe  I tried the CameraControl getters, but they only seem to work if the setters have been used to set a value first. I wanted the exposure value when AutoExposure is used, but I get a 0 if I do not set a value first. Am I missing something or is that not included in that branch yet?

Erol444, 2022-06-20:
Hello @manujtrehan , I have added a [simple demo script](https://github.com/luxonis/depthai-python/blob/report_cam_settings/examples/mixed/report_camera_settings.py) that report camera settings. It currently doesn't work for MonoCamera - only ColorCamera.

manujtrehan, 2022-06-20:
@Erol444 Thanks. I was trying it out with the MonoCamera, that was the issue I guess.

chengguizi, 2022-07-14:
@Erol444  Does it work for any color cameras? Say OV9782, AR0234?

cyberbeat, 2022-08-31:
It would be nice, if this would also work for mono-cameras. When will this land in offical release?