luxonis/depthai-core #502: build error. how to think about this problem  error ==> if(${status} STREQUAL "-")

jaiminlee, 2022-06-10:
$ cmake -S. -Bbuild -D'CMAKE_INSTALL_PREFIX=[./]'
-- Using existing generated toolchain
-- Using toolchain file: /home/kt/Camera_Test/depthai-core-main/build/generated/toolchain.cmake
-- [hunter] Calculating Toolchain-SHA1
-- [hunter] Calculating Config-SHA1
-- [hunter] HUNTER_ROOT: /home/kt/.hunter
-- [hunter] [ Hunter-ID: cb0ea1f | Toolchain-ID: a3a48bb | Config-ID: 07e4a3f ]
-- [hunter] NLOHMANN_JSON_ROOT: /home/kt/.hunter/_Base/cb0ea1f/a3a48bb/07e4a3f/Install (ver.: 3.9.1)
-- [hunter] XLINK_ROOT: /home/kt/.hunter/_Base/cb0ea1f/a3a48bb/07e4a3f/Install (ver.: luxonis-2021.4.2-develop)
-- [hunter] BZIP2_ROOT: /home/kt/.hunter/_Base/cb0ea1f/a3a48bb/07e4a3f/Install (ver.: 1.0.8-p0)
-- [hunter] FP16_ROOT: /home/kt/.hunter/_Base/cb0ea1f/a3a48bb/07e4a3f/Install (ver.: luxonis-0.0.0)
-- [hunter] LIBARCHIVE-LUXONIS_ROOT: /home/kt/.hunter/_Base/cb0ea1f/a3a48bb/07e4a3f/Install (ver.: hunter-3.5.2)
-- [hunter] SPDLOG_ROOT: /home/kt/.hunter/_Base/cb0ea1f/a3a48bb/07e4a3f/Install (ver.: 1.8.2)
-- [hunter] ZLIB_ROOT: /home/kt/.hunter/_Base/cb0ea1f/a3a48bb/07e4a3f/Install (ver.: 1.2.11-p2)
-- [hunter] BACKWARD_ROOT: /home/kt/.hunter/_Base/cb0ea1f/a3a48bb/07e4a3f/Install (ver.: 1.6)
-- [hunter] LIBNOP_ROOT: /home/kt/.hunter/_Base/cb0ea1f/a3a48bb/07e4a3f/Install (ver.: 1.0-ec8f75a)
CMake Error at shared/depthai-shared.cmake:38 (string):
  string sub-command SUBSTRING requires four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:162 (include)

CMake Error at shared/depthai-shared.cmake:39 (if):
  if given arguments:

    "STREQUAL" "-"

  Unknown arguments specified
Call Stack (most recent call first):
  CMakeLists.txt:162 (include)

-- Configuring incomplete, errors occurred!
See also "/home/kt/Camera_Test/depthai-core-main/build/CMakeFiles/CMakeOutput.log".
See also "/home/kt/Camera_Test/depthai-core-main/build/CMakeFiles/CMakeError.log".

themarpe, 2022-06-10:
@jaiminlee did you git clone or downloaded the sources as zip? If the latter, use the "prepackaged" release zip / tar, which includes the necessary git submodules

dwffls, 2022-07-06:
I'm running into the same problem. I've downloaded this [depthai-core-v2.16.0.zip](https://github.com/luxonis/depthai-core/releases/download/v2.16.0/depthai-core-v2.16.0.zip) file from the releases page.

Running the following command (from the README.md):
```bash
cmake -S. -Bbuild -D'BUILD_SHARED_LIBS=ON'
cmake --build build
```

And i'm getting the following output: 
```bash
-- Generating new toolchain...
-- Using toolchain file: /home/dwffls/edward/ros2_ws/src/sensors/oak_camera/include/depthai-core/build/generated/toolchain.cmake
-- The CXX compiler identification is GNU 11.2.0
-- The C compiler identification is GNU 11.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Setting build type to 'Release' as none was specified.
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- [hunter] Calculating Toolchain-SHA1
-- [hunter] Calculating Config-SHA1
-- [hunter] HUNTER_ROOT: /home/dwffls/.hunter
-- [hunter] [ Hunter-ID: cb0ea1f | Toolchain-ID: 3f80026 | Config-ID: 07e4a3f ]
-- [hunter] NLOHMANN_JSON_ROOT: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install (ver.: 3.9.1)
-- [hunter] XLINK_ROOT: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install (ver.: luxonis-2021.4.2-develop)
-- [hunter] BZIP2_ROOT: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install (ver.: 1.0.8-p0)
-- [hunter] FP16_ROOT: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install (ver.: luxonis-0.0.0)
-- [hunter] LIBARCHIVE-LUXONIS_ROOT: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install (ver.: hunter-3.5.2)
-- [hunter] SPDLOG_ROOT: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install (ver.: 1.8.2)
-- [hunter] ZLIB_ROOT: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install (ver.: 1.2.11-p2)
-- [hunter] BACKWARD_ROOT: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install (ver.: 1.6)
-- [hunter] LIBNOP_ROOT: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install (ver.: 1.0-ec8f75a)
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Could NOT find libdw (missing: LIBDW_LIBRARY LIBDW_INCLUDE_DIR) 
-- Could NOT find libbfd (missing: LIBBFD_LIBRARY LIBBFD_INCLUDE_DIR) 
-- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR LIBELF_LIBRARY LIBELF_INCLUDE_DIR) 
-- Found Backward: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install/lib/backward  
-- Found nlohmann_json: /home/dwffls/.hunter/_Base/cb0ea1f/3f80026/07e4a3f/Install/lib/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found suitable version "3.9.1", minimum required is "3.6.0") 
CMake Error at shared/depthai-shared.cmake:38 (string):
  string sub-command SUBSTRING requires four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:162 (include)

CMake Error at shared/depthai-shared.cmake:39 (if):
  if given arguments:

    "STREQUAL" "-"

  Unknown arguments specified
Call Stack (most recent call first):
  CMakeLists.txt:162 (include)

-- Configuring incomplete, errors occurred!
See also "/home/dwffls/edward/ros2_ws/src/sensors/oak_camera/include/depthai-core/build/CMakeFiles/CMakeOutput.log".
```

By commenting out the following lines, I am able to build the library as normal:
- depthai-shared.cmake: line 38-41
- depthai-bootloader-shared.cmake: line 32-35

themarpe, 2022-07-06:
@dwffls 
I've created a PR addressing this: #514 
Do you mind giving it a try?

dwffls, 2022-07-06:
Output of your command is: 
```
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
/home/dwffls/edward
true
```
And it is correct that the "edward" folder is also a git repository. Gonna try the PR right now

dwffls, 2022-07-06:
Checkoud out #514 and got the ```git submodule  update --init --recursive``` error on the first run and after running the command, the build succeeded. 

Should be fixed now! Thanks for the quick help!

closed 2022-07-06 by themarpe