luxonis/depthai-core #505: Building for Android

Bercon, 2022-06-16:
v2.16.0 release notes mentioned that Android is a supported platform. When trying to build for it, I run into issue that XLink is using Android logging, but it's not included as a link dependency (to my understanding), so to build fails. Are there instructions I missed on how to build should be done?

Trying this on Ubuntu 20.04 LTS and 22.04 LTS
```
mkdir target
cd target
cmake -DANDROID_ABI=arm64-v8a \
   -DCMAKE_BUILD_TYPE=Release \
   -DANDROID_PLATFORM=android-24 \
   -DANDROID_NDK="/home/jerry/Android/Sdk/ndk/23.2.8568313" \
   -DCMAKE_TOOLCHAIN_FILE="/home/jerry/Android/Sdk/ndk/23.2.8568313/build/cmake/android.toolchain.cmake" \
   -DCMAKE_FIND_ROOT_PATH="/home/jerry/Android/Sdk/ndk/23.2.8568313" \
   -DANDROID_TOOLCHAIN=clang \
   -DCMAKE_CXX_COMPILER_LAUNCHER=ccache  \
   -DBUILD_SHARED_LIBS=ON \
   -DCMAKE_INSTALL_PREFIX="./install/" \
   -DCMAKE_INSTALL_LIBDIR=lib \
   -DDEPTHAI_BUILD_EXAMPLES=OFF  \
   ..
make -j16

...
100%] Linking CXX shared library libdepthai-core.so
ld: error: undefined symbol: __android_log_vprint
>>> referenced by XLinkLog.c:166 (/home/jerry/.hunter/_Base/cb0ea1f/f05d2ab/13dbff5/Build/XLink/Source/src/shared/XLinkLog.c:166)
>>>               XLinkLog.c.o:(logprintf) in archive /home/jerry/.hunter/_Base/cb0ea1f/f05d2ab/13dbff5/Install/lib/libXLink.a
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
```

themarpe, 2022-06-16:
Hi @Bercon 

No straightforward instructions as of right now. 
Sorry for this issue - the inital efforts were mostly getting everything in line from tech perspective to be able to run it on Android. Our Discord server holds some debate on that aspect: https://discord.com/channels/790680891252932659/924798955290755082

Feel free to check the following for an example: https://github.com/ibaiGorordo/depthai-android-jni-example

You may test out the following:
 - fork & clone XLink
 - Specify DepthAI to use local XLink: `-D DEPTHAI_XLINK_LOCAL=[path/to/cloned/xlink]`
 - Add the following to XLink CMakeLists.txt
```
if(MINGW)
    target_link_libraries(${TARGET_NAME}
        PUBLIC
            libwinusb.a
            libsetupapi.a
            ws2_32
            Iphlpapi.lib
    )
endif()

if(ANDROID)
    find_library(log-lib log)
    target_link_libraries(${TARGET_NAME} ${log-lib})
endif()
```

If the following works as expected, feel free to open a PR against XLink and we'll mainline:)

Bercon, 2022-06-17:
That see seems to have done the trick, thank you!

Used `target_link_libraries(${TARGET_NAME} PUBLIC ${log-lib})`, could probably be PRIVATE too.

Will do some more testing before creating a PR

themarpe, 2022-06-17:
@Bercon thanks a lot for testing it out!

Indeed, jotted that down too fast - PRIVATE is the way to go. (Edited the initial post)