luxonis/depthai-core #507: Full example of point cloud extraction is needed

tarekmuallim, 2022-06-21:
Hi,
Is their any full c++ example on extracting the point cloud?
I mean a full working example which can be tested  directly.  
Thanks in advanced.

themarpe, 2022-06-21:
CC: @Erol444 @moratom

There are some Python experiments, but we should create an experiment targeting C++ as well

tarekmuallim, 2022-07-22:
Hi,
Any update?

themarpe, 2022-07-22:
None as of right now - currently we'll are focusing more on higher abstraction in Python. These changes will slowly boil down into C++ as well, but not immediately.

For now I'd suggest going with Python instead - if C++ is necessary, then I'm afraid you'd have to write one yourself. PCL library would be a good candidate for that.
However, If time permits, we'll create an example for this specific usecase.

tarekmuallim, 2022-07-22:
Ok
Thank you for your help.