luxonis/depthai-core #511: build error

jaiminlee, 2022-06-28:


jaiminlee, 2022-06-29:
Add to cmakefiles

target_compile_options(myLib PRIVATE -fPIC)
add_compile_options(-fPIC)
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fpic")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fpic")

alex-luxonis, 2022-06-29:
@jaiminlee It's possible this is an issue with the Hunter package manager cache, can you try a cleanup:
`rm -r build`
`rm -r ~/.hunter`

nathanieltagg, 2022-07-14:
It's not a cache problem. I just ran into this on my very first attempt to install (jetson / ARM)

themarpe, 2022-07-14:
@nathanieltagg was it exact same issue? Do you mind sharing the logs? 
Also do above commands not resolve the issue for you?

nathanieltagg, 2022-07-15:
No, eventually I got it to work. I missed that I had to delete .hunter in my home directory, not the build directory.

nathanieltagg, 2023-01-10:
Just tried a clean install, and got the same problem.  This time deleting the caches and rebuilding did NOT solve the problem.
v19.1
(I'm specifically building to a custom install location, which is likely connected to the problem.)

nathanieltagg, 2023-01-10:
For future reference:
When doing a fresh install, ensure you specify BOTH shared library AND the install directory the FIRST time you invoke cmake.
 cmake -S. -Bbuild -D'BUILD_SHARED_LIBS=ON' -D'CMAKE_INSTALL_PREFIX=/install/dir'

If you do not, you must delete your ~/.hunter directory, since the build system apparently pollutes your home directory for some reason.

jwinarske, 2023-03-11:
@jaiminlee 
```
-DBUILD_SHARED_LIBS=ON -DCMAKE_POSITION_INDEPENDENT_CODE=ON
```