luxonis/depthai-core #516: Whether the monocular and IMU are the same clock？

dongxuanlb, 2022-07-11:
>Through firmware sync, we’re monitoring for drift and aligning the capture timestamps of the cameras, which are taken at the MIPI Start-of-Frame event. The Left/Right global shutter cameras are driven **by the same clock**, started by broadcast write on I2C, so no drift will happen over time, even when running freely without a sync. With the above functionality it would be also possible to configure FSIN as an output on one sensor, and an input to the other sensor. The RGB rolling shutter **has a slight difference in clocking/frame-time**, so when we detect a small drift, we’re modifying the frame time (number of lines) for the next frame by a small amount to compensate.

mentioned in this [article](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/sync_frames.html?highlight=clock)

My question is whether the monocular camera and IMU are driven  by the same clock？

in source：
1. [RawIMUData.hpp](https://github.com/luxonis/depthai-shared/blob/cb49a5c571c2ed253dfcd0aa80564291767b7a92/include/depthai-shared/datatype/RawIMUData.hpp)
    /** Generation timestamp, synced to host time */
    Timestamp timestamp = {};

    /** Generation timestamp, direct device monotonic clock */
    Timestamp tsDevice = {};
2. [RawImgFrame.hpp](https://github.com/luxonis/depthai-shared/blob/cb49a5c571c2ed253dfcd0aa80564291767b7a92/include/depthai-shared/datatype/RawImgFrame.hpp)
    Timestamp ts = {};         // generation timestamp, synced to host time
    Timestamp tsDevice = {};   // generation timestamp, direct device monotonic clock

It seems they are driven by the same clock。

But，for my test， the mono frame’s timestamp and the imu frame's timestamp are not align.

dongxuanlb, 2022-07-12:
ok。So is there any good method or idea for how to synchronize between the two? The core problem is that I can't know the time offset between them through the sequence。

szabi-luxonis, 2022-07-12:
> ok。So is there any good method or idea for how to synchronize between the two? The core problem is that I can't know the time offset between them through the sequence。

There's a confusion.
The IMU and cameras are **not driven** by the same clock, only mono cameras are driven by the same clock.
E.g. capture time is different for IMU and stereo cameras.
The timestamp captured and stored in messages are from the same free running counter. Therefore you can compare directly the timestamp from camera and timestamp from IMU since they are from the same source.

szabi-luxonis, 2022-07-12:
IMU and stereo cameras are running independently, there's no HW sync between them.
Sequence numbers are independent, timestamps are independent. In other words they are sampled independently.
The best way to align them is based on timestamp, which is taken from common source (free running counter with the same clock as source).
Offset is 100 ms, depending on how you measure it.
If you want to align IMU frames with camera frames, the best way to do it is based on device timestamp.
IMU frame X is aligned with frame Y if the timestamp of IMU frame is inside of the frame time interval for frame Y, e.g. [frame timestamp - 33.3 ms, frame timestamp]. Another, better way (for VIO, SLAM etc.) is aligning against the middle of exposure time. (PR to access exposure time is here: https://github.com/luxonis/depthai-core/pull/191)

Erol444, 2022-10-17:
Hi @dongxuanlb ,
Here's a simple [demo of IMU + frame](https://github.com/luxonis/depthai-experiments/tree/master/gen2-syncing#imu--rgb--depth-timestamp-syncing) syncing based on timestamps.
Thanks, Erik