luxonis/depthai-core #521: [Bug Report] Yolo example 100% CPU LeonRT

Erol444, 2022-07-12:
After ~15 minutes of running the depthai-python [TinyYoloV4 example](https://docs.luxonis.com/projects/api/en/latest/samples/Yolo/tiny_yolo/#rgb-tiny-yolo), it will stop as LeonRT CPU will go to 100% and streaming of frames/results will stop.
```
Test specs:
depthai version: 2.17 and 2.16
OAK-D, `SUPER` USB speed
Ubuntu 20.04
Intel i7-7700HQ
```
Log:
```
[14442C1001615ED700] [1.3] [985.040] [system] [info] Temperatures - Average: 91.53 °C, CSS: 92.56 °C, MSS 91.67 °C, UPA: 93.44 °C, DSS: 88.44 °C
[14442C1001615ED700] [1.3] [985.040] [system] [info] Cpu Usage - LeonOS 28.48%, LeonRT: 22.02%
[14442C1001615ED700] [1.3] [986.041] [system] [info] Memory Usage - DDR: 60.68 / 340.93 MiB, CMX: 2.18 / 2.50 MiB, LeonOS Heap: 21.70 / 77.60 MiB, LeonRT Heap: 4.29 / 41.39 MiB
[14442C1001615ED700] [1.3] [986.041] [system] [info] Temperatures - Average: 91.40 °C, CSS: 91.85 °C, MSS 91.32 °C, UPA: 93.27 °C, DSS: 89.16 °C
[14442C1001615ED700] [1.3] [986.041] [system] [info] Cpu Usage - LeonOS 28.64%, LeonRT: 21.08%
[14442C1001615ED700] [1.3] [987.041] [system] [info] Memory Usage - DDR: 60.68 / 340.93 MiB, CMX: 2.18 / 2.50 MiB, LeonOS Heap: 21.70 / 77.60 MiB, LeonRT Heap: 4.29 / 41.39 MiB
[14442C1001615ED700] [1.3] [987.042] [system] [info] Temperatures - Average: 91.22 °C, CSS: 92.38 °C, MSS 90.96 °C, UPA: 92.91 °C, DSS: 88.62 °C
[14442C1001615ED700] [1.3] [987.042] [system] [info] Cpu Usage - LeonOS 28.84%, LeonRT: 21.06%
[14442C1001615ED700] [1.3] [988.042] [system] [info] Memory Usage - DDR: 60.68 / 340.93 MiB, CMX: 2.18 / 2.50 MiB, LeonOS Heap: 21.70 / 77.60 MiB, LeonRT Heap: 4.29 / 41.39 MiB
[14442C1001615ED700] [1.3] [988.042] [system] [info] Temperatures - Average: 89.07 °C, CSS: 89.52 °C, MSS 88.80 °C, UPA: 91.32 °C, DSS: 86.62 °C
[14442C1001615ED700] [1.3] [988.042] [system] [info] Cpu Usage - LeonOS 26.44%, LeonRT: 34.73%
[14442C1001615ED700] [1.3] [989.043] [system] [info] Memory Usage - DDR: 60.68 / 340.93 MiB, CMX: 2.18 / 2.50 MiB, LeonOS Heap: 21.70 / 77.60 MiB, LeonRT Heap: 4.29 / 41.39 MiB
[14442C1001615ED700] [1.3] [989.043] [system] [info] Temperatures - Average: 86.33 °C, CSS: 87.35 °C, MSS 86.07 °C, UPA: 87.90 °C, DSS: 84.03 °C
[14442C1001615ED700] [1.3] [989.043] [system] [info] Cpu Usage - LeonOS 9.08%, LeonRT: 100.00%
[14442C1001615ED700] [1.3] [990.044] [system] [info] Memory Usage - DDR: 60.68 / 340.93 MiB, CMX: 2.18 / 2.50 MiB, LeonOS Heap: 21.70 / 77.60 MiB, LeonRT Heap: 4.29 / 41.39 MiB
[14442C1001615ED700] [1.3] [990.044] [system] [info] Temperatures - Average: 84.40 °C, CSS: 85.14 °C, MSS 84.77 °C, UPA: 85.70 °C, DSS: 81.97 °C
[14442C1001615ED700] [1.3] [990.044] [system] [info] Cpu Usage - LeonOS 9.24%, LeonRT: 100.00%
[14442C1001615ED700] [1.3] [991.045] [system] [info] Memory Usage - DDR: 60.68 / 340.93 MiB, CMX: 2.18 / 2.50 MiB, LeonOS Heap: 21.70 / 77.60 MiB, LeonRT Heap: 4.29 / 41.39 MiB
```
[Full log here](https://pastebin.com/syK8AUhf)

Luxonis-Brandon, 2022-07-12:
Thanks.  CC: @szabi-luxonis on your proposed fix.  @Erol444 I'm not sure if you've already tried out @szabi-luxonis 's alignment proposed solution here.  But I think it would be worthwhile.

Erol444, 2022-07-13:
@szabi-luxonis Same sporadic issue; 100% CPU usage on LeonRT after a while. Used `2.16.0.0.dev0+c051e7a402081a195cc94593a8f05f27b8bf7060`.

szabi-luxonis, 2022-07-13:
This looks like an MDK bug.
A deadlock situation in low level driver code, hardware mutex is not taken and spins in an infinite loop.

szabi-luxonis, 2022-07-14:
@Erol444 try out latest develop which includes the fix: https://github.com/luxonis/depthai-python/commit/7c9b30a9f4fddcbca9b304a6c65b9455c1861559

closed 2022-07-15 by szabi-luxonis