luxonis/depthai-core #524: cmake is time consuming, stuck in hunter

shmm91, 2022-07-15:
When i execute cmake -Hdepthai-core -Bdepthai-core/build -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=/usr/local, it stuck in hunter get package. Then the terminal will show like this:
--- LOG BEGIN ---
         Protocol "https" not supported or disabled in libcurl
I have try my curl in terminal, it's support https. The test like this:
[image]

After i enter the above command( cmake -Hdepthai-core -Bdepthai-core/build -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=/usr/local), the terminal will stuck in this way:

[image]

After about two~three hours, it will show some error like:  Protocol "https" not supported or disabled in libcurl。
Can anyone help me? Thank you!!!

shmm91, 2022-07-15:
Supply metarial. After about 3 hours later, the error occurs like i said:
[image]

themarpe, 2022-07-15:
@shmm91 
Do you mind retrying? We observed similar in our case.
If it still doesn't work, do you mind letting me know your system configuration? Which OS, any modifications, which libcurl, etc..?

shmm91, 2022-07-15:
@themarpe, thank you for your reply! I have tried several times since yesterday. 
My os is Ubuntu18.04.5, ROS melodic, Nvidia GTX1080Ti.
curl 7.77.0 (x86_64-pc-linux-gnu) libcurl/7.77.0 OpenSSL/1.1.1 zlib/1.2.11 OpenLDAP/2.4.45
Release-Date: 2021-05-26
Protocols: dict file ftp ftps gopher gophers http https imap imaps ldap ldaps mqtt pop3 pop3s rtsp smb smbs smtp smtps telnet tftp 
Features: alt-svc Asynch

themarpe, 2022-07-15:
@shmm91 
It seems you have some different/custom version of curl.
I'm on Ubuntu 20.04 and have the following:
```
curl 7.68.0 (x86_64-pc-linux-gnu) libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Release-Date: 2020-01-08
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtmp rtsp scp sftp smb smbs smtp smtps telnet tftp 
Features: AsynchDNS brotli GSS-API HTTP2 HTTPS-proxy IDN IPv6 Kerberos Largefile libz NTLM NTLM_WB PSL SPNEGO SSL TLS-SRP UnixSockets
```
And the following is default on 18.04
```
curl 7.58.0 (x86_64-pc-linux-gnu) libcurl/7.58.0 OpenSSL/1.1.1 zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Release-Date: 2018-01-24
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtmp rtsp smb smbs smtp smtps telnet tftp 
Features: AsynchDNS IDN IPv6 Largefile GSS-API Kerberos SPNEGO NTLM NTLM_WB SSL libz TLS-SRP HTTP2 UnixSockets HTTPS-proxy PSL 
```

2 possibilities it seems:
 - https://askubuntu.com/a/821269 - your custom curl wasn't built with some options like SSL/etc.. to enable https (OpenSSL is specified however, which is strange)
 - It seems like syntax issue when specifying the URL could cause a valid https URL to fail with above message. But this hasn't happened anywhere else before, which is why I'd check your curl first.

(rechecked on Ubuntu22.04 and it comes with an even newer curl, where things also work as expected)

Just to make sure we are on the same page - can you try cloning the repo with https/ssh, or use the release package? What was the current means of retrieving the sources?

shmm91, 2022-07-20:
@themarpe , thank you for your reply!
I clone the package with the following command: git clone git@github.com:luxonis/depthai-core.git.
After several days test, I have found the key issue.
 **The cmake is build without curl support.**
I re-install cmake with curl support, then everything is ok. I will close this issue, thank you!!!

closed 2022-07-20 by shmm91